lovyan03/LovyanGFX #25: Confirming the pin setup

docop, 2020-07-11:
Hi So i just want to confirm how to do the pin config for the lib. I do have a ST7789 lcd and i see in you config : Lgfx-config-esp-Wrover-kit.hpp  this :

static lgfx::Panel_ST7789 panel;
      panel.spi_3wire = false;
      panel.spi_cs   = 22;
      panel.spi_dc   = 21;
      panel.gpio_rst = 18;
      panel.gpio_bl  = 5;
      panel.pwm_ch_bl = 7;

So is this the pin connected from the esp32 to the lcd ?  As GPIO 22/21/18/5 .. but for 7 : is it Gpio 26 ?  Fow the wrover esp32 normal 30pin version, not the 36pin.
And i think loading the tft-graphictest.ino will load the LovyanGFX.hpp and this call to the Lgfx-config... 
 And is there a way to force to lcd st7789 or it goes only with an autodetect ?
Thank again

tobozo, 2020-07-11:
autodetection feature is for the ESP32-Wrover-Kit only since it exists both with ILI9341 and ST7789, but you probably don't need that

[this example](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_spi_setting/2_spi_setting.ino) shows how to configure you own custom display profile directly from the .ino file

lovyan03, 2020-07-12:
ESP-WROVER-KIT  is [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html)
[image]
[image]

closed 2020-07-17 by lovyan03