lovyan03/LovyanGFX #26: Dual display setup

a-RN-au-D, 2020-07-16:
Hello, I've been trying to setup a dual display setup, sadly it still doesn't work, 
you can find the full code here:  [https://pastebin.com/vXLQESUd](https://pastebin.com/vXLQESUd) 
my issue is that when I use this code, I can see both screen trigger during setup but the loop doesn't move forward, it just stay stuck and doesn't start the loop,  could you tell me what I'm doing wrong.

Thanks

I've define both panels like this:

`struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 23;
  static constexpr int spi_miso = -1;
  static constexpr int spi_dlen = 8;
};

//static LGFX lcd;
static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::LGFX_SPI<LGFX_Config> lcdB;
static lgfx::Panel_ST7789 panel;
static lgfx::Panel_ST7789 panelB;
static TCPReceiver recv;
static TCPReceiver recvB;`

then in setup I did this:

`//SETTING LEFT PANEL
  panel.freq_write = 40000000;
  panel.freq_fill = 40000000;
  panel.freq_read = 6000000;
  panel.spi_mode = 3;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;
  panel.spi_cs = 5;
  panel.spi_dc = 2;
  panel.gpio_rst = 4;
  panel.gpio_bl = 32;
  panel.pwm_ch_bl = 7;
  panel.backlight_level = true;
  panel.invert = true;

  //SETTING RIGHT PANEL
  panelB.freq_write = 40000000;
  panelB.freq_fill = 40000000;
  panelB.freq_read = 6000000;
  panelB.spi_mode = 3;
  panelB.spi_mode_read = 0;
  panelB.len_dummy_read_pixel = 8;
  panelB.spi_read = true;
  panelB.spi_3wire = false;
  panelB.spi_cs = 17;
  panelB.spi_dc = 2;
  panelB.gpio_rst = 4;
  panelB.gpio_bl = 32;
  panelB.pwm_ch_bl = 7;
  panelB.backlight_level = true;
  panelB.invert = true;
  //  panel.rgb_order = false;

  lcd.setPanel(&panel);
  lcdB.setPanel(&panelB); //STARTING RIGHT PANEL`

lovyan03, 2020-07-16:
Hello.
ScreenShotReceiver uses a command and DMA transfer to occupy the SPI bus.
This makes it difficult to control two panels on the same SPI bus.
The most reliable solution is to assign a separate SPI to each of the two panels.
One to the DMA1 channel of VSPI.
Set the other one to HSPI's DMA2 channel.
Assign different GPIOs to each one.

a-RN-au-D, 2020-07-16:
Great news, I was able to setup the dual display using your Party_Parrots examples and my st7789 based displays, 
I do not know how to use github but I will leave the working codes here in case someone wants to replicate it: https://pastebin.com/vQFhi7Z3
Sadly for the screenReceiver, I'm still stuck, I can not run them at the same time, it's either the right one or the left one, but not both, 
I have tried to create another instance of recv.loop, like this '
//  recv.loop(); // left display
  recvB.loop(); // right display'
but recvB doesn't receive any data, I can see that both screen work by alternating between &lcd // &lcdB in the setup, here's my current codes
https://pastebin.com/Vsu0A5zH

Thanks a lot for your help, I'm making some progress +1

lovyan03, 2020-07-16:
Congratulations. I'm glad your multi-monitor setup worked out.
From here on out, this is the topic of ScreenShotReceiver, so it is out of the scope of support for LovyanGFX.

ScreenShotReceiver has never been in demand to run multiple simultaneous It is designed to wait for a TCP connection on a specific port. It is not currently designed to handle multiple connections at the same time.
I think you'll need to modify the source code in various ways.

a-RN-au-D, 2020-07-17:
Thanks a lot for your help, I will take a better look at the tcp connection files and try to see if I can make it work :)
Btw it would suck if I didn't show you what I'm using your codes for, I've been trying to build the cheapest open source A.R. goggles and your screenshotreceiver code helped me greatly in testing video output and positioning.

Here are the links, if you have an I.G. profile, I'll gladly tag you in all the posts, 
[my profile](https://www.instagram.com/thebestphotographerinthegalaxy/)
[CheApR_glasses A ](https://www.instagram.com/p/CCsn5xvDzNB/?utm_source=ig_web_copy_link)
[CheApR_glasses B](https://www.instagram.com/p/CCi1oH8jzxp/?utm_source=ig_web_copy_link)
[CheApR_glasses C](https://www.instagram.com/p/CBsVovgjQ7Q/?utm_source=ig_web_copy_link)

lovyan03, 2020-07-17:
Your AR glasses are interesting !!
By the way, probably increasing the number of TCP connections would be difficult and slow in terms of speed.
In my opinion, it would be better to split a single horizontal JPG in the center and extract it to separate DMA buffers on the left and right sides.

If you want to continue on this topic after this, please create a new Issue in ESP32_ScreenShotReceiver.

closed 2020-07-18 by lovyan03