lovyan03/LovyanGFX #27: On Wemos D32 Pro + TFT3.5inch touch , how to test 「 MovingIcons」 example ?

BlynkGO, 2020-07-19:
I am using Wemos D32 Pro + TFT3.5inch (ILI9488) . and connect line like the following image

[image]

When I test by 「 ClockSample」example, it 's fine for me by the following config.

```cpp
struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 23;
  static constexpr int spi_miso = 19;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;

// Panelクラスのインスタンスを作成します。使用するパネルにあった記述をコメントアウトしてください。
//static lgfx::Panel_HX8357B panel;
//static lgfx::Panel_HX8357D panel;
//static lgfx::Panel_ILI9163 panel;
//static lgfx::Panel_ILI9341 panel;
//static lgfx::Panel_ILI9342 panel;
//static lgfx::Panel_ILI9486 panel;
static lgfx::Panel_ILI9488 panel;
//static lgfx::Panel_SSD1351 panel;
//static lgfx::Panel_ST7789 panel;
//static lgfx::Panel_ST7735S panel;

void panel_config(){
  panel.freq_write = 20000000;
  panel.freq_fill  = 27000000;
  panel.freq_read  = 16000000;
  panel.spi_mode = 0;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;
  panel.spi_cs = 14;
  panel.spi_dc = 27;
  panel.gpio_rst = 33;
  panel.gpio_bl  = 32;
  panel.pwm_ch_bl = 7;
  panel.backlight_level = true;
  panel.invert = false;
  panel.rgb_order = false;
  panel.memory_width  = 320;//320;
  panel.memory_height = 480;//240;
  panel.panel_width  = 320;//320;
  panel.panel_height = 480;//240;
  panel.offset_x = 0;
  panel.offset_y = 0;
  panel.rotation = 3;
  panel.offset_rotation = 0;  
}

// ....

void setup(){
  Serial.begin(115200);

#if defined(ARDUINO_M5Stick_C)
  axp.begin();
#endif

  panel_config();
  lcd.setPanel(&panel);
  lcd.init();

  //. ....
}

void loop(){
   //.....
}
```

However when I use this config for 「 MovingIcons」 example ,
it occurs some strange display like the following clip.

https://streamable.com/4qw95u

How to solve this example ?
Thank you very much.

lovyan03, 2020-07-19:
Thank you for reporting this !
I have just modified the examples to work properly with ILI9488 . The fix is now reflected in the "develop" branch.
We'll release it in the master branch soon, but if you want to try it out right away, you can get examples from the develop branch.

tobozo, 2020-07-19:
@BlynkGO just out of curiosity I've been wondering why is the espressif logo hidden on this image ?
Are there any fake/counterfeit LoLinD32Pro models out there ?

BlynkGO, 2020-07-19:
@lovyan03  Thank you very much. I will try.

@tobozo . No, there are not any fake models. It's a ESP32 Wrover module.

BlynkGO, 2020-07-19:
@lovyan03 
I have just tried by "develop" branch.
「 MovingIcons」 example  is still occurs the same result.

Any my config  is mistake ?

lovyan03, 2020-07-19:
@BlynkGO 
I'm sorry.
I updated my local branch but didn't update github.
I just redid it.
I also fixed "CollisionCircles". It's now capped at 320x240, so there's a gap in the screen with ILI9488, but I've made This is a response to the lack of memory.

BlynkGO, 2020-07-20:
「 MovingIcons」 example's result :

[image]

BlynkGO, 2020-07-20:
By the lastest 'developer' branch,
「 MovingIcons」 example  is fine for 480x320 now .

One more thing, 
I wish to use the same SPI both TFT and Touch interface.
Is there any easy way to config Touch-SPI interface can be used
the same with TFT's SPI that used by LovyanGFX ?

Thank  you very much.

BlynkGO, 2020-07-20:
[image]

Finally by the lastest 'developer' branch,
「 MovingIcons」 example  is fine for 480x320 now .
Thank  you very much.

One more thing, 
I wish to use the same SPI both TFT and Touch interface.
Is there any easy way to config Touch-SPI interface can be used
the same with TFT's SPI that used by LovyanGFX ?

lovyan03, 2020-07-20:
@BlynkGO 
Thanks for the report !
The "CollisionCirlces" sample has also been updated. If memory allocation fails, PSRAM is used. If it still fails, it reduce the screen size.

Touch screen support is planned to be implemented.
I haven't been able to work on the touch screen function because I'm giving priority to the display function at the moment, but I plan to provide a function similar to TFT_eSPI.

It is possible to use SPI when TFT is not drawing, so if you already have a library that handles the touch screen, you may be able to use it.

BlynkGO, 2020-07-20:
https://streamable.com/ro4zdt

The "CollisionCirlces" sample can display
on 480*320  between  7-15 fps , thanks again.

I wish to replace TFT-eSPI to LovyanGFX for more high-performance with TFT Touch interface.
Waiting for the next version.

Akigato Go zaimasu.

lovyan03, 2020-08-31:
@BlynkGO 
Thank you for your patience.
LovyanGFX is now touch screen compatible ! (in the master branch. )
The release will happen in the next few days.
Currently available are XPT2046 and STMPE610 (SPI) , and FT5x06 (I2C).

closed 2020-09-27 by lovyan03