lovyan03/LovyanGFX #30: ILI9486L display support

Dimision, 2020-07-27:
Hello
I recently bought this screen from Aliexpress: https://www.aliexpress.com/item/3-5-inch-8P-SPI-TFT-LCD-Color-Screen-Module-ILI9486-Drive-IC-320-480-RGB/32828284227.html

After wired it to my ESP32 board and changed some settings  in the example sketch "2_spi_setting", the display does not work at all. 
By studying the datasheet, I realized that the ILI9486L controller only support RGB111 and RGB666 mode when using SPI mode(Chapter 7.5.3, Page 45). So I changed the configuration "write_depth" from "rgb565_2Byte" to "rgb666_3Byte" in the "PanelCommon.hpp"(Line 39), and then, I upload the code, the display seems worked flawlessly.
[image]

But when I moved to the project "ScreenShotReceiver" with same configurations, the WiFi information displayed normally, but the image seems got twisted.
[image]

Here is my configuration codes, and did I miss anything important?
`struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 23;
  static constexpr int spi_miso = 19;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ILI9486 panel;

void pannel_cfg() 
{
  panel.freq_write = 20000000;
  panel.freq_fill  = 27000000;
  panel.freq_read  = 16000000;
  panel.spi_mode = 3;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = false;
  panel.spi_3wire = true;
  panel.spi_cs = -1;
  panel.spi_dc = 5;
  panel.gpio_rst = 4;
  panel.gpio_bl  = 15;
  panel.pwm_ch_bl = 7;
  panel.backlight_level = true;
  panel.invert = false;
  panel.rgb_order = false;
  panel.memory_width  = 320;
  panel.memory_height = 480;
  panel.panel_width  = 320;
  panel.panel_height = 480;
  panel.offset_x = 0;
  panel.offset_y = 0;
  panel.rotation = 0;
  panel.offset_rotation = 0;
  lcd.setPanel(&panel);
}`

lovyan03, 2020-07-27:
Hello @Dimision  .
ScreenShotReceiver was fixed about 10 days ago and now supports RGB888.
Is your ScreenShotReceiver new?

https://github.com/lovyan03/ESP32_ScreenShotReceiver/issues/3

It is now available in the master branch.

lovyan03, 2020-07-27:
and, I've just added Panel_ILI9486L to the "develop" branch, which will disable RGB565 and force it to use 3-byte colors.
You can try the "develop" branch if you like.

Dimision, 2020-07-27:
Hi Lovyan
Thanks for your reminding, I realized my local codes are actually out of date. After download the latest project, and copy the configuration codes to the new project, it works like a charm! 
I got about 10 fps at 480x320 resolution, but when I increased the SPI speed to 80 Mhz, the image getting blurry.
[image]
[image]

lovyan03, 2020-07-27:
Hi @Dimision 
I'm glad it seemed to work.
The ST7789 runs stably at SPI 80MHz, but it's difficult to get to 80MHz on an ILI panel.
In most cases, it's safe to limit freq_fill to 80 MHz and freq_write to 40 MHz .
When rewriting the entire screen, The maximum frame rate is as follows.
320x240 RGB565 40MHz  = 32 fps
320x240 RGB666 40MHz  = 22 fps
480x320 RGB666 40MHz  = 11 fps

Dimision, 2020-07-27:
Hi Lovyan
Thanks for your suggestions. I tried your settings, But unfortunately, my ST7789 panel works at 40Mhz only, it just black out If I try to increase any SPI clock to 80Mhz. 
Anyway, LGFX is the most efficiency library which I have used, and it will be better if the touch screen function integrated.

moononournation, 2020-07-27:
Please be noted that, only SPI native pins can run 80 MHz, other pins only can run up to 40 MHz

lovyan03, 2020-07-27:
@Dimision 
The upper speed limit varies depending on the state of the wire connection. It is true that 80MHz is not easy to use, but products like TTGO T-Watch can use 80MHz stably.

I have plans to support a touch screen, but we have multiple types and methods of touch controllers, and it's a bit of a headache.

@moononournation 
Thank you for your comment. But that information is not accurate.
1: The reception speed is limited when going through the GPIO matrix. However, transmission only is unaffected.

2: There is a bug in the SPI master driver for ESP-IDF3.3 (and the ArduinoESP32 that uses it) when using DMA at 80 MHz, the error checking mechanism prevents it.
However, the LovyanGFX is unaffected by this, as it works directly on the peripheral registers.

I use the TTGO-T-Watch, running the DMA at 80 MHz.

Dimision, 2020-07-28:
Hi @lovyan03 
I use some cheap jumper wires to connect ESP32 and ST7789 panel, It's definitely not a good choice, So I think that might be the reason why I cannot reach 80Mhz. I'm going to make a homemade breakboard later and try to increase the clock speed.
As far as i know, most of TFT modules are using XPT2046 to drive the resistive touch screen, and that could be a good start.

Hi @moononournation 
Thanks for reminding, I mentioned my configuration codes at the beginning of this topic, I was actually used the Hardware SPI interface(VSPI), After talking with Lovyan, I realized that the messy wiring could be the main reason.

lovyan03, 2020-08-31:
@Dimision 
Thank you for your patience.
LovyanGFX is now touch screen compatible ! (in the master branch. )
The release will happen in the next few days.
Currently available are XPT2046 and STMPE610 (SPI) , and FT5x06 (I2C).

Dimision, 2020-09-02:
Hi @lovyan03 
Thanks for your reply. After updated my local library, and uploaded some testing codes to the ESP32 board, the touch screen which on my ILI9341 panel finally got worked flawlessly. The library is pretty awesome!

By the way, is there any plan to add more demo examples to the library to make the beginners easier to understand and use it, just like SparkFun and Adafruit does?

lovyan03, 2020-09-02:
I'm glad that your touch screen worked properly !

You are right about the inadequate documentation and examples...

I'd like to add more better example.
What examples do you think I should add?

lovyan03, 2020-09-02:
@Dimision 
I'm glad that your touch screen worked properly !

You are right about the inadequate documentation and examples...

I'd like to add more better example.
What examples do you think I should add?

closed 2021-01-09 by lovyan03

Nadeemranm, 2023-07-26:
In which version of lovyanGFX library deceleration is available  of the ILI9486 LCD DISPLAY  for esp-idf?