lovyan03/LovyanGFX #35: TFT SPI 4 inch (ST7796) support

BlynkGO, 2020-08-14:
Request for TFT SPI 4.0inch (ST7796) too

https://www.aliexpress.com/item/4000572980706.html

Thank you very much.

TridentTD, 2020-08-15:
I have tried to edit the panel for ST7796. As the following is worked for me.

```cpp
//-----------------------------------
//  [Panel_ST7796.hpp]
//-----------------------------------
#ifndef LGFX_PANEL_ST7796_HPP_
#define LGFX_PANEL_ST7796_HPP_

#include "PanelIlitekCommon.hpp"

namespace lgfx
{
  struct Panel_ST7796 : public PanelIlitekCommon
  {
    Panel_ST7796(void)
    {
      panel_width  = memory_width  = 320;
      panel_height = memory_height = 480;

      freq_write = 40000000;
      freq_read  = 20000000;
      freq_fill  = 40000000;
      read_depth  = rgb888_3Byte;
      write_depth = rgb888_3Byte;

      spi_mode = 0;
      spi_mode_read = 0;
      len_dummy_read_pixel = 8;
      spi_read = true;
      spi_3wire = false;

      pwm_ch_bl       = 0;
      backlight_level = true;
      invert    = false;
      rgb_order = false;
      offset_x = 0;
      offset_y = 0;
      rotation = 3;
      offset_rotation = 0;
    }

  protected:

    struct CMD : public CommandCommon
    {
      static constexpr std::uint8_t FRMCTR1 = 0xB1;
      static constexpr std::uint8_t FRMCTR2 = 0xB2;
      static constexpr std::uint8_t FRMCTR3 = 0xB3;
      static constexpr std::uint8_t INVCTR  = 0xB4;
      static constexpr std::uint8_t DFUNCTR = 0xB6;
      static constexpr std::uint8_t ETMOD   = 0xB7;
      static constexpr std::uint8_t PWCTR1  = 0xC0;
      static constexpr std::uint8_t PWCTR2  = 0xC1;
      static constexpr std::uint8_t PWCTR3  = 0xC2;
      static constexpr std::uint8_t PWCTR4  = 0xC3;
      static constexpr std::uint8_t PWCTR5  = 0xC4;
      static constexpr std::uint8_t VMCTR   = 0xC5;
      static constexpr std::uint8_t GMCTRP1 = 0xE0; // Positive Gamma Correction
      static constexpr std::uint8_t GMCTRN1 = 0xE1; // Negative Gamma Correction
      static constexpr std::uint8_t DOCA    = 0xE8; // Display Output Ctrl Adjust 
      static constexpr std::uint8_t CSCON   = 0xF0; // Command Set Control
    };

    const std::uint8_t* getInitCommands(std::uint8_t listno) const override {
      static constexpr std::uint8_t list0[] = {
          CMD::SWRESET, CMD_INIT_DELAY, 120, 
          CMD::SLPOUT,  CMD_INIT_DELAY, 120, 
          CMD::CSCON,   1, 0xC3,  // Enable extension command 2 partI
          CMD::CSCON,   1, 0x96,  // Enable extension command 2 partII
          CommandCommon::MADCTL, 1, 0x48, //X-Mirror, Top-Left to right-Buttom, RGB 
          CommandCommon::COLMOD, 1, 0x55, //Control interface color format set to 16
          CMD::INVCTR,  1, 0x01,  //1-dot inversion
          CMD::DFUNCTR, 3, 0x80,  //Display Function Control //Bypass
                           0x02,  //Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2
                           0x3B,  //LCD Drive Line=8*(59+1)
          CMD::DOCA,    8, 0x40,
                           0x8A,
                           0x00,
                           0x00,
                           0x29,  //Source eqaulizing period time= 22.5 us
                           0x19,  //Timing for "Gate start"=25 (Tclk)
                           0xA5,  //Timing for "Gate End"=37 (Tclk), Gate driver EQ function ON
                           0x33,
          CMD::PWCTR2,  1, 0x06,  //Power control2   //VAP(GVDD)=3.85+( vcom+vcom offset), VAN(GVCL)=-3.85+( vcom+vcom offset)
          CMD::PWCTR3,  1, 0xA7,  //Power control 3  //Source driving current level=low, Gamma driving current level=High
          CMD::VMCTR,   1+CMD_INIT_DELAY, 0x18, 120, //VCOM Control    //VCOM=0.9
          CMD::GMCTRP1,14, 0xF0, 0x09, 0x0B, 0x06, 0x04, 0x15, 0x2F,
                           0x54, 0x42, 0x3C, 0x17, 0x14, 0x18, 0x1B,
          CMD::GMCTRN1,14+CMD_INIT_DELAY, 
                           0xE0, 0x09, 0x0B, 0x06, 0x04, 0x03, 0x2B,
                           0x43, 0x42, 0x3B, 0x16, 0x14, 0x17, 0x1B,
                           120,
          CMD::CSCON,   1, 0x3C, //Command Set control // Disable extension command 2 partI
          CMD::CSCON,   1, 0x69, //Command Set control // Disable extension command 2 partII
          0xFF,0xFF, // end
      };
      static constexpr std::uint8_t list1[] = {
          CMD::SLPOUT,  CMD_INIT_DELAY, 120,    // Exit sleep mode
          CMD::DISPON,  CMD_INIT_DELAY,  20,    // Set display on
          0xFF,0xFF, // end
      };
      switch (listno) {
      case 0: return list0;
      case 1: return list1;
      default: return nullptr;
      }
    }
    std::uint8_t getMadCtl(std::uint8_t r) const override {
      static constexpr std::uint8_t madctl_table[] = {
               MAD_MX|MAD_MH              ,
        MAD_MV                            ,
                             MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MY|MAD_MH|MAD_ML,
               MAD_MX|MAD_MH|MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MH              ,
                                         0,
        MAD_MV|              MAD_MY|MAD_ML,
      };
      r = ((r + offset_rotation) & 3) | ((r & 4) ^ (offset_rotation & 4));
      return madctl_table[r];
    }

    std::uint8_t getColMod(std::uint8_t) const override { return RGB666_3BYTE; }
  };
}
#endif

```

If you have any points for the better for cleaning up , 
do you mind to suggestion ?

BlynkGO, 2020-08-15:
I have tried to edit the panel for ST7796. As the following is worked for me.

```cpp
//-----------------------------------
//  [Panel_ST7796.hpp]
//-----------------------------------
#ifndef LGFX_PANEL_ST7796_HPP_
#define LGFX_PANEL_ST7796_HPP_

#include "PanelIlitekCommon.hpp"

namespace lgfx
{
  struct Panel_ST7796 : public PanelIlitekCommon
  {
    Panel_ST7796(void)
    {
      panel_width  = memory_width  = 320;
      panel_height = memory_height = 480;

      freq_write = 40000000;
      freq_read  = 20000000;
      freq_fill  = 40000000;
      read_depth  = rgb888_3Byte;
      write_depth = rgb888_3Byte;

      spi_mode = 0;
      spi_mode_read = 0;
      len_dummy_read_pixel = 8;
      spi_read = true;
      spi_3wire = false;

      pwm_ch_bl       = 0;
      backlight_level = true;
      invert    = false;
      rgb_order = false;
      offset_x = 0;
      offset_y = 0;
      rotation = 3;
      offset_rotation = 0;
    }

  protected:

    struct CMD : public CommandCommon
    {
      static constexpr std::uint8_t FRMCTR1 = 0xB1;
      static constexpr std::uint8_t FRMCTR2 = 0xB2;
      static constexpr std::uint8_t FRMCTR3 = 0xB3;
      static constexpr std::uint8_t INVCTR  = 0xB4;
      static constexpr std::uint8_t DFUNCTR = 0xB6;
      static constexpr std::uint8_t ETMOD   = 0xB7;
      static constexpr std::uint8_t PWCTR1  = 0xC0;
      static constexpr std::uint8_t PWCTR2  = 0xC1;
      static constexpr std::uint8_t PWCTR3  = 0xC2;
      static constexpr std::uint8_t PWCTR4  = 0xC3;
      static constexpr std::uint8_t PWCTR5  = 0xC4;
      static constexpr std::uint8_t VMCTR   = 0xC5;
      static constexpr std::uint8_t GMCTRP1 = 0xE0; // Positive Gamma Correction
      static constexpr std::uint8_t GMCTRN1 = 0xE1; // Negative Gamma Correction
      static constexpr std::uint8_t DOCA    = 0xE8; // Display Output Ctrl Adjust 
      static constexpr std::uint8_t CSCON   = 0xF0; // Command Set Control
    };

    const std::uint8_t* getInitCommands(std::uint8_t listno) const override {
      static constexpr std::uint8_t list0[] = {
          CMD::SWRESET, CMD_INIT_DELAY, 120, 
          CMD::SLPOUT,  CMD_INIT_DELAY, 120, 
          CMD::CSCON,   1, 0xC3,  // Enable extension command 2 partI
          CMD::CSCON,   1, 0x96,  // Enable extension command 2 partII
          CommandCommon::MADCTL, 1, 0x48, //X-Mirror, Top-Left to right-Buttom, RGB 
          CommandCommon::COLMOD, 1, 0x55, //Control interface color format set to 16
          CMD::INVCTR,  1, 0x01,  //1-dot inversion
          CMD::DFUNCTR, 3, 0x80,  //Display Function Control //Bypass
                           0x02,  //Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2
                           0x3B,  //LCD Drive Line=8*(59+1)
          CMD::DOCA,    8, 0x40,
                           0x8A,
                           0x00,
                           0x00,
                           0x29,  //Source eqaulizing period time= 22.5 us
                           0x19,  //Timing for "Gate start"=25 (Tclk)
                           0xA5,  //Timing for "Gate End"=37 (Tclk), Gate driver EQ function ON
                           0x33,
          CMD::PWCTR2,  1, 0x06,  //Power control2   //VAP(GVDD)=3.85+( vcom+vcom offset), VAN(GVCL)=-3.85+( vcom+vcom offset)
          CMD::PWCTR3,  1, 0xA7,  //Power control 3  //Source driving current level=low, Gamma driving current level=High
          CMD::VMCTR,   1+CMD_INIT_DELAY, 0x18, 120, //VCOM Control    //VCOM=0.9
          CMD::GMCTRP1,14, 0xF0, 0x09, 0x0B, 0x06, 0x04, 0x15, 0x2F,
                           0x54, 0x42, 0x3C, 0x17, 0x14, 0x18, 0x1B,
          CMD::GMCTRN1,14+CMD_INIT_DELAY, 
                           0xE0, 0x09, 0x0B, 0x06, 0x04, 0x03, 0x2B,
                           0x43, 0x42, 0x3B, 0x16, 0x14, 0x17, 0x1B,
                           120,
          CMD::CSCON,   1, 0x3C, //Command Set control // Disable extension command 2 partI
          CMD::CSCON,   1, 0x69, //Command Set control // Disable extension command 2 partII
          0xFF,0xFF, // end
      };
      static constexpr std::uint8_t list1[] = {
          CMD::SLPOUT,  CMD_INIT_DELAY, 120,    // Exit sleep mode
          CMD::DISPON,  CMD_INIT_DELAY,  20,    // Set display on
          0xFF,0xFF, // end
      };
      switch (listno) {
      case 0: return list0;
      case 1: return list1;
      default: return nullptr;
      }
    }
    std::uint8_t getMadCtl(std::uint8_t r) const override {
      static constexpr std::uint8_t madctl_table[] = {
               MAD_MX|MAD_MH              ,
        MAD_MV                            ,
                             MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MY|MAD_MH|MAD_ML,
               MAD_MX|MAD_MH|MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MH              ,
                                         0,
        MAD_MV|              MAD_MY|MAD_ML,
      };
      r = ((r + offset_rotation) & 3) | ((r & 4) ^ (offset_rotation & 4));
      return madctl_table[r];
    }

    std::uint8_t getColMod(std::uint8_t) const override { return RGB666_3BYTE; }
  };
}
#endif

```

If you have any points for the better for cleaning up , 
do you mind for suggestion ?

Thank you.

lovyan03, 2020-08-15:
@BlynkGO 
Thank you for your work !

I ordered the ST7796 today and it seems to take about two weeks to arrive.
It will be a little while before I get to try that, but I will consider this suggestion as soon as possible.

I'm reading the specsheet now.
I want to check a few things before I merge it.

Confirmation 1:
Need to override "getColMod" ?
If the ST7796 supports 16-bit color mode, do not need to override getColMod.
Remove the override and use setColorDepth(16) and try to draw something.
If the drawing seems strange, need to override "getColMod".

Confirmation 2:
Does the read pixel sample work correctly?
"examples/Test/test_readpixel/test_readpixel.ino"
If it works correctly, the test pattern will move as it deforms.
If the settings are not correct, the colors will be wrong or collapsed.
In that case, you may need to adjust "len_dummy_read_pixel" and "freq_read".

The following settings are described in user code, not in the panel structure.
```c
      spi_read = true;
      spi_3wire = false;
      pwm_ch_bl       = 0;
      backlight_level = true;
      invert    = false;
      rgb_order = false;
      offset_x = 0;
      offset_y = 0;
      rotation = 3;
      offset_rotation = 0;
```

BlynkGO, 2020-08-15:
@lovyan03 
Thank you for the suggestion.

By your suggestion,
I have just removed the user-panel structure and `getColMod(..)` as the following

```cpp
#ifndef LGFX_PANEL_ST7796_HPP_
#define LGFX_PANEL_ST7796_HPP_

#include "PanelIlitekCommon.hpp"

namespace lgfx
{
  struct Panel_ST7796 : public PanelIlitekCommon
  {
    Panel_ST7796(void)
    {
      panel_width  = memory_width  = 320;
      panel_height = memory_height = 480;

      freq_write = 40000000;
      freq_read  = 20000000;
      freq_fill  = 40000000;
      read_depth  = rgb888_3Byte;
      write_depth = rgb888_3Byte;
    }

  protected:

    struct CMD : public CommandCommon
    {
      static constexpr std::uint8_t FRMCTR1 = 0xB1;
      static constexpr std::uint8_t FRMCTR2 = 0xB2;
      static constexpr std::uint8_t FRMCTR3 = 0xB3;
      static constexpr std::uint8_t INVCTR  = 0xB4;
      static constexpr std::uint8_t DFUNCTR = 0xB6;
      static constexpr std::uint8_t ETMOD   = 0xB7;
      static constexpr std::uint8_t PWCTR1  = 0xC0;
      static constexpr std::uint8_t PWCTR2  = 0xC1;
      static constexpr std::uint8_t PWCTR3  = 0xC2;
      static constexpr std::uint8_t PWCTR4  = 0xC3;
      static constexpr std::uint8_t PWCTR5  = 0xC4;
      static constexpr std::uint8_t VMCTR   = 0xC5;
      static constexpr std::uint8_t GMCTRP1 = 0xE0; // Positive Gamma Correction
      static constexpr std::uint8_t GMCTRN1 = 0xE1; // Negative Gamma Correction
      static constexpr std::uint8_t DOCA    = 0xE8; // Display Output Ctrl Adjust 
      static constexpr std::uint8_t CSCON   = 0xF0; // Command Set Control
    };

    const std::uint8_t* getInitCommands(std::uint8_t listno) const override {
      static constexpr std::uint8_t list0[] = {
          CMD::SWRESET, CMD_INIT_DELAY, 120, 
          CMD::SLPOUT,  CMD_INIT_DELAY, 120, 
          CMD::CSCON,   1, 0xC3,  // Enable extension command 2 partI
          CMD::CSCON,   1, 0x96,  // Enable extension command 2 partII
          CommandCommon::MADCTL, 1, 0x48, //X-Mirror, Top-Left to right-Buttom, RGB 
          CommandCommon::COLMOD, 1, 0x55, //Control interface color format set to 16
          CMD::INVCTR,  1, 0x01,  //1-dot inversion
          CMD::DFUNCTR, 3, 0x80,  //Display Function Control //Bypass
                           0x02,  //Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2
                           0x3B,  //LCD Drive Line=8*(59+1)
          CMD::DOCA,    8, 0x40,
                           0x8A,
                           0x00,
                           0x00,
                           0x29,  //Source eqaulizing period time= 22.5 us
                           0x19,  //Timing for "Gate start"=25 (Tclk)
                           0xA5,  //Timing for "Gate End"=37 (Tclk), Gate driver EQ function ON
                           0x33,
          CMD::PWCTR2,  1, 0x06,  //Power control2   //VAP(GVDD)=3.85+( vcom+vcom offset), VAN(GVCL)=-3.85+( vcom+vcom offset)
          CMD::PWCTR3,  1, 0xA7,  //Power control 3  //Source driving current level=low, Gamma driving current level=High
          CMD::VMCTR,   1+CMD_INIT_DELAY, 0x18, 120, //VCOM Control    //VCOM=0.9
          CMD::GMCTRP1,14, 0xF0, 0x09, 0x0B, 0x06, 0x04, 0x15, 0x2F,
                           0x54, 0x42, 0x3C, 0x17, 0x14, 0x18, 0x1B,
          CMD::GMCTRN1,14+CMD_INIT_DELAY, 
                           0xE0, 0x09, 0x0B, 0x06, 0x04, 0x03, 0x2B,
                           0x43, 0x42, 0x3B, 0x16, 0x14, 0x17, 0x1B,
                           120,
          CMD::CSCON,   1, 0x3C, //Command Set control // Disable extension command 2 partI
          CMD::CSCON,   1, 0x69, //Command Set control // Disable extension command 2 partII
          0xFF,0xFF, // end
      };
      static constexpr std::uint8_t list1[] = {
          CMD::SLPOUT,  CMD_INIT_DELAY, 120,    // Exit sleep mode
          CMD::DISPON,  CMD_INIT_DELAY,  20,    // Set display on
          0xFF,0xFF, // end
      };
      switch (listno) {
      case 0: return list0;
      case 1: return list1;
      default: return nullptr;
      }
    }
    std::uint8_t getMadCtl(std::uint8_t r) const override {
      static constexpr std::uint8_t madctl_table[] = {
               MAD_MX|MAD_MH              ,
        MAD_MV                            ,
                             MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MY|MAD_MH|MAD_ML,
               MAD_MX|MAD_MH|MAD_MY|MAD_ML,
        MAD_MV|MAD_MX|MAD_MH              ,
                                         0,
        MAD_MV|              MAD_MY|MAD_ML,
      };
      r = ((r + offset_rotation) & 3) | ((r & 4) ^ (offset_rotation & 4));
      return madctl_table[r];
    }
  };
}

#endif
```

And then tested by the example `examples/Test/test_readpixel/test_readpixel.ino`.
The result as the following

[image]

Does the result display correctly?

By the way, do you mind to add a API for returning `spi_t*` function to the library?

such as
```
__attribute__ ((always_inline)) inline spi_t* getSPIHandle(void) const { return _spi_handle; }
```

I would like to use the same spi_handle  for the touch interface.

Thank you.

lovyan03, 2020-08-15:
@BlynkGO 
thankyou !
Yes, that's correct for the test_readpixel display, but it should move down from there, transforming into a triangle.
Make sure the color does not change or darken at that time.

Please be patient, I have plans to support touchscreens.
getSPIHandle needs to be decided carefully as exposing it will interfere with ESP-IDF compatibility, but probabry you can simply use the SPI class.

BlynkGO, 2020-08-15:
@lovyan03  
I tried by the Arduino's SPI class for the touch interface last night.

When after call  `SPI.begin(....);`,  the touch interface can use, 
however the spi from LovyanGFX is stop. can't display other thing anymore.

I recreated by using the LovyanGFX's spi_handle,
it's fine for both TFT and touch interface.

Waiting for LovyanGFX come with touchscreens.
Thank you very much.

lovyan03, 2020-08-15:
@BlynkGO 
Perhaps you need to use SPIClass beginTransaction(); and endTransaction(); function.
The same goes for startWrite() and endWrite() of LovyanGFX.
These must be called properly to use the SPI only when necessary.

closed 2021-01-27 by lovyan03