lovyan03/LovyanGFX #36: How to wire ESP32 I2S mode to TFT ?

BlynkGO, 2020-08-16:
I would like to try TFT parallel i2s mode.

Is there any ESP32 I2S + TFT wire-mapping 
for trying LovyanGFX with parellel I2S mode ?

Thank you very much.

lovyan03, 2020-08-16:
@BlynkGO 
Sorry.
I haven't created that document yet, but I've included it as a comment on the 2_spi_setting example.

examples/HowToUse/2_spi_setting
```c
/* for ESP32 Parallel
struct LGFX_Config
{
  // Select the type of I2S  (I2S_NUM_0 or I2S_NUM_1)
  // default value is I2S_NUM_0
  static constexpr i2s_port_t i2s_port = I2S_NUM_0;

  // Set the parallel pin number
  // パラレル接続の各種ピン番号を設定します。
  static constexpr int gpio_wr  =  4;
  static constexpr int gpio_rd  =  2;
  static constexpr int gpio_rs  = 15;
  static constexpr int gpio_d0  = 12;
  static constexpr int gpio_d1  = 13;
  static constexpr int gpio_d2  = 26;
  static constexpr int gpio_d3  = 25;
  static constexpr int gpio_d4  = 17;
  static constexpr int gpio_d5  = 16;
  static constexpr int gpio_d6  = 27;
  static constexpr int gpio_d7  = 14;
};
```
```c
static lgfx::LGFX_PARALLEL<LGFX_Config> lcd;
```
```c
  panel.spi_cs = 14;
```

I have added "gpio_" to the name of the pin, but there is a possibility of changing it to "pin_".
Also, it is possible that ”gpio_rs" will be abolished and "panel.spi_dc" will be used.
I have a lot to think about. This may change when I do major upgrades.

BlynkGO, 2020-08-16:
@lovyan03 
I will try!
どうもありがとうございました。

closed 2021-01-27 by lovyan03