lovyan03/LovyanGFX #46: Incorrect character alignment at drawChar

yamamaya, 2020-09-24:
Those characters are not aligned correctly.

canvas.setTextFont( &fonts::FreeMonoBold18pt7b );
canvas.setTextDatum( textdatum_t::top_left );
canvas.drawChar( '0', 50, 2 );
canvas.setTextFont( &fonts::Font4 );
canvas.drawChar( '0', 70, 2 );

lovyan03, 2020-09-24:
@yamamaya 
Thank you for reporting this issue !

The following functions are affected by textdatum.
・Functions of the "drawString" family.
・Functions of the "print" family. (Vertical direction only)
It has no effect on drawChar.

And the GFX font is based on a baseline of reference coordinates. This is based on the specifications of AdafruitGFX and is inherited by LovyanGFX.

https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts

I recommend that you use "print" or "drawString" even if you only want to output a single character.

yamamaya, 2020-09-24:
Thank you for your advice!

closed 2020-09-24 by yamamaya