lovyan03/LovyanGFX #52: Request for RA8875 800x480 support

BlynkGO, 2020-10-05:
Request for TFT SPI 5.0inch (RA8875) SPI4pin  too.

https://www.buydisplay.com/5-inch-tft-lcd-display-capacitive-touchscreen-ra8875-controller-800x480

Thank you very much.

lovyan03, 2020-10-05:
@BlynkGO 
Thanks for the request.
I've looked into it and it looks like it's a very different specs from the traditional panels and seems to be a high performance panel.
I'm not sure if I can achieve sufficient support, but I've ordered it anyway.

BlynkGO, 2020-10-06:
Thank you.

楽しみにしています!

lovyan03, 2020-10-31:
The shop contacted me and it seems that somehow the package was returned to the shop.
I was informed that it will be shipped again. It will still take a while for it to arrive.
(´;ω;`)

yashm, 2021-09-21:
+1 for this!
Any updates for the RA8875?
Great Library on nonRA8875 setups!

lovyan03, 2022-03-25:
@BlynkGO @yashm 
Thank you for your patience.
I have updated the develop branch and added RA8875 support.

Tests of operation with SPI connections have been completed, but parallel connections have not yet been tested.

DanielFranco227, 2022-06-10:
Hello. I am stucked trying to test operation of RA8875. Do you have any existing src code already tested and setup if possible to share with me? I am developing it with ESP32 . I would appreciate it.

lovyan03, 2022-06-10:
@DanielFranco227 
Perhaps this configuration will work.
If it does not work, please tell us a little more about your situation.

btw, Parallel connection operation of the RA8875 has not yet been implemented.

```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_RA8875 _panel_instance;
  lgfx::Bus_SPI _bus_instance;

public:

  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();

      cfg.spi_host = SPI2_HOST;
      cfg.freq_write = 40000000;
      cfg.freq_read  =  5000000;
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 23;
      cfg.pin_miso = 19;
      cfg.spi_3wire = false;

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();

      cfg.pin_cs = 14;
      cfg.pin_rst = 33;
      cfg.pin_busy = 27;

      cfg.dummy_read_pixel = 16;
      cfg.dummy_read_bits  = 0;
      cfg.readable = true;

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance);
  }
};
```

DanielFranco227, 2022-06-11:
Hello! Thank you for your quick reply.
I tried the configuration you just attached above but the problem continues, I can't see anything on my screen and I noticed a weird behavior.

Hardware used:
ESP32-WROOM-32U
[RA8875 Adafruit driver board](https://www.adafruit.com/product/1590)
[7 inch Adafruit TFT display](https://www.adafruit.com/product/2354)

Software used: ESP-IDF V4.4 with VScode

src code:

```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

extern "C"
{
  void app_main();
}

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_RA8875 _panel_instance;
  lgfx::Bus_SPI _bus_instance;

public:

  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();

      cfg.spi_host = VSPI_HOST;
      cfg.freq_write = 40000000;
      cfg.freq_read  =  5000000;
      cfg.pin_sclk = 4;
      cfg.pin_mosi = 16;
      cfg.pin_miso = 17;
      cfg.spi_3wire = false;

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();

      cfg.pin_cs = 5;
      cfg.pin_rst = 18;
      cfg.pin_busy = 34;

      cfg.dummy_read_pixel = 16;
      cfg.dummy_read_bits  = 0;
      cfg.readable = true;

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance);
  }
};

//THEN THE EXAMPLE FROM (https://github.com/lovyan03/LovyanGFX/blob/master/examples/Standard/Graph/Graph.ino)

static void guiTask(void *pvParameter);

unsigned long IRAM_ATTR millis_custom()
{
  return (unsigned long)(esp_timer_get_time() / 1000ULL);
}
#include <vector>
#define LINE_COUNT 6

LGFX lcd;
static std::vector<int> points[LINE_COUNT];
static int colors[] = { TFT_RED, TFT_GREEN, TFT_BLUE, TFT_CYAN, TFT_MAGENTA, TFT_YELLOW };
static int xoffset, yoffset, point_count;
int getBaseColor(int x, int y)
{
  return ((x^y)&3 || ((x-xoffset)&31 && y&31) ? TFT_BLACK : ((!y || x == xoffset) ? TFT_WHITE : TFT_DARKGREEN));
}

void app_main()
{

  lcd.init();

  if (lcd.width() < lcd.height()) lcd.setRotation(lcd.getRotation() ^ 1);

  yoffset = lcd.height() >> 1;
  xoffset = lcd.width()  >> 1;
  point_count = lcd.width() + 1;

  for (int i = 0; i < LINE_COUNT; i++)
  {
    points[i].resize(point_count);
  }

  lcd.startWrite();
  lcd.setAddrWindow(0, 0, lcd.width(), lcd.height());
  for (int y = 0; y < lcd.height(); y++)
  {
    for (int x = 0; x < lcd.width(); x++)
    {
      lcd.writeColor(getBaseColor(x, y - yoffset), 1);
    }
  }
  lcd.endWrite();
  xTaskCreatePinnedToCore(guiTask, "gui", 4096 * 2, NULL, 0, NULL, 1);
  vTaskDelete(NULL);

}

static void guiTask(void *pvParameter)
{
  for (;;)
  {
  static int prev_sec;
  static int fps;
  ++fps;
  int sec = millis_custom() / 1000;
  if (prev_sec != sec)
  {
    prev_sec = sec;
    lcd.setCursor(0,0);
    lcd.printf("fps:%03d", fps);
    fps = 0;
  }

  static int count;

  for (int i = 0; i < LINE_COUNT; i++)
  {
    points[i][count % point_count] = (sinf((float)count / (10 + 30 * i))+sinf((float)count / (13 + 37 * i))) * (lcd.height() >> 2);
  }

  ++count;

  lcd.startWrite();
  int index1 = count % point_count;
  for (int x = 0; x < point_count-1; x++)
  {
    int index0 = index1;
    index1 = (index0 +1) % point_count;
    for (int i = 0; i < LINE_COUNT; i++)
    {
      int y = points[i][index0];
      if (y != points[i][index1])
      {
        lcd.writePixel(x, y + yoffset, getBaseColor(x, y));
      }
    }

    for (int i = 0; i < LINE_COUNT; i++)
    {
      int y = points[i][index1];
      lcd.writePixel(x, y + yoffset, colors[i]);
    }
  }
  lcd.endWrite();
  }

}

```

Note before results:
*I need to turn ON the GPO pin of the RA8875 so I added the following lines to Panel_RA8875.cpp:
[image]
*The backlight of the screen is forced to be fully ON and it does not depend on the driver

When running after 12 sec the WDT is triggered and I get the following backtrace:
[image]
[image]

Notes after the backtrace:
*The GPO pin of the RA8875 was never turned ON, it keeps as 0 state(default after reset)

*Line 153 of main.cpp is: lcd.init();

*The timestamp of  the last 2 green lines are 1 second apart as you can see and it is stucked inside "_wait_busy()" because the timeout is 1 sec and the reading of the wait pin is 0 as if it was always busy.

*The communication lines are properly connected but I don't know if it caused by the ~WAIT pin or the speed of the bus. What is the minimum speed that I can test it with? and where do you recommend me to change it? I have tested raw commands with only the SPI library and it communicates successfully at 1.25 MHz. I have not tried higher speeds

*I tested the program on a screen with TFT driver ILI9488 and I can see the program running with no issues (obviously the configuration is not the same  and there are some differences like the DC pin and other things but this tells me that the example provided is not the issue...)

I would appreciate your help and time to help me solving this issue.

DanielFranco227, 2022-06-11:
Update: I tried with the pinout config you provided above and it is working now. Thank you so much for your help

DanielFranco227, 2022-06-11:
Update: Testing GPIO 34 as cfg.pin_busy crashes the program.

> Update: I tried with the pinout config you provided above and it is working now. Thank you so much for your help

lovyan03, 2022-06-11:
@DanielFranco227 
Thanks for the information.
We have updated the develop branch as we found that the pin mode setting for the busy pins was not implemented during initialization.

DanielFranco227, 2022-06-23:
Hello.
I've been using lgvl v8 but during transition of page, animations are really slow. Do you know any tips to improve performance? or do you have any example that is working with no issues? Thanks in advance

lovyan03, 2022-06-23:
@DanielFranco227 
Page transitions mean that you are rewriting the entire screen.
If SPI 40MHz is set up, the limit is about 6.5FPS.
The amount of communication goes like this.

```
amount of pixel data : 800 x 480 x 16bpp = 6,144,000 bit 
SPI speed : 40,000,000 Hz

Time required for transmission : 6,144,000 / 40,000,000 = 0.1536 = 153.6 msec
Frame rate : 40,000,000 / 6,144,000 = 6.51 FPS
```

Please quantify how slow is "really slow."

DanielFranco227, 2022-06-24:
Video of the screen running with a test program:
[https://youtube.com/shorts/4HiK7YT8tL4](url)

Video of the test program in squareline simulator:
[https://youtu.be/tGtuNzJfM3I](url)

You can see the spinner running when it is idle but when you change page or select an object, it takes more than 1 sec to show next screen values

lovyan03, 2022-06-27:
Frankly, this topic is not about LovyanGFX.
I don't want to go into details here because you are asking me to show you how to use LVGL.

I think that's about the speed you get when you run LVGL on a microcontroller without setting it up properly.

At the very least, it will never run at the same speed as a PC simulator.
Such speeds are not possible, so do not expect too much.

Check out this series of tweets and give it a try.

https://twitter.com/lovyan03/status/1504298966577725441
https://twitter.com/lovyan03/status/1504298970574893058
https://twitter.com/lovyan03/status/1504298974874042368
https://twitter.com/lovyan03/status/1504298978950938626
https://twitter.com/lovyan03/status/1504324849174720513

BlynkGO, 2022-07-04:
Hi, great news for RA8875 supported.

By the way, what is the pin_busy?

[image]

The pin_busy is the WAIT pin (pin 9) of the TFT?

[image]

Thank you.

lovyan03, 2022-07-04:
@BlynkGO 
yes. It's busy state pin.

BlynkGO, 2022-07-04:
> @BlynkGO yes. It's busy state pin.

Thank you very much.

DanielFranco227, 2022-07-04:
> Frankly, this topic is not about LovyanGFX. I don't want to go into details here because you are asking me to show you how to use LVGL.
>
> I think that's about the speed you get when you run LVGL on a [...]

Thank you very much for your time and recommendations.

I tried to compare 2 different tft drivers with the same program without using lvgl. I used one of your test programs and I noticed a huge difference in the fps. 
Panel_RA8875: 6 FPS
Panel_ILI9488: 30 FPS

Is the same behavior on your side too? I would appreciate any help you can provide me. 

The code is the following:
```

LGFX display;

unsigned long IRAM_ATTR millis_custom()
{
  return (unsigned long)(esp_timer_get_time() / 1000ULL);
}
#include <vector>
#define LINE_COUNT 6

// LGFX lcd;
#define lcd display
static std::vector<int> points[LINE_COUNT];
static int colors[] = { TFT_RED, TFT_GREEN, TFT_BLUE, TFT_CYAN, TFT_MAGENTA, TFT_YELLOW };
static int xoffset, yoffset, point_count;
int getBaseColor(int x, int y)
{
  return ((x^y)&3 || ((x-xoffset)&31 && y&31) ? TFT_BLACK : ((!y || x == xoffset) ? TFT_WHITE : TFT_DARKGREEN));
}

void app_main()
{
  display.init();

  if (lcd.width() < lcd.height()) lcd.setRotation(lcd.getRotation() ^ 1);

  yoffset = lcd.height() >> 1;
  xoffset = lcd.width()  >> 1;
  point_count = lcd.width() + 1;

  for (int i = 0; i < LINE_COUNT; i++)
  {
    points[i].resize(point_count);
  }

  lcd.startWrite();
  lcd.setAddrWindow(0, 0, lcd.width(), lcd.height());
  for (int y = 0; y < lcd.height(); y++)
  {
    for (int x = 0; x < lcd.width(); x++)
    {
      lcd.writeColor(getBaseColor(x, y - yoffset), 1);
    }
  }
  lcd.endWrite();
  xTaskCreatePinnedToCore(guiTask, "gui", 4096 * 2, NULL, 0, NULL, 1);
  vTaskDelete(NULL);
}

static void guiTask(void *pvParameter)
{
  for (;;)
  {
  static int prev_sec;
  static int fps;
  ++fps;
  int sec = millis_custom() / 1000;
  if (prev_sec != sec)
  {
    prev_sec = sec;
    lcd.setCursor(0,0);
    lcd.printf("fps:%03d", fps);
    fps = 0;
  }

  static int count;

  for (int i = 0; i < LINE_COUNT; i++)
  {
    points[i][count % point_count] = (sinf((float)count / (10 + 30 * i))+sinf((float)count / (13 + 37 * i))) * (lcd.height() >> 2);
  }

  ++count;

  lcd.startWrite();
  int index1 = count % point_count;
  for (int x = 0; x < point_count-1; x++)
  {
    int index0 = index1;
    index1 = (index0 +1) % point_count;
    for (int i = 0; i < LINE_COUNT; i++)
    {
      int y = points[i][index0];
      if (y != points[i][index1])
      {
        lcd.writePixel(x, y + yoffset, getBaseColor(x, y));
      }
    }

    for (int i = 0; i < LINE_COUNT; i++)
    {
      int y = points[i][index1];
      lcd.writePixel(x, y + yoffset, colors[i]);
    }
  }
  lcd.endWrite();
  }
}

```

DanielFranco227, 2022-07-04:
> Thank you very much for your time and recommendations.
>
> I tried to compare 2 different tft drivers with the same program without using lvgl. I used one of your test programs and I noticed a huge [...]

Sorry, I respond to myself: It is normal due to the size of the screen and number of pixels. It is exactly the fps expected. Besides any other advice you can give me, if you have the same fps number, it is ok for me.

Thank you in advance

lovyan03, 2022-07-05:
@DanielFranco227 
Yes, as I previously answered to you.

> If SPI 40MHz is set up, the limit is about 6.5FPS.

Any higher speed is theoretically impossible.
Alternatively, if you call setColorDepth(8) and operated in RGB332 mode, you should be able to cut the communication time in half and reducing the number of colors.

Since stable operation at SPI 80MHz will be difficult, the only remaining method is a parallel connection.
However, the LovyanGFX RA8875 parallel connection has not yet been tested.
I would like to eventually support parallel connection operation, but it will be a while before we can do that because the tasks are piling up.

BlynkGO, 2022-07-05:
Hi, the RA8875 TFT can display now,
the next issue, if the TFT's touch interface is FT5206, how to config the touch interface?

[Current Wiring]
CTP I2C SDA ---> ESP32 PIN21
CTP I2C SCL ---> ESP32 PIN22
CTP INTERRUPT ---> ESP32 PIN14

[Current Touch config's code]
```cpp
class LGFX : public lgfx::LGFX_Device {
  private:
    ....
    lgfx::Touch_FT5x06 _touch_instance;  
  public:
    LGFX (void)
    {
      ....
      {
        auto cfg = _touch_instance.config();
        cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
        cfg.x_max      = 799;  // タッチスクリーンから得られる最大のX値(生の値)
        cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
        cfg.y_max      = 479;  // タッチスクリーンから得られる最大のY値(生の値)
        cfg.pin_int    = 14;   // INTが接続されているピン番号
        cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
        cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

  // I2C接続の場合
        cfg.i2c_port =  0; //1;      // 使用するI2Cを選択 (0 or 1)
        cfg.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
        cfg.pin_sda  = 21;     // SDAが接続されているピン番号
        cfg.pin_scl  = 22;     // SCLが接続されているピン番号
        cfg.freq = 400000;     // I2Cクロックを設定
        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
      }
      setPanel(&_panel_instance);
    }
};
 ```
 
 [Result]
 Some error occurs like the following
 ```
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jul 29 2019 12:21:46
 ```
 
 
 How to fix the FT5206 touch interface config?
 Thank you.

BlynkGO, 2022-07-07:
@lovyan03 
Hi, my tft now can display and can touch already.
The next issue  when  I wire TFT's backlight pin to ESP32's GPIO32 with
the following backlight config.

```cpp
class LGFX: public lgfx::LGFX_Device {
  private:
    lgfx::Panel_RA8875  _panel_instance;
     .....
    lgfx::Light_PWM     _light_instance;
  public:
    LGFX(void){
      { 
         ....
      }
      { // バックライト制御の設定を行います。（必要なければ削除）
        auto cfg = _light_instance.config();    // バックライト設定用の構造体を取得します。

        cfg.pin_bl      = 32;                // バックライトが接続されているピン番号
        cfg.invert      = false;             // バックライトの輝度を反転させる場合 true
        cfg.freq        = 44100;           // バックライトのPWM周波数
        cfg.pwm_channel = 15;        // 使用するPWMのチャンネル番号

        _light_instance.config(cfg);
        _panel_instance.setLight(&_light_instance);  // バックライトをパネルにセットします。
      }
#endif
      setPanel(&_panel_instance);
   }
}
```

The result occurs the following.
[image]

How to solve the issue for backlight-PWM.
Thank you.

lovyan03, 2022-07-07:
@BlynkGO 
Is your ESP32 the first generation ESP32 and not the ESP32S2 or S3?
I tried the following code with the first generation ESP32 and the backlight flickers without any problem.

```
void setup()
{
  auto cfg = _pwm.config();
  cfg.pin_bl = 32;
  cfg.invert = false;
  cfg.pwm_channel = 15;
  cfg.freq = 44100;
  _pwm.config(cfg);
  _pwm.init(128);

  for (int i = 0; i < 65535; ++i)
  {
    _pwm.setBrightness(i & 255);
    delay(1);
  }
}
```

BlynkGO, 2022-07-07:
@lovyan03 
Thank you , backlight PWM has already done.

By the way, TFT 5 inch RA8875 800x480 with capacitive GSL1680F touch interface.

I tried by the following code, but it's not worked.

[Wiring ]
TFT's pin 34 ( CTP I2C SDA )  <----> ESP32 GPIO 21 (I2C SDA)
TFT's pin 35 ( CTP I2C SCL )  <----> ESP32 GPIO 22 (I2C SCL)

[Code]

```
class LGFX : public lgfx::LGFX_Device {
  private:
    ....
    lgfx::Touch_GSL1680F_800x480 _touch_instance;
    ....

  public:
    LGFX(void)
    {
      lgfx::Panel_RA8875  _panel_instance;
      .....
      .....
      
      { // タッチスクリーン制御の設定を行います。（必要なければ削除）
        auto cfg = _touch_instance.config();

        cfg.x_min      = 0;
        cfg.x_max      = 799;
        cfg.y_min      = 0;
        cfg.y_max      = 479;
        cfg.pin_int    = -1; //CTP_INTERRUPT;//38;   // INTが接続されているピン番号
        cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
        cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

        cfg.i2c_port = 0;           // 使用するI2Cを選択 (0 or 1)
        cfg.i2c_addr = 0x40;        // I2Cデバイスアドレス番号
        cfg.pin_sda  = 21;          // SDAが接続されているピン番号
        cfg.pin_scl  = 22;          // SCLが接続されているピン番号
        cfg.freq     = 400000;      // I2Cクロックを設定

        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
      }

      setPanel(&_panel_instance);
    }
};
```

How to config GSL1680F touch interface for 
https://www.buydisplay.com/5-inch-tft-lcd-display-capacitive-touchscreen-ra8875-controller-800x480 ?

lovyan03, 2022-07-07:
@BlynkGO 
[image]

If #32 remains LOW, TP is not activated.
Try connecting it to the same pin as the RST on the LCD.

BlynkGO, 2022-07-08:
@lovyan03 

Thank you very much.

dmartauz, 2022-09-07:
> @DanielFranco227 Yes, as I previously answered to you.
>
>
> Any higher speed is theoretically impossible. Alternatively, if you call setColorDepth(8) and operated in RGB332 mode, you should be able to [...]

@lovyan03 any news on RA8875 parallel connection support?

closed 2022-12-25 by github-actions[bot]