lovyan03/LovyanGFX #54: pushImage no longer compiles with a transparency color

m1cr0lab, 2020-10-16:
Hi lovyan03,

First of all, I would like to thank you for your high quality work and the services offered by your LovyanGFX library.
Thank you so much for making it available to all of us.

However, since your last update v0.2.3, I can no longer compile my project when I specify a transparency color with the `pushImage` function :

```cpp
// it doesn't compile anymore when :
// - TRANS_COLOR is defined as a single variable
// - TRANS_COLOR is defined as a global constant
// - TRANS_COLOR is defined as a static constant within a class
sprite.pushImage(x, y, w, h, data, TRANS_COLOR);
```

**While it compiled perfectly with version 0.2.2.**

Here's the error I get :

```
.pio/libdeps/m5/LovyanGFX/src/lgfx/LGFXBase.hpp: In instantiation of 'void lgfx::LGFXBase::pushImage(int32_t, int32_t, int32_t, int32_t, const T*, const U&) [with T = lgfx::rgb565_t; U = short unsigned int; int32_t = int]':
.pio/libdeps/m5/LovyanGFX/src/lgfx/LGFXBase.hpp:265:9:   required from 'void lgfx::LGFXBase::pushImage(int32_t, int32_t, int32_t, int32_t, const uint16_t*, const U&) [with U = short unsigned int; int32_t = int; uint16_t = short unsigned int]'
src/Player.cpp:14:91:   required from here
.pio/libdeps/m5/LovyanGFX/src/lgfx/LGFXBase.hpp:246:90: error: invalid use of non-static member function
       pixelcopy_t p(data, _write_conv.depth, get_depth<T>::value, hasPalette, nullptr, tr);
```

Am I misusing the new version of `pushImage()`?

lovyan03, 2020-10-16:
@m1cr0lab 
Thank you for reporting this issue !
And I'm sorry for the inconvenience.

Perhaps I removed a necessary definition while cleaning up my code.
I'm going to fix it now, so please be patient.

lovyan03, 2020-10-16:
@m1cr0lab 
I have fixed the bug and updated the master branch.
Please give this a try.

m1cr0lab, 2020-10-17:
Thank you so much @lovyan03!
It compiles perfectly now.
Thanks for your responsiveness and your great job.

closed 2020-10-17 by m1cr0lab