lovyan03/LovyanGFX #57: LovyanGFX version awareness

tobozo, 2020-10-30:
Does it make sense to add something like this ?

https://github.com/tobozo/M5Stack-SD-Updater/blob/master/src/gitTagVersion.h

lovyan03, 2020-10-31:
I have adopted and committed to the proposal. 
Thank you!

closed 2020-10-31 by lovyan03

tobozo, 2020-10-31:
great :+1: 

Please note that  the version number must be increased on the master branch *after every release*, not before.

Benefits for developers: 
- no more version confusion between the *latest release* and the *master/develop branches*
- optional ability to add "-develop" or "-master" or other branch name suffixes

For example, just after releasing 0.2.5 the next commit message should be "raising version":

- library.properties -> 0.2.6
- library.json -> 0.2.6
- gitTagVersion.h -> 0.2.6

tobozo, 2020-10-31:
I have automated the process on my repository.

The result is this pull request received after producing the 1.0.2 release: 

https://github.com/tobozo/M5Stack-SD-Updater/pull/151

Let me know if you're interested in using this.

lovyan03, 2020-10-31:
@tobozo 
Thanks!
I'm interested. I would like to consider that approach.

tobozo, 2020-11-01:
You will need to copy those two files into your .github folder, no editing is required, just keep the same path:

- [.github/workflows/onrelease.yml](https://github.com/tobozo/ESP32-Chimera-Core/blob/master/.github/workflows/onrelease.yml)
- [.github/scripts/semver.sh](https://github.com/tobozo/ESP32-Chimera-Core/tree/master/.github/scripts/semver.sh)

After a release is published, this workflow will make a pull request with the modified version files.

This workflow will increase the "patch" version number ( 1.0.1 => 1.0.2 ) by default.

"Major" and "Minor" version numbers can only be increased manually by using "Run workflow" from github action.

It may be wise to try that on a private repository first in order to get familiar with the manual override.

Forcing any version to "1.4.6"
[image]

Default setting (increase "patch" version number level)
[image]

lovyan03, 2020-11-01:
Thanks for the explanation !
I have created the necessary files.
I will make sure it works on the next release.