lovyan03/LovyanGFX #59: sprite.setPsram(true)  + sprite.setColorDepth(16)

tobozo, 2020-11-08:
- Observed on M5Core2 and Wrover-Kit (ILI9341)
- Using this for creating a 320x240 sprite:

```C
    static LGFX_Sprite sprite = LGFX_Sprite( &tft );
```

these work:

```C
    sprite.setColorDepth( 8 );
    sprite.setPsram( true );
```
```C
    sprite.setColorDepth( 24 );
    sprite.setPsram( true );
```

[image]

however these do not seem to produce a valid result

```C
    sprite.setColorDepth( tft.getColorDepth() );
    sprite.setPsram( true );
```
```C
    sprite.setColorDepth( 16 );
    sprite.setPsram( true );
```
[image]

Is that a limitation of the display ?

lovyan03, 2020-11-09:
@tobozo 
Thanks for reporting this!

I found a flaw in the behavior when using PSRAM.
I'm working on fixing it and will be making a release today.

lovyan03, 2020-11-09:
@tobozo 
The fix is complete and ver 0.2.6 has been released.
thanks !

tobozo, 2020-11-09:
fix confirmed, closing this

closed 2020-11-09 by tobozo