lovyan03/LovyanGFX #71: pushRotateZoom() / Is it a specification,my mistake or bug?

GOB52, 2020-12-03:
M5Stack gray, Arduino, LovyanGFX 0.2.7

pushRotateZoom() 
想定範囲を超えて描画が行われているように見えますが、仕様でしょうか?
または私の使い方に誤りがありますか?
It appears to be out of the expected range.
Is it a specification,my mistake or bug?

大きさは問題なく描画位置が zoom/2 pixelずれている気がしますが...。

[image]

```
#include <M5Stack.h>
#include <LovyanGFX.hpp>

static LGFX lcd;
static LGFX_Sprite sprite(&lcd);

#define SW (18)
#define SH (18)

void setup()
{
    M5.begin(false /* LCD*/, true /* SD*/ , true /*Serial*/);

    lcd.init();
    lcd.setRotation(1);
    lcd.setBrightness(32);
    
    sprite.createSprite(SW,SH);

    float z = 1.0f;

    sprite.fillScreen(TFT_DARKGREEN);
    sprite.setPivot(0,0);
    for(int i=0; i<4; i++, z+=1.0f)
    {
	sprite.pushRotateZoom(i * sprite.width(), 32, 0, 1.0f, z);
	sprite.pushRotateZoom(180, 32 + i * sprite.height(), 0, z, 1.0f);
    }
    lcd.drawLine(0,32,SW*5,32,TFT_RED);
    lcd.drawLine(180,32, 180, 32+SH*5, TFT_BLUE);

    z = 1.0f;
    sprite.fillScreen(TFT_YELLOW);
    sprite.setPivot(SW/2,SH/2); // center
    for(int i=0; i<4; i++, z+=1.0f)
    {
	sprite.pushRotateZoom(i * sprite.width() + SW/2, 180, 0 ,1.0f, z);
	sprite.pushRotateZoom(216, 144 + i * sprite.height() + SH/2, 0, z, 1.0f);
    }

    for(int i = 0; i < 5; ++i)
    {
	lcd.drawLine(0,180 - i*SH/2, 100, 180 - i*SH/2, TFT_RED);
	lcd.drawLine(0,180 + i*SH/2, 100, 180 + i*SH/2, TFT_RED);
	lcd.drawLine(216 - i*SW/2 ,144,216 - i*SW/2, 240, TFT_BLUE);
	lcd.drawLine(216 + i*SW/2 ,144,216 + i*SW/2, 240, TFT_BLUE);
    }
    lcd.drawLine(0,180,100,180, TFT_RED);
    lcd.drawLine(216,144,216,240, TFT_BLUE);
}

void loop()
{
    delay(1);
}
```

lovyan03, 2020-12-03:
これはバグではありません。
ベースとなっている TFT_eSPI の pushRotate の挙動に準拠しています。
TFT_eSPIでは pivot が "ピクセルの中心" を指す仕様になっているため、
厳密には +0.5ピクセル 右下にオフセットする仕様になっています。

LovyanGFX では pivot を実数指定できるように変更を加えたので、
必要に応じて pivotを -0.5してズレを相殺することができます。

GOB52, 2020-12-03:
挙動について説明とアドバイスありがとうございます。
pivotの補正でいけそうです。

closed 2020-12-03 by GOB52