lovyan03/LovyanGFX #73: error: no matching function for call to 'max(int, long int)'

matsujirushi, 2020-12-08:
I got below error for Wio Terminal.

```
.pio\libdeps\seeed_wio_terminal\LovyanGFX\src\lgfx\panel\Panel_IT8951.cpp:411:82: error: no matching function for call to 'max(int, long int)'
                 pixel = std::min(15, std::max(0, pixel + btbl[(x + prev_pos) & 3]) >> 6);
                                                                                  ^
```

[image]

lovyan03, 2020-12-08:
Thanks for reporting this !
I've hastily reflected the fixes in the develop branch.

matsujirushi, 2020-12-08:
Super!

I checked. It is fine.
[image]

Thank you.

Kabron287, 2020-12-10:
I applyed the fix, but got:
Panel_IT8951.cpp: 411:82: error: no matching function for call to 'max(int, long int)
                 pixel = std*: min(15, std::max(0, pixel + btbl[(x + prev_pos) & 3]) >> 6)

Kabron287, 2020-12-10:
pixel = std::min(15, std::max(0, (int)(pixel + btbl[(x + prev_pos) & 3] >> 6)));
Solved it.

lovyan03, 2020-12-10:
The fix has been made to the develop branch and is not yet reflected in the master branch.

commit 79343480 (2020-12-08): fix #73
closed 2020-12-12 by lovyan03