lovyan03/LovyanGFX #83: DOIT Devkit V1.0 + TFT 3.5 Touch (ILI9488)  about touch interface can't calibrate. How to fix it?

BlynkGO, 2020-12-20:
I used  Lolin D32 Pro + TFT3.5 Touch , it can calibrate well,

However when change the ESP32 board to DOIT Devkit V1.0 (30pins) + TFT3.5 Touch.
It can't calibrate. The result is the following clip.
https://streamable.com/iaebdt

[Code]
```
#include <LovyanGFX.hpp>

struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 23;
  static constexpr int spi_miso = 19;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ILI9488 panel;
static lgfx::Touch_XPT2046 touch;

void panel_d32pro_config(){
  panel.freq_write = 20000000;
  panel.freq_fill  = 27000000;
  panel.freq_read  = 16000000;
  panel.spi_mode = 0;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;
  panel.spi_cs = 14;
  panel.spi_dc = 27;
  panel.gpio_rst = 33;
  panel.gpio_bl  = 32;
  panel.pwm_ch_bl = 7;
  panel.backlight_level = true;
  panel.invert = false;
  panel.rgb_order = false;
  panel.memory_width  = 320;//320;
  panel.memory_height = 480;//240;
  panel.panel_width  = 320;//320;
  panel.panel_height = 480;//240;
  panel.offset_x = 0;
  panel.offset_y = 0;
  panel.rotation = 3;
  panel.offset_rotation = 0;  
}

void panel_devkit_config(){
  panel.freq_write = 20000000;
  panel.freq_fill  = 27000000;
  panel.freq_read  = 16000000;
  panel.spi_mode = 0;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;
  panel.spi_cs = 15;
  panel.gpio_rst = 2;
  panel.spi_dc = 4;
  panel.gpio_bl  = 5;
  panel.pwm_ch_bl = 1;
  panel.backlight_level = true;
  panel.invert = false;
  panel.rgb_order = false;
  panel.memory_width  = 320;//320;
  panel.memory_height = 480;//240;
  panel.panel_width  = 320;//320;
  panel.panel_height = 480;//240;
  panel.offset_x = 0;
  panel.offset_y = 0;
  panel.rotation = 3;
  panel.offset_rotation = 0;  
}

void touch_config(){
  touch.spi_host = VSPI_HOST;  // VSPI_HOST or HSPI_HOST
  touch.spi_sclk = 18;
  touch.spi_mosi = 23;
  touch.spi_miso = 19;
  touch.spi_cs   = 27;
  touch.freq = 1000000;
  touch.bus_shared = true;  // If the LCD and touch share SPI, set to true.

  touch.x_min = 470; //0;
  touch.x_max = 3740; //319;
  touch.y_min = 3850;//0;
  touch.y_max = 447; //319;  
}

void setup(void)
{
  Serial.begin(115200);
  
//  panel_d32pro_config();   // D32Pro + TFT3.5 Touch
  panel_devkit_config();     // DOIT Devkit V1.0  + TFT3.5 Touch
  lcd.setPanel(&panel);
  touch_config();
  lcd.setTouch(&touch);

  lcd.init();

  if (lcd.width() > 240 || lcd.height() > 240) lcd.setTextSize(2);

  if (lcd.touch())
  {
    if (lcd.width() < lcd.height()) lcd.setRotation(3 & (lcd.getRotation() + 1));
    lcd.drawString("touch the arrow marker.", 0, lcd.height()>>1);
    lcd.calibrateTouch(nullptr, 0xFFFFFFU, 0x000000U, 30);
    lcd.clear();
  }
}

uint32_t count = ~0;
void loop(void)
{
  delay(10);
  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);

  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30,30,lcd.width()-60,lcd.height()-60,random(65536));

  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    Serial.printf("x,y = %d, %d\n", x,y);
    lcd.fillRect(x-2, y-2, 5, 5, random(65536));
  }
}

```

How to fix the issue?

Thank you very much.

lovyan03, 2020-12-20:
@BlynkGO 
If it is working correctly with LolinD32Pro, it is unlikely to be a hardware failure or a library failure.
As far as the symptoms are concerned, it is always misidentified as being touched.
Perhaps it is not receiving data from XPT2046.

First, check the wiring.

touch.spi_cs = 27;
static constexpr int spi_miso = 19;

I can't verify that this matches the actual connection.
You may want to try using a different GPIO just to be sure.

tobozo, 2020-12-20:
pinout numbering differs between ESP32-Wrover and ESP32-Wroom, it's very likely the Touch config needs to be changed

BlynkGO, 2020-12-20:
@lovyan03 @tobozo 
When I tested DOIT + TFT 3.5 by TFT_eSPI library 
config pin as LovyanGFX library , it can work well.

For me , LovyanGFX is better than TFT_eSPI, 
I would like to use LovyanGFX .

The issue maybe occurs from GPIO19, but I don't know how to set
----------------------------------------
static constexpr int spi_miso = 19;
----------------------------------------

How to put this line in the code?

lovyan03, 2020-12-21:
@BlynkGO 
Hmmm, there is no cause I can think of right now.
The superficial appearance of the phenomenon means that it is always misinterpreted as being touched, but we need more in-depth information.
If you have an oscilloscope, observing the waveform might tell you something.

I can't reproduce the phenomenon with what I have, so I'm relying on your debugging.

BlynkGO, 2020-12-21:
Thank you very much.
I have tried to understand the LovyanGFX's touch in library code,
but I don't know how to fix.

At this point I backward to another method that 
send `spi_t spi_handle[spi_host]` to outside from LovyanGFX' spi system and create another touch library. Now it can work.

However I feel satisfied to use the original of LovyanGFX' touch more than this way.

If the issue can know cause and is fixed ,
please update for the issue.

Arigato gozaimasu.

tobozo, 2022-10-23:
closing this issue as solved, feel free to reopen if needed

closed 2022-10-23 by tobozo