lovyan03/LovyanGFX #84: How to display image with ESP32 3.5" TFT touch(capacitive) board

gigwegbe, 2020-12-20:
Hello @lovyan03,
I have ESP32 3.5" TFT touch(capacitive) board and I want to push the image from the camera with CAMERA_PIXEL_FORMAT -> PIXFORMAT_GRAYSCALE however, I got the result below. 
[image]

``` camera_fb_t *img = capture();
    if (img == nullptr || img == 0) 
    {
        Serial.printf("snap fail\n");
        return;
    }

    tft.pushImage(0, 0, 96, 96, (lgfx::swap565_t*)img->buf); 
}
```
How do I display a grayscale image on the TFT screen?
For more details check the repo - [Link](https://github.com/gigwegbe/person-detection-with-esp32)

lovyan03, 2020-12-21:
@gigwegbe 
I have now added a new type "lgfx::grayscale_t" to the develop branch.
Try to get the latest develop branch and cast it to lgfx::grayscale_t instead of lgfx::swap565_t .

gigwegbe, 2020-12-21:
Thanks @lovyan03, 
Worked like a charm.
[image]

closed 2020-12-21 by gigwegbe