lovyan03/LovyanGFX #87: please add support for the ILI9481 display if possible

ihayri, 2020-12-30:
Hi Lovyan, first of all, thank you for your great work on making this library. Was using the TFT_eSPI from Bodmer, thanks to him as well of course, but yours seems a bit faster and I love your analog gauge. Today I received my ILI9481 IPS displays and wonder if you could add that driver to your library. The display is a 3.5" 480x320 pixel with the FT6236 capacitive touchscreen controller, which you library supports as well, and has the ILI9481 driver chip as opposed to the ILI9488 which my Makerfabs display uses. Btw, Bodmer's library does support the ILI9481 and I tried to add it to your library but the configuration files are quite different. I am of course happy to test and give feedback. The reason why I did go with the ILI9481 is that the display is IPS and fully sunlight readable which I need for my project. Thank you very much in advance and if there is anything I can do to help. please let me know.
Happy New Year.

lovyan03, 2020-12-31:
@ihayri 
Thank you for your interest !
I don't own an ILI9481 yet, so I can't check how it works.
I would also like to get ILI9481. Please let me know the product page URL and other information of the panel you purchased.

I've tentatively added the definition to the develop branch, so please try it out.

I think the basic functions will probably work, but there are many things like adjusting the rotation direction and reading pixel data that I need to see in action.

ihayri, 2020-12-31:
Hi, thank you so much for replying. This is the link where I did buy my displays, it is the ILI9481 with single point touch display that uses the FT6236 chip that your great library supports as well.

https://www.aliexpress.com/item/1005001309491536.html

Here is a video I made today, where I do show the display and after much searching I finally found a library that does work with the ILI9481 and over SPI, as most other libraries do use parallel connection, which I would like not do do as it uses too many pins.

Your library works great with the ILI9488 which you can see in the video as well.

https://youtu.be/bl1exToW0Wo 

https://github.com/moononournation/Arduino_GFX

I hope it is ok if I put links here, if not please remove them or I can do it I presume.

If there is anything I can do to help please let me know, thanks.

lovyan03, 2020-12-31:
@ihayri 
Thanks for letting me know where to buy it !
I placed an order immediately.
It will probably take about two weeks to arrive.

In the meantime, I've added support for ILI9481 to the "develop" branch, so you can try it out.

ihayri, 2020-12-31:
You are most welcome. I will try the the library and let you know later on.
Will go to bed soon as it is 5 o' clock in the morning here.
Take care and thank you.

ihayri, 2020-12-31:
Hi lovyan, I am at it since this morning but so far I could not get it to work, I am probably doing something wrong. This is the code I am using:
``// How to set up the SPI with LovyanGFX
// LovyanGFX SPIバスおよび使用パネルの設定を伴う使い方

#include <LovyanGFX.hpp>
#include "Panel_ILI9481.hpp"

// Create a structure for SPI configuration
// SPI設定用の構造体を作成します。

// for ESP32 SPI
struct LGFX_Config
{
  // You can change the name of the structure from "LGFX_Config"
  // But do not change the name and type of the members.
  // 構造体の名称 "LGFX_Config" は変更しても構いませんが、
  // 構造体の各メンバ変数の名前と型は例の通りにしてください。

  // Select the type of SPI  (VSPI_HOST or HSPI_HOST)
  //  使用するSPIを VSPI_HOST または HSPI_HOST で設定します。
  static constexpr spi_host_device_t spi_host = HSPI_HOST;

  // Set the DMA channel number (1 or 2)
  // If you don't want to use DMA, omit it or set it to 0.
  // 使用するDMAチャンネルを 1か2で設定します。
  // 使用しない場合は省略するか0を設定します。
  static constexpr int dma_channel = 0;

  // Set the SPI SCLK pin number
  // SPIのSCLKのピン番号を設定します。
  static constexpr int spi_sclk = 14;

  // Set the SPI MOSI pin number
  // SPIのMOSIのピン番号を設定します。
  static constexpr int spi_mosi = 13;

  // Set the SPI MISO pin number
  // If you share the SPI bus with an SD card, be sure to set up MISO as well.
  // If you don't want to use MISO, omit it or set it to -1.
  // SPIのMISOのピン番号を設定します。
  // SDカード等と共通のSPIバスを使う場合はMISOも必ず設定してください。
  // 使わない場合は省略するか-1を設定します。
  static constexpr int spi_miso = 12;

  static constexpr int spi_cs = 18;
  static constexpr int spi_dc = 22;
  static constexpr int spi_rst = 21;

  // Set the data length of SPI communication.
  // If you want to use the LCD for RPi, set it to 16.
  // SPI通信のデータ長を指定します。
  // RaspberryPi用のLCD等を使用する場合に16を指定します。
  // 省略時は 8 です。大抵のパネルは8ですので、基本的には省略してください。
  static constexpr int spi_dlen = 8;
};

/* for ESP32 Parallel
  struct LGFX_Config
  {
  // Select the type of I2S  (I2S_NUM_0 or I2S_NUM_1)
  // 使用するI2Sを I2S_NUM_0 または I2S_NUM_1 で設定します。
  // 省略時は I2S_NUM_0です。
  static constexpr i2s_port_t i2s_port = I2S_NUM_0;

  // Set the parallel pin number
  // パラレル接続の各種ピン番号を設定します。
  static constexpr int gpio_wr  =  4;
  static constexpr int gpio_rd  =  2;
  static constexpr int gpio_rs  = 15;
  static constexpr int gpio_d0  = 12;
[... 263 lines cut ...]
  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30, 30, lcd.width() - 60, lcd.height() - 60, random(65536));

  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    lcd.fillRect(x - 2, y - 2, 5, 5, random(65536));

    Serial.print("x = ");
    Serial.print(x);
    Serial.print("  y = ");
    Serial.println(y);

  }
}

ihayri, 2020-12-31:
Hope this is the correct way to add code :-)

ihayri, 2020-12-31:
I did add a new tab with Panel_ILI9481.hpp , with #include "Panel_ILI9481.hpp"
Which might not be needed or completely wrong, sorry.

lovyan03, 2020-12-31:
@ihayri 
Yes, you don't need that include statement.

Are you currently experiencing any errors?
Or is there something wrong with the operation?
I'm counting on your reports. Please let me know what's happening.

ihayri, 2020-12-31:
Hi there, I am working on it with Tobozo since this morning and we have some steps toward it working.
this is the stripped down version:

#include <LovyanGFX.hpp>
#include "Panel_ILI9481.hpp"

struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = HSPI_HOST;

  static constexpr int dma_channel = 0;

  static constexpr int spi_sclk = 14;

  static constexpr int spi_mosi = 13;

  static constexpr int spi_miso = 12;

  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;

static lgfx::Panel_ILI9481 panel;

static lgfx::Touch_FT5x06 touch;

void setup(void)
{

  Serial.begin(115200);

  panel.freq_write = 12000000;

  panel.freq_fill  = 12000000;

  panel.freq_read  = 12000000;

  panel.spi_mode = 0;

  panel.spi_mode_read = 0;

  panel.len_dummy_read_pixel = 8;

  panel.spi_read = true;

  panel.spi_3wire = false;

  panel.spi_cs = 15;

  panel.spi_dc = 33;

  panel.gpio_rst = 26;

  panel.gpio_bl  = 5;

  panel.pwm_ch_bl = 5;

  panel.backlight_level = true;

  panel.reverse_invert = true;

  panel.rgb_order = false;

  panel.memory_width  = 320;
  panel.memory_height = 480;

  panel.panel_width  = 320;
  panel.panel_height = 480;

  panel.offset_x = 0;
  panel.offset_y = 0;

  panel.rotation = 1;
  
  panel.offset_rotation = 0; // tried with 1, no change on the display

  lcd.setPanel(&panel);

  lcd.init();
  Serial.println("lcd.init done");

  if (lcd.width() > 240 || lcd.height() > 240) lcd.setTextSize(2);

  if (lcd.touch())
  {
    if (lcd.width() < lcd.height()) lcd.setRotation(3 & (lcd.getRotation() + 1));

    lcd.drawString("touch the arrow marker.", 0, lcd.height() >> 1);

    lcd.clear();
  }
}

uint32_t count = ~0;

void loop(void)
{
  delay(10);
  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);

  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30, 30, lcd.width() - 60, lcd.height() - 60, random(65536));

  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    lcd.fillRect(x - 2, y - 2, 5, 5, random(65536));

  }
}

ihayri, 2020-12-31:
and a picture
[image]

ihayri, 2021-01-02:
Hello Lovyan,

I did some progress and it does work now, the only thing I can not add are custom colors. Here is the current code:

#include <LovyanGFX.hpp>

struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = HSPI_HOST;

  static constexpr int dma_channel = 0;

  static constexpr int spi_sclk = 14;

  static constexpr int spi_mosi = 13;

  static constexpr int spi_miso = 12;

  static constexpr int spi_dlen = 8;

  static constexpr int TFT_DARKGRAY    = 0x7BEF;  // does not work
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;

static lgfx::Panel_ILI9481 panel;

//static lgfx::Touch_FT5x06 touch;

void setup() {

  Serial.begin(115200);

  panel.freq_write = 17000000;

  panel.freq_fill  = 17000000;

  panel.freq_read  = 27000000;

  panel.spi_mode = 0;

  panel.spi_mode_read = 0;

  panel.len_dummy_read_pixel = 8;

  panel.spi_read = true;

  panel.spi_3wire = false;

  panel.spi_cs = 15;

  panel.spi_dc = 33;

  panel.gpio_rst = 26;

  panel.gpio_bl  = 5;

  panel.pwm_ch_bl = 5;

  panel.backlight_level = true;

  panel.reverse_invert = true;

  panel.rgb_order = false;

  panel.memory_width  = 320;
  panel.memory_height = 480;

  panel.panel_width  = 320;
  panel.panel_height = 480;

  panel.offset_x = 0;
  panel.offset_y = 0;

  panel.rotation = 1;

  panel.offset_rotation = 0;

  lcd.setPanel(&panel);

  lcd.init();
  lcd.setBrightness(255);
  lcd.setColorDepth(24);
//  lcd.fillScreen (TFT_DARKGRAY);  // produces vertical lines and wrong color
  lcd.fillScreen (TFT_BLACK);
  
  lcd.setTextSize(4);
  lcd.setTextColor(TFT_RED, TFT_BLACK);
  lcd.drawString("R", 30, 30);
  lcd.setTextColor(TFT_GREEN);
  lcd.drawString("G", 60, 30);
  lcd.setTextColor(TFT_BLUE);
  lcd.drawString("B", 90, 30);
  lcd.setTextColor(TFT_YELLOW, TFT_BLUE);
  lcd.setTextSize(3);
  lcd.drawString("ILI9481 Working", 30, 90);

  lcd.drawRect(30, 30, lcd.width() - 60, lcd.height() - 60, TFT_BLUE);

  delay(3000);
}

void loop() {

  lcd.setTextSize(4);
  lcd.setTextColor(TFT_WHITE, TFT_RED);
  lcd.drawString("R", 30, 30);
  delay(1000);
  lcd.setTextColor(TFT_BLACK, TFT_GREEN);
  lcd.drawString("G", 60, 30);
  delay(1000);
  lcd.setTextColor(TFT_YELLOW, TFT_BLUE);
  lcd.drawString("B", 90, 30);

}

ihayri, 2021-01-02:
PS: it does not like random colors, that was the problem in the first picture.

I am sure you can fix that. Somehow the panel frequencies are pretty low.

[ILI9481_v0.35.pdf](https://github.com/lovyan03/LovyanGFX/files/5760226/ILI9481_v0.35.pdf)

ihayri, 2021-01-02:
maybe the pdf file will help you further if you don't have it already.

Thank you very much for all your work and thanks to Tobozo as well of course.

Just noticed that he edited my first code upload to me more readable, thanks Tobozo :-)

lovyan03, 2021-01-03:
@ihayri 
Thank you for the information. 

Do not make TFT_DARKGRAY a member of LGFX_Config.
It should be placed outside of LGFX_Config.

I ordered ILI9481 and it seems it has already shipped.
I will make adjustments when I receive the product.

ihayri, 2021-01-03:
No problem, anyway I can help is good.

Correct, never mind the color definition, did remove that and put it at the beginning of the code.
It works rather nicely with digits updating without any flicker by just using fore and background colors.

So far I am loving this display and tomorrow I can finally try it outside in sunlight, where it is supposed to be seen without a problem. Viewing angles are great of course, being a IPS display.

Thank you again for your great library and all your help.

fsender, 2021-11-13:
Now I'm using ILI9481 on an ESP32 with 12MHz SPI. It's too slow for me. And its pixels cannot be read in your examples. What can I do? My sketch needs to read pixels on it.

lovyan03, 2021-11-14:
@fsender 
Try changing the settings here.
```
  panel.spi_3wire = false; // or true
```

tobozo, 2022-10-23:
closing this issue as solved, feel free to reopen if needed

closed 2022-10-23 by tobozo