lovyan03/LovyanGFX #88: Object oriented programming crashes my M5StickC Plus

m1cr0lab, 2021-01-06:
Hi @lovyan03,

Sorry to bother you, but I can't figure out what's going on?

I'm trying to run simple code on an M5StickC Plus, but the ESP32 reacts strangely.

If it helps, I work with Visual Studio Code and PlatformIO on macOS 10.14.6:
- VSCode v1.52.1
- PIO Core v5.0.4
- PIO Home v3.3.1
- Espressif 32 v2.1.0

Here is the configuration file of my project :

```ini
; platformio.ini

[env:stick+]
platform   = espressif32
board      = m5stick-c
framework  = arduino
lib_deps   =
    m5stack/M5StickCPlus @ ^0.0.1
    lovyan03/LovyanGFX @ ^0.3.4
```

Here is the tree structure of my `dummy` project:

```
dummy
├── platformio.ini
└── src
    └── main.cpp
```

And here is my initial code:

```cpp
// main.cpp

#include <M5StickCPlus.h>
#include <LovyanGFX.h>

LGFX lcd;
LGFX_Sprite fb(&lcd);

void setup() {
    M5.begin(false, true, false);
    lcd.init();
    lcd.setColorDepth(16);
    fb.createSprite(135, 240);
    
}

void loop() {
    M5.update();
    fb.clear(0xfe40);
    fb.pushSprite(0, 0);
}
```

Well... so far so good, I'm getting an orange screen on my stick.

Now I want to deport the content of my `loop()` in a `GraphicsEngine` class :

```
dummy
├── lib
│   └── GraphicsEngine
│       └── GraphicsEngine.h
├── platformio.ini
└── src
    └── main.cpp
```

So here is the new version of the code :

```cpp
// main.cpp

#include <M5StickCPlus.h>
#include <LovyanGFX.h>
#include "GraphicsEngine.h"

LGFX lcd;
LGFX_Sprite fb(&lcd);

GraphicsEngine ge;

void setup() {
    M5.begin(false, true, false);
    lcd.init();
    lcd.setColorDepth(16);
    fb.createSprite(135, 240);
}

void loop() {
    M5.update();
    ge.draw(fb);
}
```

```cpp
// lib/GraphicsEngine/GraphicsEngine.h

#pragma once

#include <LovyanGFX.h>

class GraphicsEngine {

    public:

        void draw(LGFX_Sprite &fb) {
            fb.clear(0xfe40);
            fb.pushSprite(0, 0);
        }
};
```

Everything functions normally, as before.

Now I want to separate the declarative part from the implementation of the class:

```
dummy
├── lib
│   └── GraphicsEngine
│       ├── GraphicsEngine.cpp
│       └── GraphicsEngine.h
├── platformio.ini
└── src
    └── main.cpp
```

And here's the new code (`main.cpp` does not change):

```cpp
// lib/GraphicsEngine/GraphicsEngine.h

#pragma once

#include <LovyanGFX.h>

class GraphicsEngine {

    public:

        void draw(LGFX_Sprite &fb);

};
```

```cpp
// lib/GraphicsEngine/GraphicsEngine.cpp

#include "GraphicsEngine.h"

void GraphicsEngine::draw(LGFX_Sprite &fb) {

    fb.clear(0xfe40);
    fb.pushSprite(0, 0);

}
```

**And that's when it all goes to hell!**
Black screen... and the stick restarts continuously.

Here are the crash logs if it can help you:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7f6c  PS      : 0x00060330  A0      : 0x800d3db6  A1      : 0x3ffb1f00  
A2      : 0x400d268c  A3      : 0x3ffc01b8  A4      : 0x000d268c  A5      : 0x0000268c  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb1ee0  
A10     : 0x3ffc01b8  A11     : 0x00010000  A12     : 0x00000000  A13     : 0x0000ffff  
A14     : 0x000000f0  A15     : 0x00000012  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000078  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d7f6c:0x3ffb1f00 0x400d3db3:0x3ffb1f90 0x400dab8d:0x3ffb1fb0 0x40088c99:0x3ffb1fd0

Rebooting...
```

There's probably something I'm missing, but I don't see what!

I seem to remember that I didn't encounter this kind of problem some time ago, when I was using version 0.2.4 of LovyanGFX...

Has something happened in the meantime that makes this way of structuring my code not work anymore?

tobozo, 2021-01-06:
I've had a similar problem with Arduino IDE, and found out all my .cpp files are ignored at compilation unless they are included as libraries :-(

A quick test to see if this is the problem you're having is to move the contents of `GraphicsEngine.cpp` into `GraphicsEngine.h` and have the problem go away.

If so, maybe adding `lib_ldf_mode = deep` in your platformio.ini (and playing with the different values) may help

m1cr0lab, 2021-01-06:
Thanks for the idea @tobozo ... but :

- actually, if I move the content of the `.cpp`file to the `.h` file it works,
- but when I separate the code and specify the `lib_ldf_mode = deep` it doesn't help
 _(the `chain+` and `deep+` options don't help, of course)_

In fact, the compilation goes well. It's at runtime that it goes wrong.

Yet I've always done it this way in my projects on ESP32. I've never had this problem before. And I remember that this way of structuring the code worked well with an older version of LovyanGFX (at least the 0.2.4 version anyway).

And now that I've switched to 0.3.4, the problem occurs! It's really curious. Does it come from the way the library is structured? The `deep` option however analyzes all dependencies recursively. Rhaaa... it's frustrating not to understand.

Anyway, **thank you very much** for answering me so quickly.
Maybe @lovyan03 will have an idea?

m1cr0lab, 2021-01-06:
I have a new lead...

If I rewrite the code this way, it works:

```cpp
// main.cpp

#include <M5StickCPlus.h>
#include <LovyanGFX.h>
#include "GraphicsEngine.h"

LGFX lcd;

GraphicsEngine ge;

void setup() {
    M5.begin(false, true, false);
    lcd.init();
    lcd.setColorDepth(16);
}

void loop() {
    M5.update();
    ge.draw(lcd);
}
```

```cpp
// lib/GraphicsEngine/GraphicsEngine.h

#pragma once

#include <LovyanGFX.h>

class GraphicsEngine {

    public:

        void draw(LGFX &lcd);

};
```

```cpp
// lib/GraphicsEngine/GraphicsEngine.cpp

#include "GraphicsEngine.h"

void GraphicsEngine::draw(LGFX &lcd) {

    LGFX_Sprite fb(&lcd);
    fb.createSprite(135, 240);
    fb.clear(0xfe40);
    fb.pushSprite(0, 0);
    fb.deleteSprite();

}
```

But why, when I create the framebuffer as a global variable, doesn't it work?

tobozo, 2021-01-06:
LGFX can handle multiple displays, and the global instance has no config before the init(), so you were probably accessing a configless instance when the crash occured.

noob idea: the GraphicsEngine could probably be instanciated after lcd.init() and also get the `lcd` object injected in order to avoid the global access.

```C

GraphicsEngine *ge = nullptr;

void setup()
{
   lcd.init();
   ge = new GraphicsEngine( &lcd );
}

```

also I'm used to pointers when it comes to passing sprites as arguments

lovyan03, 2021-01-07:
@m1cr0lab 
There is a fatal bug, sorry about that.
Try the one on the "develop" branch.
It was probably fixed in the commit on December 19 last year.

The cause is that the structure of the LovyanGFX type is different in each compilation unit.
In the main.cpp, LovyanGFX file system support is enabled by the presence of the M5StickCPlus library, but GraphicsEngine.cpp no longer includes file system support.
This problem was fixed late last year, but we have not yet released it.

m1cr0lab, 2021-01-07:
Indeed, by including in priority the `M5StickCPlus` library in the `.cpp` file, the problem disappears :

```cpp
// lib/GraphicsEngine/GraphicsEngine.cpp

#include <M5StickCPlus.h>
#include "GraphicsEngine.h"

void GraphicsEngine::draw(LGFX_Sprite &fb) {

    fb.clear(0xfe40);
    fb.pushSprite(0, 0);

}
```

@lovyan03 @tobozo Thank you both for your precious help and diligence.
And above all thank you for the fantastic work you share with us with this absolutely great library.

lovyan03, 2021-01-09:
@m1cr0lab
I have created a new release ver 0.3.5 which fixes this issue.
I hope you will try it when you have time and see if it solves issue.
thank you !

m1cr0lab, 2021-01-09:
Thank you very much @lovyan03, everything is working fine now.

closed 2021-01-09 by lovyan03