lovyan03/LovyanGFX #89: Idea: MultiSprite

luisllamasbinaburo, 2021-01-07:
Hello, I have created a "MultiSprite" concept, based on the examples of the library.

Basically, it manage the case where you don't have memory enough to store a sprite, and you have to split in "bands".

```c++
class MultiSprite
{
public:
	MultiSprite()
	{

	}

	void Init(LGFX* lcd, std::int_fast8_t div = 1)
	{
		_lcd = lcd;
		lcd_width = _lcd->width();
		lcd_height = _lcd->height();

		for (;;) {
			sprite_height = (lcd_height + div - 1) / div;
			bool fail = false;
			for (std::uint32_t i = 0; !fail && i < 2; ++i)
			{
				sprite[i].setColorDepth(lcd->getColorDepth());
				fail = !sprite[i].createSprite(lcd_width, sprite_height);
			}
			if (!fail) break;
			for (std::uint32_t i = 0; i < 2; ++i)
			{
				sprite[i].deleteSprite();
			}
			div *= 2;
		}
		_div = div;
	}

	LGFX* _lcd;

	uint32_t _div;
	uint16_t currentStrip = 0;

	std::int_fast16_t lcd_width;
	std::int_fast16_t lcd_height;

	MultiSpriteStrip sprite[2];
	int_fast16_t sprite_height;

	void startWrite() {_lcd->startWrite();};
	void endWrite() {_lcd->endWrite();};
	void display() {_lcd->display();};

	void Render(void (*render)(MultiSpriteStrip&, uint8_t))
	{
		currentStrip = 0;
		for (std::int32_t y = 0; y < lcd_height; y += sprite_height)
		{
			auto index = currentStrip % 2;
			sprite[index].lowerY = y;
			sprite[index].upperY = y + sprite_height;

			render(sprite[index], currentStrip);

			std::uint32_t len = sprite_height * lcd_width;
			if (y + sprite_height > lcd_height) {
				len = (lcd_height - y) * lcd_width;
			}
			_lcd->pushPixelsDMA(sprite[index].getBuffer(), len);
			currentStrip++;
		}
	}
};
```
No conversion between global (full screen) and local (sprite band) is required
This way, for example, the "moving circle" demo is as easy as
```c++
void draw(MultiSpriteStrip& sprite, uint8_t spriteNum)
{
	sprite.clear();

	for (std::uint32_t i = 0; i < obj_count; i++)
	{
		sprite.drawCircle(circles[i].x, circles[i].y, circles[i].r, circles[i].color);
	}

	if (spriteNum >= 3) drawFps(sprite);
}

void loop(void)
{
	fpsCounter.Update(millis());

	for (std::uint32_t i = 0; i != obj_count; i++)
	{
		circles[i].move();
	}

	multiSprite.Render(draw);
}
```

Please, let me know if you find it interesting, and if there is anything more that i can do for helping (share the code, fork, PR,...)

Thank you for you great work

lovyan03, 2021-01-08:
@luisllamasbinaburo 
Thank you for posting your ideas, they are very helpful.

I think it is very good to have this kind of device on the user code side.

On the other hand, I also believe that examples need to be in a state that makes it easy to understand the behavior of the library.

(However, I also think that the current examples are not easy to understand.)

I would like to increase the number of examples with plain expressions as much as possible, but the progress is not good…

luisllamasbinaburo, 2021-01-08:
Hello @lovyan03 . Sorry, i don't explain my self.  The point is not to have a cleaner or shorter code. It's about interoperatibility.
This object "wraps" the managing of an array of sprited-bands in a single class. I created the bands depending of the availeble memory, and also handle drawing on that band transforming global to local coordinates

So, for example, you can draw a X,Y pixel of screen, and you don't have to worry about coordinate transformation.
This way, tha same  program will run in any device, with more/less memory or pixels, it doesn't matter if it requires 1, 2 ... or 64 bands.

In fact, you can use this object to run any code (for example, all the examples of other libraries) with no code modification, only changind the original LCD object for a MultiSprite object.

But it's OK, i understand you want to keep you code as clean as you can. I will add this to a fork.
Thanks again!!

closed 2021-01-08 by luisllamasbinaburo