lovyan03/LovyanGFX #94: Introducing CI using GitHub Actions

estshorter, 2021-01-12:
How about introducing CI using GitHub Actions to ensure successful build when new PRs are submitted? (though this doesn't help to detect runtime error...

Web pages below would be good examples.

PlatformIO document:
https://docs.platformio.org/en/latest/integration/ci/github-actions.html

Files in FastLED:
https://github.com/FastLED/FastLED/blob/master/ci/ci-compile
https://github.com/FastLED/FastLED/blob/master/.github/workflows/build.yml

tobozo, 2021-01-12:
@estshorter thanks for your input :+1: 

Given the scope of this library, a few basic parallel jobs are required:

- ESP32 / Arduino
  - windows
  - mac
  - linux
- ESP32 / Platformio
  - windows
  - mac
  - linux
- ESP32 / esp-idf
  - windows
  - mac
  - linux
- SAMD / Arduino
  - windows
  - mac
  - linux
  
This is only the shallowest scope though, each job should also be tested against every combination of board selection + display device selection in order to be exhaustive.

The problem I've had when researching this is that such exhaustivity triggers billing on the github account (or exhausts whatever budget there is)

Unfortunately without this exhaustivity the tests are weak and question the utility of the CI, so maybe a self-hosted test suite is more appropriate ?

lovyan03, 2021-01-13:
I'm having problems with my recent release of LovyanGFX causing compile errors on SAMD51. Sorry about that.

The internal type representation is different between ESP32 and SAMD51, so we need to describe the types strictly.
Since I usually work with ESP32, I often miss compile errors in SAMD51.

lovyan03, 2021-10-03:
The implementation appears to have been a success.
Thank you very much.

closed 2021-10-03 by lovyan03

tobozo, 2021-10-06:
#151 added 3 separate workflows [`ArduinoBuild`](https://github.com/lovyan03/LovyanGFX/blob/CI-Tests/.github/workflows/ArduinoBuild.yml), [`PlatformioBuild`](https://github.com/lovyan03/LovyanGFX/blob/CI-Tests/.github/workflows/PlatformioBuild.yml) and [`IDFBuild`](https://github.com/lovyan03/LovyanGFX/blob/CI-Tests/.github/workflows/IDFBuild.yml)

The existing [`build`](https://github.com/lovyan03/LovyanGFX/blob/CI-Tests/.github/workflows/build.yml) using Arduino CI has been kept as is.

On push/pull-request, each workflow will create parallel jobs to test different boards.

|        Board        | Arduino IDE | esp-idf 4.1 | esp-idf 4.2 | esp-idf 4.3.1 | Plaformio | Arduino CI |
|:------------------- | :---------- | :------     | :------     | :------       |:--------- | :--------- |
| ESP32               | ✓           | ✓           | ✓           | ✓             | ✓         | ☐         |
| ESP32 S2            | ✓           | ☐           | ☐           | ✓             | ✓         | ☐         |
| ESP32 C3            | ☐           | ☐           | ☐           | ☐             | ☐         | ☐         |
| M5stickC            | ✓           | ☐           | ☐           | ☐             | ✓         | ✓         |
| M5stack Classic     | ✓           | ☐           | ☐           | ☐             | ✓         | ☐         |
| M5core2             | ✓           | ☐           | ☐           | ☐             | ✓         | ☐         |
| ESP8266 (D1 Mini)   | ✓           | ☐           | ☐           | ☐             | ✓         | ☐         |
| Wio Terminal        | ✓           | ☐           | ☐           | ☐             | ✓         | ✓         |
| Adafruit Pybadge M4 | ☐           | ☐           | ☐           | ☐             | ✓         | ✓         |