lovyan03/LovyanGFX #96: Setting a specific sprite color to act as alpha channel (ie. not drawPixel if color =r,g,b)

loxai, 2021-01-15:
Hi,

I've just started with ESP32 and using M5StickC to make some games.
I'm using your library, which seems great although I have a hard time finding out how to use it (eg. no idea how to generate bitmap/jpg/png hex arrays to be read by sprite.generateFromBitmap or similar, as that's not shown in the examples).
So I've made my custom pixel loader, with input generated in java and including a palette array and an array of indices, and I'd like to know how to set a color to act as a no draw pixel (to act as alpha, showing the pixel below the sprite).
Anyway, how can I achieve this simple alpha color feature... whether using existing bitmap loading (doc?) or compatible with my custom pixel loading?
I use sprite.drawPixel(x, y, lcd.color888(r,g,b)), where I set r, g, b from my custom image loader... maybe there's something like drawPixel(x, y, lcd.ALPHA)?
I understand maybe this is not possible, if data is sent in bulk to the display (really no idea how all this gfx stuff works internally, but saw some DMA reference that hints at that).
In that case, I guess my only option is to draw all relevant pixels directly to the lcd (ie. without sprites), skipping the alpha ones, which is going to be rather slow.

any other link to documentation, api reference would be great :)

Cheers

tobozo, 2021-01-15:
is that what you want to achieve ?

```C
#define ALPHA_COLOR 0xABCDEFu // like html rgb code #ABCDEF

sprite.pushSprite( x, y, ALPHA_COLOR );
```

loxai, 2021-01-15:
awesome, that works for me. thanks for the prompt solution.
as for the included options to load pictures, how do you convert a bmp into an array like in the PartyParrot.createFromBitmap example (https://github.com/lovyan03/LovyanGFX/blob/master/examples/Sprite/PartyParrot/PartyParrot.ino) or FlashMemSprite.setBuffer (https://github.com/lovyan03/LovyanGFX/blob/master/examples/Sprite/FlashMemSprite/image320x240x16.h) ?

tobozo, 2021-01-15:
I use the `xxd` command line or I go to [hexed.it](hexed.it), drop my file, select all bytes (CTRL+A) then right click and choose "Export selected bytes as code snippet"

closed 2021-01-27 by lovyan03

loxai, 2021-02-12:
Hope it's ok to reopen this issue. Not really a problem, just wanted to share the project I started using LovyanGFX, a set of mini games: https://github.com/loxai/rotaryGame

tobozo, 2021-02-12:
@loxai the repository you linked is still private :-)

loxai, 2021-02-12:
Oops, sorry about that