lovyan03/LovyanGFX #99: How to config for TTGO T-Display (ST7789)

BlynkGO, 2021-01-31:
I wish to use TTGO T-Display 1.3inch (ST7789  240x135) on LovyaGFX library.

[image]

If TTGO T-Display config by TFT-eSPI, it's configed by the following.

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

```c
#define ST7789_DRIVER

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required

//#define TFT_MISO -1

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

//#define SPI_FREQUENCY  27000000
  #define SPI_FREQUENCY  40000000   // Maximum for ILI9341

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
```

When I config on LovyanGFX by the following code , it occurs some error , how to fix it ?
Thank you.

```c
#include <LovyanGFX.hpp>

struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_miso = 19;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = -1;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ST7789 panel;

void setup(void)
{

//  panel.freq_write = 20000000;
  panel.freq_fill   = 40000000; //27000000;
  panel.freq_read   = 6000000;//16000000;
  panel.spi_mode    = 0;
  panel.spi_mode_read         = 0;
  panel.len_dummy_read_pixel  = 8;
  panel.spi_read    = true;
  panel.spi_3wire   = false;

  panel.spi_cs      = 5;
  panel.spi_dc      = 16;
  panel.gpio_rst    = 23;//-1;//33;

  panel.gpio_bl     = 4;//32;

  panel.pwm_ch_bl   = 7;
  panel.backlight_level = true;
  panel.reverse_invert = false;
  panel.rgb_order = false;
  
  panel.memory_width  = 240; //320;
  panel.memory_height = 135; //240;
  panel.panel_width   = 240; //320;
  panel.panel_height  = 135; //240;

  panel.offset_x = 0;
  panel.offset_y = 0;

  panel.rotation = 0;
  panel.offset_rotation = 0;

  lcd.setPanel(&panel);
  lcd.init();
}

uint32_t count = ~0;
void loop(void)
{
  delay(10);
  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);

  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30,30,lcd.width()-60,lcd.height()-60,random(65536));

  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    lcd.fillRect(x-2, y-2, 5, 5, random(65536));
  }
}

```

lovyan03, 2021-01-31:
missing mosi setting.
```
static constexpr int spi_mosi = -1;
```

BlynkGO, 2021-01-31:
> missing mosi setting.
> 
> ```
> static constexpr int spi_mosi = -1;
> ```

Thanks for the fast reply.

At this point ?
[image]

lovyan03, 2021-01-31:
Perhaps you are confusing MOSI with MISO.
Isn't it correct that MISO is -1 and MOSI is 19?

BlynkGO, 2021-01-31:
> Perhaps you are confusing MOSI with MISO.
> Isn't it correct that MISO is -1 and MOSI is 19?

Oh, it has already displayed now. I mistake between MISO and MOSI .
Thanks again.

closed 2021-01-31 by lovyan03