lovyan03/LovyanGFX #100: LovyanGFX_Clock on M5Stack Core2

ondrej1024, 2021-02-22:
I flashed the LovyanGFX_Clock firmware on an M5Stack Core2 with M2Burner. Now the Core2 is not switching on again. Tried the reset and power buttons. 

Any idea what happened? Did I corrupt the boot flash? How can I recover the device?

Any idea welcome.

Thanks, Ondrej

lovyan03, 2021-02-22:
Ah... that's not the farm I made. It's probably for M5Stack OLD Core.
If you write the firmware for Core2, it should come back.

lovyan03, 2021-02-22:
The M5Stack staff imported the LovyanGFX sample code into M5Burner without consulting me, so they are not under my control.
I'm having a bit of trouble with that.

ondrej1024, 2021-02-23:
> If you write the firmware for Core2, it should come back.

I had tried that but it didn't work. I was afraid I had bricked the device. Now I tried again and it worked :grinning: 

Yes, the firmware is probably only for the old Core device, but it doesn't mention it in the description of M5Burner.

Thanks for your help

closed 2021-02-23 by lovyan03