lovyan03/LovyanGFX #101: How to config for TFT 1.3inch ST7789

BlynkGO, 2021-02-22:
Hi,

I have tried to config for TFT1.3inch ST7789
by the following code 

```c
#include <LovyanGFX.hpp>

struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel  = 1;
  static constexpr int spi_miso     = -1; //19;
  static constexpr int spi_sclk     = 18;
  static constexpr int spi_mosi     = 23;
  static constexpr int spi_dlen     = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ST7789 panel;

void setup(void)
{
  panel.freq_write            = 20000000;
  panel.freq_fill             = 27000000;
  panel.freq_read             = 16000000;
  panel.spi_mode              = 0;
  panel.spi_mode_read         = 0;
  panel.len_dummy_read_pixel  = 8;
  panel.spi_read              = true;
  panel.spi_3wire             = true; //false;

  panel.spi_cs                = -1;
  panel.spi_dc                = 4;
  panel.gpio_rst              = 2;
  panel.gpio_bl               = 5;

  panel.pwm_ch_bl             = 7;
  panel.backlight_level       = true;
  panel.reverse_invert        = false;
  panel.rgb_order             = false;
  
  panel.memory_width          = 240;
  panel.memory_height         = 240;
  panel.panel_width           = 240;
  panel.panel_height          = 240;

  panel.offset_x              = 0;
  panel.offset_y              = 0;

  panel.rotation              = 0;
  panel.offset_rotation       = 0;

  lcd.setPanel(&panel);
  lcd.init();
}

uint32_t count = ~0;
void loop(void)
{
  delay(10);
  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);

  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);

  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);

  lcd.drawRect(30,30,lcd.width()-60,lcd.height()-60,random(65536));

  lcd.endWrite();
}
```

But by the code , it can't display anything.

When I tried on TFT_eSPI as the same pins-config.

```c
#define ST7789_DRIVER
#define TFT_WIDTH  240
#define TFT_HEIGHT 240

#define TFT_MISO -1   //19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS     -1
#define TFT_DC      4 
#define TFT_RST    2 
#define TFT_BL       5
```

TFT can display.
How to solve the issue?
Thank you.

lovyan03, 2021-02-22:
It seems that the ST7789 without CS pin needs to use SPI mode 3.

BlynkGO, 2021-02-23:
> It seems that the ST7789 without CS pin needs to use SPI mode 3.

Thank for the reply.

This TFT 1.3inch ST7789's pin as the image, 
it doesn't have CS pin.

[image]

How to replace -1 in cs pin in the panel ?

closed 2021-02-23 by lovyan03

summerfind, 2023-06-09:
Hi all. I got compiling error: 'LGFX_SPI' in namespace 'lgfx' does not name a template type for the line:
static lgfx::LGFX_SPI<LGFX_Config> tft;
how to fix?
Thanks.