lovyan03/LovyanGFX #104: Help documenting...

ropg, 2021-03-05:
Hi there...

I'd like to help creating a truly english version of the documentation, and possibly create spiffy documentation on Read The Docs by creating a documentation source directory. Is there interest for that?

ropg, 2021-03-05:
Have a look at PR #105. It's just a start of how it could be... I'd like feedback before doing lots of work...

lovyan03, 2021-03-05:
Hello,  @ropg !
Thanks for your cooperation and very interesting offer.
In fact, I am very bad at English. I use the help of google and DeepL for translation.
Also, I understand that the documentation for LovyanGFX is quite lacking at the moment.

ropg, 2021-03-06:
Better suited for discussions, not really an issue. See #111

closed 2021-03-06 by ropg