lovyan03/LovyanGFX #106: Feature idea: Hierarchical sprites that pass-thru when not set up ...

ropg, 2021-03-05:
I created [Core2ez](https://core2ez.m5ez.com), a widget-based toolkit for the Core2. It's still very fresh, and it'll need more work. Given that M5Stack is going for your display driver, it might be good if I tell you something about what I have done. I think some of the things that I have implemented would be possibly be cleaner and useful for more people if they migrated to your driver. Also I'd just like to hear what you think...

## The thing

I have a class called `ezDisplayZone`, which inherits from `ezZone`, which just defines a rectangle and you can ask it whether a given `ezPoint` is inside of it, etc etc. `ezDisplayZone` then lets you create a hierarchy of zones that are either directly on the screen, or inside a parent `ezDisplayZone`. `ezDisplayZone` then implements the methods of the display, so `drawRoundRect` is implemented. If you do not set it to buffer in a sprite, all it does is send it to its parent, after subtracting its own position within that parent from the coordinates.

When you set it to buffer, it will draw whatever comes 'from below' and will similarly 'pass up' when pushed, until it meets another sprite-buffered `ezDisplayZone` or the display itself. I implemented this as a shim, but it works remarkable well: an entire widget toolkit runs on it.

## The idea

Could I maybe interest you to have this as part of the display driver instead. A sort of sprite that is pass-through when not active, that stores and corrects for its position in the parent sprite or screen?

I'd be happy to discuss further...

lovyan03, 2021-03-06:
@ropg 
I think this idea is interesting, but I think we need to think carefully about whether it is the job of the display driver or not.
If performance can be sufficiently achieved by providing another class that uses or derives from LovyanGFX or LGFX_Sprite, then I believe that should be the job of that other class.
However, if it is desirable to add new functionality to LovyanGFX or LGFX_Sprite in order to achieve sufficient performance, then I will consider doing so.

By the way, I'm trying to create it as LovyanGUI.
(But the plan is stalled at the moment)

ropg, 2021-03-06:
You are right, it may be the job of an extra class. Please have a look at my first attempt with Core2ez. Let me see if I can adapt what I have to use the new features of LovyanGFZ better.

lovyan03, 2021-03-06:
Sorry, I can't read your intentions.
What do you want me to do?

ropg, 2021-03-06:
I am only showing you what I did. As you say, it probably isn't the job of the display library, but I may come up later with small things that would be better for the display library to handle as I adapt my code to LovyanGFX 1.0.0. Later...

closed 2021-03-06 by ropg