lovyan03/LovyanGFX #108: Error compiling with M5Stack / M5StickCPlus since update 0.3.9

m1cr0lab, 2021-03-06:
Hello @lovyan03,

Thanks for your last update... but since I switched to version 0.3.9, I have compilation problems with the `<M5Stack.h>` and `<M5StickCPlus.h>` libraries:

```
In file included from .pio/libdeps/stick/M5StickCPlus/src/M5Display.h:7:0,
                 from .pio/libdeps/stick/M5StickCPlus/src/M5StickCPlus.h:69,
                 from src/stick.cpp:2:
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:494:18: error: expected unqualified-id before numeric constant
 #define TL_DATUM 0 // Top left (default)
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:168:34: note: in expansion of macro 'TL_DATUM'
     static constexpr textdatum_t TL_DATUM   = textdatum_t::top_left;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:495:18: error: expected unqualified-id before numeric constant
 #define TC_DATUM 1 // Top centre
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:169:34: note: in expansion of macro 'TC_DATUM'
     static constexpr textdatum_t TC_DATUM   = textdatum_t::top_center;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:496:18: error: expected unqualified-id before numeric constant
 #define TR_DATUM 2 // Top right
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:170:34: note: in expansion of macro 'TR_DATUM'
     static constexpr textdatum_t TR_DATUM   = textdatum_t::top_right;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:497:18: error: expected unqualified-id before numeric constant
 #define ML_DATUM 3 // Middle left
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:171:34: note: in expansion of macro 'ML_DATUM'
     static constexpr textdatum_t ML_DATUM   = textdatum_t::middle_left;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:498:18: error: expected unqualified-id before numeric constant
 #define CL_DATUM 3 // Centre left, same as above
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:172:34: note: in expansion of macro 'CL_DATUM'
     static constexpr textdatum_t CL_DATUM   = textdatum_t::middle_left;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:499:18: error: expected unqualified-id before numeric constant
 #define MC_DATUM 4 // Middle centre
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:173:34: note: in expansion of macro 'MC_DATUM'
     static constexpr textdatum_t MC_DATUM   = textdatum_t::middle_center;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:500:18: error: expected unqualified-id before numeric constant
 #define CC_DATUM 4 // Centre centre, same as above
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:174:34: note: in expansion of macro 'CC_DATUM'
     static constexpr textdatum_t CC_DATUM   = textdatum_t::middle_center;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:501:18: error: expected unqualified-id before numeric constant
 #define MR_DATUM 5 // Middle right
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:175:34: note: in expansion of macro 'MR_DATUM'
     static constexpr textdatum_t MR_DATUM   = textdatum_t::middle_right;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:502:18: error: expected unqualified-id before numeric constant
 #define CR_DATUM 5 // Centre right, same as above
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:176:34: note: in expansion of macro 'CR_DATUM'
     static constexpr textdatum_t CR_DATUM   = textdatum_t::middle_right;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:503:18: error: expected unqualified-id before numeric constant
 #define BL_DATUM 6 // Bottom left
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:177:34: note: in expansion of macro 'BL_DATUM'
     static constexpr textdatum_t BL_DATUM   = textdatum_t::bottom_left;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:504:18: error: expected unqualified-id before numeric constant
 #define BC_DATUM 7 // Bottom centre
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:178:34: note: in expansion of macro 'BC_DATUM'
     static constexpr textdatum_t BC_DATUM   = textdatum_t::bottom_center;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:505:18: error: expected unqualified-id before numeric constant
 #define BR_DATUM 8 // Bottom right
                  ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:179:34: note: in expansion of macro 'BR_DATUM'
     static constexpr textdatum_t BR_DATUM   = textdatum_t::bottom_right;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:506:21: error: expected unqualified-id before numeric constant
 #define L_BASELINE  9 // Left character baseline (Line the 'A' character would sit on)
                     ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:180:34: note: in expansion of macro 'L_BASELINE'
     static constexpr textdatum_t L_BASELINE = textdatum_t::baseline_left;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:507:20: error: expected unqualified-id before numeric constant
 #define C_BASELINE 10 // Centre character baseline
                    ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:181:34: note: in expansion of macro 'C_BASELINE'
     static constexpr textdatum_t C_BASELINE = textdatum_t::baseline_center;
                                  ^
.pio/libdeps/stick/M5StickCPlus/src/utility/In_eSPI.h:508:20: error: expected unqualified-id before numeric constant
 #define R_BASELINE 11 // Right character baseline
                    ^
.pio/libdeps/stick/LovyanGFX/src/lgfx/v0/lgfx_common.hpp:182:34: note: in expansion of macro 'R_BASELINE'
     static constexpr textdatum_t R_BASELINE = textdatum_t::baseline_right;
                                  ^
Archiving .pio/build/stick/libFrameworkArduinoVariant.a
Indexing .pio/build/stick/libFrameworkArduinoVariant.a
Compiling .pio/build/stick/FrameworkArduino/Esp.cpp.o
*** [.pio/build/stick/src/stick.cpp.o] Error 1

```

Do you know why?

I don't have these problems with the previous version.

Thank you in advance,
Steph

lovyan03, 2021-03-06:
@m1cr0lab 
Thanks for reporting this. and Sorry for the inconvenience.
I've uploaded a bug-fixed version to the "develop" branch, so please try it out.

m1cr0lab, 2021-03-06:
Sorry for the wait... it took ages to download.

```
git clone --branch develop --single-branch https://github.com/lovyan03/LovyanGFX.git
```

Well, now everything works again 😅

Thanks a lot @lovyan03 🥇

lovyan03, 2021-03-06:
I released v0.3.10 ! thanks to @m1cr0lab .

closed 2021-03-06 by lovyan03