lovyan03/LovyanGFX #109: Does not seem to work on Core2 when co-existing with M5Core2 library ?

ropg, 2021-03-06:
On my Core2, with the master branches of LovyanGFX and m5core2, if I try this:

```c++
#include <M5Core2.h>
#define LGFX_M5STACK_CORE2
#include <LovyanGFX.hpp>

void setup() {
  M5.begin(false);
  static LGFX scr;
  scr.fillCircle(160, 120, 100, TFT_RED); 
}

void loop() {
}
```

I get a boot loop with this:

``` text
PC: 0x400d3ac4: lgfx::LGFX_SPI ::begin_transaction() at /Users/rop/code/Arduino/libraries/LovyanGFX/src/lgfx/platforms/LGFX_SPI_ESP32.hpp line 233
EXCVADDR: 0x00000008

Decoding stack results
0x400d3ac4: lgfx::LGFX_SPI ::begin_transaction() at /Users/rop/code/Arduino/libraries/LovyanGFX/src/lgfx/platforms/LGFX_SPI_ESP32.hpp line 233
0x400d3bc1: lgfx::LGFX_SPI ::beginTransaction_impl() at /Users/rop/code/Arduino/libraries/LovyanGFX/src/lgfx/platforms/LGFX_SPI_ESP32.hpp line 225
0x400db32b: lgfx::LGFXBase::fillCircle(int, int, int) at /Users/rop/code/Arduino/libraries/LovyanGFX/src/lgfx/LGFXBase.hpp line 167
0x400d32e2: setup() at /Users/rop/code/Arduino/libraries/LovyanGFX/src/lgfx/LGFXBase.hpp line 95
0x400deec3: loopTask(void*) at /Users/rop/Library/Arduino15/packages/m5stack/hardware/esp32/1.0.5/cores/esp32/main.cpp line 14
0x4008b619: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

&nbsp;

```c++
#include <M5Core2.h>

void setup() {
  M5.begin();
  M5.Lcd.fillCircle(160, 120, 100, TFT_RED); 
}

void loop() {
}
```
works as expected

lovyan03, 2021-03-06:
need scr.init();
LGFX is uninitialized

closed 2021-03-06 by lovyan03

ropg, 2021-03-06:
Whoops, right...