lovyan03/LovyanGFX #115: Request for GC9A01 support

neofuturism, 2021-03-12:
Hello Lovyan03,
I have two questions for you:
1: What is the display type you have found to be the fastest with your library?
2: I have gotten this [Open Smart Watch](https://open-smartwatch.github.io/) to work with [tft_eSpi ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup200_GC9A01.h) and would love to make it run on yours. The display type is [GC9A01 (pdf datasheet)](https://www.buydisplay.com/download/ic/GC9A01A.pdf), do you think it will work?
Thanks, have a nice day

lovyan03, 2021-03-12:
Hello @neofuturism How have you been?

The fastest one I've tried so far is the ST7789. It can run at 80MHz SPI as long as the connection is good.
Alternatively, an 8-bit parallel connection can be used. This can be used up to 20MHz on most panels.
The parallel 20MHz is faster than the SPI 40MHz. However, there is still room for improvement in the library, and the difference may not be noticeable when compared to SPI 80MHz.

As far as I can tell from the GC9A01 specifications, the command system is very similar to the ILI and ST panels. I think it is possible to support it.

neofuturism, 2021-03-12:
Yeah, the ST7789 are quite fast, I have tried all kind of displays and none come close to the quality and speed.
It's very good news to know that there's a possibility for support for the GC9A01

closed 2021-04-21 by neofuturism

lovyan03, 2021-05-11:
The code that I have in development is already supported, but it will take a few more days before it is released to the public because of structural changes from the current version. Please stay tuned.

lovyan03, 2021-05-30:
@neofuturism 
@RealCorebb 
Hello. Thank you very much for your patience.

I have now uploaded a new version v1 to the develop branch that you can try out, and support for GC9A01 has been added, so you can try it out if you feel like it.

The configuration method is completely new. Please see the example settings and make adjustments.

https://github.com/lovyan03/LovyanGFX/blob/develop/examples/HowToUse/2_user_setting/2_user_setting.ino

RealCorebb, 2021-05-30:
Thank you so much