lovyan03/LovyanGFX #116: Backlight flickers using init_without_reset()

mettyw, 2021-03-20:
Hi,
thanks for this wonderful library!

I am using an M5Stack module with a fairly low backlight setting of 8. I am using deep_sleep and some ULP code to keep the backlight on during sleep.. Whenever the device comes out of deepsleep, I use init_without_reset() which is great to avoid the screen from going blank. However, the screen is initialized with a backlight setting higher than mine which causes the screen to flicker for a split second. 

I think I have traced down and avoided the problem. However, my C is not very good. Maybe you can use the code below to trace down the issue and develop a proper fix.

There seem to be two problems:
1) the panel backlight value is initialized at 128, but `lcd.setBacklight()` can only be called after init
2) `PanelCommon.init()` calls `initPWM()` without the optional duty parameter and therefore duty=128

I "fixed" 1) with repeating the init code in my own class and setting the panel backlight value after `autodetect()`:
```C++
void WeatherDisplay::init_without_reset(std::uint8_t brightness) {
  int retry = 3;
  while (!mylcd.autodetect(false) && --retry);
  mylcd.getPanel()->brightness = brightness;
  //taken from private function mylcd.lgfx::LGFX_SPI<lgfx::LGFX_Config>::init_impl(false);
  mylcd.lgfx::LGFX_SPI<lgfx::LGFX_Config>::initBus(); 
  mylcd.lgfx::LGFX_SPI<lgfx::LGFX_Config>::initPanel(false); 
  mylcd.lgfx::LGFX_SPI<lgfx::LGFX_Config>::initTouch(); 

//  mylcd.init_without_reset();
}
```
I "fixed" 2) by adding the parameter in `PanelCommon.hpp`:
```C++
virtual void init(bool use_reset)
{
  ...
          initPWM(gpio_bl, pwm_ch_bl, pwm_freq, backlight_level ? brightness : (255 - brightness));
  ...
}
```

lovyan03, 2021-03-20:
@mettyw 
Thank you for your report !
I am currently planning a major upgrade of LovyanGFX with a revised basic design.

Your report made me come up with a solution.
I'm going to fix setBrightness so that it can be used before init.

The new version v1.0.0 will be a bit of a wait, but please stay tuned.

tobozo, 2022-10-23:
closing this issue as solved, feel free to reopen if needed

closed 2022-10-23 by tobozo