lovyan03/LovyanGFX #121: Not working on TWatch 2020 V1

CelliesProjects, 2021-05-19:
If I use the library on a TWatch 2020 I get no picture or other reaction from the screen.
It does work with the TTGO drivers, so the HW works.

I get this response on boot:
```
[W][LGFX_Config_AutoDetectESP32.hpp:288] autodetect(): [Autodetect] load from NVS : board:7
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400DA2A4 arg=3FFBDD74
[W][LGFX_Config_AutoDetectESP32.hpp:306] autodetect(): [Autodetect] panel id:00528585
[W][LGFX_Config_AutoDetectESP32.hpp:308] autodetect(): [Autodetect] TWatch
[E][esp32-hal-cpu.c:115] removeApbChangeCallback(): not found func=400DA2A4 arg=3FFBDD74
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400DA2A4 arg=3FFBDD74
```

Anybody else got this issue or am I the only one?

tobozo, 2021-05-19:
hey @CelliesProjects I just tested a display sketch on my own twatch-2020-v1 and got it to work successfully (using the master branch of this project + the esp32-chimera-core layer).

Your logs indicate that it was detected and loaded successfully anyway, so the problem may be from elsewhere e.g. deepsleep with incomplete wakeup sequence or invalid use of AXP?

lovyan03, 2021-06-17:
@CelliesProjects 
It seems to be working fine as far as the logs are concerned,
 but please try running this sample and let me know the results.
If the blue square blinks, then it is working fine.
```
#define LGFX_USE_V1
#define LGFX_AUTODETECT
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

LGFX display;

void setup(void)
{
  display.init();
  display.fillRect(0,0,100,100,TFT_BLUE);
}

void loop(void)
{
  static int count;
  display.setBrightness(++count);
  delay(1);
}
```

closed 2022-06-24 by lovyan03