lovyan03/LovyanGFX #123: [branch:develop] ESP32-Chimera-Core + LGFX v1 observations and ideas

tobozo, 2021-05-30:
Compilation Tests
=====================

`#define LGFX_USE_V1` was added to ESP32-Chimera-Core as instructed.

Tests scope:

  - M5Stack Classic
  - LoLin D32 Pro + ST7735
  - LoLin D32 Pro + ILI9341 + Touch
  - TTGO T-Watch 2020 v1
  - Odroid-Go

Test Sketches: [esp32-doomfire](https://www.youtube.com/watch?v=Kb8k513iJjk), [USB Soft Host](https://www.youtube.com/watch?v=qgxr4optExk), [esp32-sidview](https://www.youtube.com/watch?v=MSlHcBb6tsY)

:warning: some changes to fix compilation errors:

1) Temporary fix for `'Font_Blah' was not declared in this scope`

```C
  // ESP32-Chimera-Core/src/M5Display.h
  static auto Font_Blah = fonts::Font_Blah;
  static auto Font_Blah1 = fonts::Font_Blah1;
```

2) Commenting this out, but skipping tests using SD Card

```C
  // ESP32-Chimera-Core/src/M5Display.cpp
  /*
  if ( lgfx::LGFX_Config::spi_host == HSPI_HOST ) {
    Lcd.setSPIShared(false);
  }
  */
```

-----------------------------

Detection Fail
============

:warning: **Found one device compatibility problem with *LoLin D32 Pro + ST7735 128x128 + no touch***

Panel id `3ff0897c` (ST7735) is ignored when `LGFX_LOLIN_D32_PRO` is used.

<details>
  <summary>lgfx::V1 :broken_heart: (click to expand):</summary>
  <p>

    [W][LGFX_AutoDetect_ESP32.hpp:284] init_impl(): [Autodetect] load from NVS : board:0
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:09 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 3ff0897c <<<< detect ignored
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:70 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:09 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 3ff0897c <<<< detect ignored
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:70 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:09 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 3ff0897c <<<< detect ignored
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:70 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff7f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:09 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 3ff0897c <<<< detect ignored
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = ffffffff
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:70 = ffffff3f
    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 00000000
    [W][LGFX_AutoDetect_ESP32.hpp:361] init_impl(): [Autodetect] save to NVS : board:0

  </p>
</details>

<details>
  <summary>lgfx::V0 :heavy_check_mark: (click to expand):</summary>
  <p>

    [W][LGFX_Config_AutoDetectESP32.hpp:288] autodetect(): [Autodetect] load from NVS : board:0
    [W][LGFX_Config_AutoDetectESP32.hpp:306] autodetect(): [Autodetect] panel id:ffffff7f
    [W][LGFX_Config_AutoDetectESP32.hpp:363] autodetect(): [Autodetect] panel id:ffffff7f
    [W][LGFX_Config_AutoDetectESP32.hpp:424] autodetect(): [Autodetect] panel id:ffffffff
    [W][LGFX_Config_AutoDetectESP32.hpp:466] autodetect(): [Autodetect] panel id:3ff0897c <<<< detect success
    [W][LGFX_Config_AutoDetectESP32.hpp:476] autodetect(): [Autodetect] LoLinD32Pro ST7735
    [W][LGFX_Config_AutoDetectESP32.hpp:1216] autodetect(): [Autodetect] save to NVS : board:12

  </p>
</details>

-----------------------------

Idea: Minimize impact of breaking changes
==========

Extend `gitTagVersion.h` as follows:

```C++

  #define LGFX_VERSION_MAJOR 7
  #define LGFX_VERSION_MINOR 8
  #define LGFX_VERSION_PATCH 9
  #define LOVYANGFX_VERSION F( LGFX_VERSION_MAJOR "." LGFX_VERSION_MINOR "." LGFX_VERSION_PATCH )
  // #define LOVYANGFX_VERSION F("7.8.9") 

```

So any library/sketch can extend compatibility from v0 to v1 by doing this:

```C
#if defined( LGFX_VERSION_MAJOR ) && LGFX_VERSION_MAJOR >= 1
  // newer driver version, assume `LGFX_USE_V1` is enabled
#else    
  // older driver version, use `lgfx::v0` syntax
#endif 

```

lovyan03, 2021-05-30:
@tobozo 
Thanks for the report !
I've made the fix and updated the develop branch as soon as possible.

・fix autodetect for LolinD32Pro with ST7735S
・add "using namespace fonts;"
・modify gitTagVersion.h

```C
  // ESP32-Chimera-Core/src/M5Display.cpp
  /*
  if ( lgfx::LGFX_Config::spi_host == HSPI_HOST ) {
    Lcd.setSPIShared(false);
  }
  */
```
This description is a remnant from the days when we relied on the board manager's selection, which is no longer necessary now that autodetect is available.

tobozo, 2021-05-30:
thanks @lovyan03 the last update solved the ST7735S detection problem with LoLin D32 Pro

    [W][LGFX_AutoDetect_ESP32.hpp:252] _read_panel_id(): [Autodetect] read cmd:04 = 3ff0897c
    [W][LGFX_AutoDetect_ESP32.hpp:898] autodetect(): [Autodetect] LoLinD32Pro ST7735

As expected, SD Card support got broken on Chimera Core:

    [W][sd_diskio.cpp:169] sdCommand(): no token received
    [E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x11
    [W][sd_diskio.cpp:169] sdCommand(): no token received
    [W][sd_diskio.cpp:169] sdCommand(): no token received
    [E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
    [W][sd_diskio.cpp:505] ff_sd_initialize(): GO_IDLE_STATE failed
    [E][vfs_api.cpp:249] VFSFileImpl(): opendir(/sd/) failed

Applied the following fix: do not call `sd_begin()` before `Lcd.begin()`:

```
    // in ESP32-Chimera-Core.cpp
    /*
    if (SDEnable == true) {
      sd_begin();
    }
    */
```

As this may be a breaking change for Chimera Core, my next test scope will be **devices using TFT + SD**.

I'm assuming everything works fine with M5Stack product family so I will focus on ESP32-Wrover-Kit, Odroid-Go and TTGO-Twatch-Base first, please let me know if I forgot a specific device.

closed 2021-06-01 by tobozo