lovyan03/LovyanGFX #127: drawJpgFile doesn't work with datum

kazyam53, 2021-06-06:
I found that drawJpgFile can't display image if I set datum except top_left.
` lcd.drawJpgFile(SD, "/hoge.jpg", 0, 0,0, 0, 0, 0, 1.0, 1.0,top_left);  //Work
  lcd.drawJpgFile(SD, "/hoge.jpg", 0, 0,0, 0, 0, 0, 1.0, 1.0,top_center); //Can't display`

Unitil the commit 48bee0d, both works well but from the next(a6bfce4), it doesn't.
It also doesn't work on M5GFX.

Thank you in advance for your help.

lovyan03, 2021-06-06:
@kazyam53 
This is not a problem, because the criteria cannot be determined without specifying the width and height to be displayed.
Try specifying the desired drawing size in width and height.

```
lcd.drawJpgFile(SD, "/hoge.jpg", 0, 0, lcd.width(), lcd.height(), 0, 0, 1.0, 1.0,top_center); 
```

lovyan03, 2021-06-06:
The M5GFX jpg drawing sample contained a bug, which has been fixed.
Thank you very much .

kazyam53, 2021-06-07:
It worked.
Thank you for your support.

closed 2021-06-07 by kazyam53