lovyan03/LovyanGFX #128: Implementing a image processing filter

don41382, 2021-06-15:
I would like to change the brightness of an image before pushing it to the LCD.

What would be the best way of accomplishing this? It would be nice if the library would support a lambda style processing.

here my pseudo code:
```
auto brightness_up = [](uint8_t r, uint8_t g, uint8_t b) ->  return {min(255, r + 10), min(255, g + 10), min(255, b + 10)}
sprite.pushSprite(lcd, brightness_up);
```

For now: What is the easiest way to implement such processing?

lovyan03, 2021-06-16:
@don41382 
Thanks for your suggestion !
I will proceed with the consideration of implementing it.
Currently there is no easy way to do the process you want. You need to prepare an image with the brightness adjusted beforehand and then push it.

don41382, 2021-06-16:
Thanks @lovyan03! 

Let me now if I can help. This library is really awesome. Thank you for the great work.

Can't wait until image processing could work. That would be the cherry on the top!