lovyan03/LovyanGFX #131: TFT 1.44" (ST7735) blinks with sample code (2_spi_setting)

takijo, 2021-06-21:
Hi,thank you for your great library.
I tried the sample code (2_spi_setting) with TFT 1.44" (ST7735).
But, the TFT blinked.
I tuned some parameters but , cannot solve this problem.
Please advise me if you know anything.
(My setting are below
 TFT_CS 5,TFT_DC 19,TFT_SCK 18,TFT_RST 21,TFT_BL 22,TFT_MOSI 23,FT_MISO -1,panel.freq_write,panel.freq_fill,panel.freq_read=10Mhz,other parameters are default)
[image]

lovyan03, 2021-06-21:
@takijo 
I can't tell anything from that information, so please show me the entire source code.
( You can delete the comment part if you like. )

takijo, 2021-06-21:
@lovyan03 
Thank you for your reply.
My entire source code is below.
I use ESP32.

```
// How to set up the SPI with LovyanGFX
// LovyanGFX SPIバスおよび使用パネルの設定を伴う使い方
#include <LovyanGFX.hpp>

// for ESP32 SPI
struct LGFX_Config
{
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 23;
  static constexpr int spi_miso = -1;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ST7735  panel;

void setup(void)
{
  panel.freq_write = 10000000;
  panel.freq_fill  = 10000000;
  panel.freq_read  = 10000000;
  panel.spi_mode = 0;
  panel.spi_mode_read = 0;
  panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;
  panel.spi_cs = 5;
  panel.spi_dc = 19;
  panel.gpio_rst = 21;
  panel.gpio_bl  = -1;
  panel.pwm_ch_bl = -1;
  panel.backlight_level = true;
  panel.reverse_invert = false;
  panel.rgb_order = false;
  panel.memory_width  = 132;
  panel.memory_height = 132;
  panel.panel_width  = 132;
  panel.panel_height = 132;
  panel.offset_x = 0;
  panel.offset_y = 0;
  panel.rotation = 0;
  panel.offset_rotation = 0;
  lcd.setPanel(&panel);

  lcd.init();

  if (lcd.width() > 240 || lcd.height() > 240) lcd.setTextSize(2);

  if (lcd.touch())
  {
    if (lcd.width() < lcd.height()) lcd.setRotation(3 & (lcd.getRotation() + 1));
    lcd.drawString("touch the arrow marker.", 0, lcd.height()>>1);
    lcd.calibrateTouch(nullptr, 0xFFFFFFU, 0x000000U, 30);
    lcd.clear();
  }
  Serial.begin(115200);
  Serial.println(F("Hello! ST77xx TFT Test"));
}

uint32_t count = ~0;
void loop(void)
{
  delay(10);

  lcd.startWrite();
  lcd.setRotation(++count & 7);
  lcd.setColorDepth((count & 8) ? 16 : 24);
  lcd.setTextColor(random(65536));
  lcd.drawNumber(lcd.getRotation(), 16, 0);
  lcd.setTextColor(0xFF0000U);
  lcd.drawString("R", 30, 16);
  lcd.setTextColor(0x00FF00U);
  lcd.drawString("G", 40, 16);
  lcd.setTextColor(0x0000FFU);
  lcd.drawString("B", 50, 16);
  lcd.drawRect(30,30,lcd.width()-60,lcd.height()-60,random(65536));
  lcd.endWrite();

  int32_t x, y;
  if (lcd.getTouch(&x, &y)) {
    lcd.fillRect(x-2, y-2, 5, 5, random(65536));
  }
}
```

lovyan03, 2021-06-21:
@takijo 
umm, I don't see any particular problem.

I don't know if this will help.
Try to use Panel_ST7735S instead of Panel_ST7735.

takijo, 2021-06-21:
@lovyan03 
Thank you for your check.
OK, I will try another panel.

takijo, 2021-06-22:
@lovyan03 
I have tried Panel_ST7735S,but it didn't work well.
My tft was blinked with  Adafruit library and DMA too.
But,It was improved by changing spi-mode from VSPI to HSPI.
I tried your library with HSPI again, the blinking was stopped.(But, whiteout)
I will continue to tune but I close this issue.(The blinking has stopped.)
Thank you for the quick help.
[image]

closed 2021-06-22 by takijo