lovyan03/LovyanGFX #133: Local Variable LGFX_Sprite Crash

SyotaTanaka, 2021-07-01:
```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x400dccff on core 1
```

Spriteをローカル変数で作成するとクラッシュします。

問題はSpriteBufferのreleaseです。
SpriteにsetBufferでデータを割り当てた後、
releaseが走ると割り当てたデータをheep_freeしてクラッシュしてるようです。
Preallocatedの場合は、何もしないのが正しい様に思えます

lovyan03, 2021-07-01:
@SyotaTanaka 
ご報告いただきありがとうございます！
ご指摘の通り、Preallocated時はheep_freeしないのが正しい動作です。
developブランチにて早速修正を行いました。

lovyan03, 2021-07-12:
本件修正を含めたリリースを公開しました。ありがとうございました！

closed 2021-07-12 by lovyan03