lovyan03/LovyanGFX #134: Request LovyanGFX support for  ESP32-C3-DevKitM-1

BlynkGO, 2021-07-04:
LovyanGFX is the great library for TFT.
And  now Arduino ESP32 core 2.0.0 alpha1 support with ESP32-C3-DevKitM-1.

I have tried to compile LovyanGFX  on  ESP32-C3-DevKitM-1. But it's not worked.
Do you mind to support LovyanGFX  for  ESP32-C3-DevKitM-1 too?
 
[image]

Thank you very much.

lovyan03, 2021-07-04:
@BlynkGO 
Thank you for the request.
I don't have it yet, so I made arrangements.

lovyan03, 2021-07-12:
@BlynkGO 
A new release has been created.
The SPI of ESP32-C3 is now available in the v1 specification.
It does not support DMA transfer yet, but you can try it anyway.

BlynkGO, 2021-07-12:
Thank you very much.

lovyan03, 2021-09-28:
@BlynkGO 
Sorry for the late notice.
Support for SPI bus and I2C bus of ESP32-C3 has been completed.

closed 2021-11-29 by lovyan03

javiser, 2022-01-29:
Great work @lovyan03! I have a ESP32C3 as well and got a basic example with esp-idf / platformio running but only after setting `cfg.dma_channel = 0;`. Just out of curiosity, are there any plans to implement DMA transfer? I will not open an issue for that myself.