lovyan03/LovyanGFX #138: Keep backlight off during deep sleep

tobozo, 2021-07-20:
The problem: ` esp_deep_sleep_start()` re-enables the back light on on some devices (e.g LoLin D32 Pro + ILI9341, using GPIO32 as TFT_BL, default high).

The workaround: use `gpio_hold_en()` and `gpio_deep_sleep_hold_en()` before enabling deep sleep.

```C
  tft.setBrightness(0);
  tft.getPanel()->setSleep(true);

  int16_t pin_bl = ( (lgfx::Light_PWM*) tft.getPanel()->getLight() )->config().pin_bl;
  if( pin_bl > -1 ) {
    gpio_hold_en((gpio_num_t) pin_bl ); // keep TFT_LED off while in deep sleep
    gpio_deep_sleep_hold_en();
  }

  esp_deep_sleep_start();
```

is there a more elegant way to achieve this ?

lovyan03, 2021-07-20:
@tobozo 
During sleep mode, LGFX does not control GPIO, so users basically need to control it by themselves.
Do you want gpio_hold_en to be enabled when setSleep ?

tobozo, 2021-07-20:
> Do you want gpio_hold_en to be enabled when setSleep ?

I'm not sure, maybe the ESP sleep control is better left outside LGFX.

There is no issue or missing feature here.
I can retrieve the `pin_bl` value.
But is there a cleaner way to achieve that ?

```C
    int16_t pin_bl = ( (lgfx::Light_PWM*) tft.getPanel()->getLight() )->config().pin_bl;
```

If you confirm it's the right way, then I can close this issue.

lovyan03, 2021-07-21:
@tobozo 
I see, you need an easy way to get the settings.
Currently the only way to do this is the way you wrote it, but it will crash if the light control is not set.
I would consider implementing a function to get all the settings at once.

tobozo, 2021-07-21:
> but it will crash if the light control is not set.

I see, only LGFX_AUTODETECT has implicit `ILight`.

Will it still crash with a proper check ?

```C

#if defined LGFX_AUTODETECT

  auto backlight = (lgfx::Light_PWM*)tft.getPanel()->getLight();
  if( backlight != nullptr ) {
    auto cfg = backlight->config();
    int16_t pin_bl = cfg.pin_bl;
    if( pin_bl > -1 ) {
      gpio_hold_en((gpio_num_t) pin_bl ); // keep TFT_LED off while in deep sleep
      gpio_deep_sleep_hold_en();
    }
  }

#endif 
```

> I would consider implementing a function to get all the settings at once.

Do you mean from outside LGFX ?

tobozo, 2022-10-23:
closing this issue as solved

closed 2022-10-23 by tobozo