lovyan03/LovyanGFX #140: Support display R61529

ajsb85, 2021-07-25:
Datasheet [R61529.pdf](https://github.com/ZinggJM/GFX_TFT/files/6873653/R61529.pdf)

https://es.aliexpress.com/item/32940608152.html
https://es.aliexpress.com/item/4001050619990.html

modi12jin, 2021-07-27:
[image]

```
https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test
```

@ajsb85

lovyan03, 2021-07-27:
Huh..

I don't want to take care of a child who doesn't even know how to say hello.

closed 2022-02-25 by lovyan03

modi12jin, 2022-03-13:
@lovyan03 Sorry for not saying hello, I was wondering if you have any plans for R61529 support?

> China Taobao store, the information given by the seller.

[LiangHaoCai-3.5.inch.IPS-40p-R61529A-data.package.zip](https://github.com/lovyan03/LovyanGFX/files/8239093/LiangHaoCai-3.5.inch.IPS-40p-R61529A-data.package.zip)

[image]

modi12jin, 2022-03-13:
@lovyan03 R61529 and ILI9481 are compatible and have been tested.

https://user-images.githubusercontent.com/40233017/158057616-cb7336d6-e5eb-4530-bd5a-b55ae32aa35a.mp4

lovyan03, 2022-03-15:
I have purchased the R61529 but have not had time to work on it because of the troublesome handling of the flexible cable.
I will eventually implement it when I have time to work on it, but it is low on my priority list.

modi12jin, 2022-04-10:
@lovyan03 I tried it on ESP32S3，It supports color depth 16 or 18

[image]

```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

   lgfx::Panel_ILI9481 _panel_instance; //R61529兼容ILI9481驱动

   lgfx::Bus_Parallel16 _bus_instance;

   lgfx::Light_PWM _light_instance;

   lgfx::Touch_FT5x06  _touch_instance; 

public:
   LGFX(void)
   {
      {  
         auto cfg = _bus_instance.config();

         cfg.port = 0;  
         cfg.freq_write = 20000000; 
         cfg.pin_wr = 15;           // WR を接続しているピン番号
         cfg.pin_rd = -1;           // RD を接続しているピン番号
         cfg.pin_rs = 16;           // RS(D/C)を接続しているピン番号
         cfg.pin_d0 = 37;            // D0を接続しているピン番号
         cfg.pin_d1 = 36;            // D1を接続しているピン番号
         cfg.pin_d2 = 35;            // D2を接続しているピン番号
         cfg.pin_d3 = 34;            // D3を接続しているピン番号
         cfg.pin_d4 = 33;            // D4を接続しているピン番号
         cfg.pin_d5 = 48;            // D5を接続しているピン番号
         cfg.pin_d6 = 21;            // D6を接続しているピン番号
         cfg.pin_d7 = 18;            // D7を接続しているピン番号
         cfg.pin_d8 = 17;            // D8を接続しているピン番号
         cfg.pin_d9 = 39;            // D9を接続しているピン番号
         cfg.pin_d10 = 38;            // D10を接続しているピン番号
         cfg.pin_d11 = 2;            // D11を接続しているピン番号
         cfg.pin_d12 = 7;            // D12を接続しているピン番号
         cfg.pin_d13 = 12;            // D13を接続しているピン番号
         cfg.pin_d14 = 6;            // D14を接続しているピン番号
         cfg.pin_d15 = 10;            // D15を接続しているピン番号
         
         _bus_instance.config(cfg);              // 設定値をバスに反映します。
         _panel_instance.setBus(&_bus_instance); // バスをパネルにセットします。
      }

      {                                       // 表示パネル制御の設定を行います。
         auto cfg = _panel_instance.config(); // 表示パネル設定用の構造体を取得します。

         cfg.pin_cs = -1;   // CSが接続されているピン番号   (-1 = disable)
         cfg.pin_rst = -1;  // RSTが接続されているピン番号  (-1 = disable)
         cfg.pin_busy = -1; // BUSYが接続されているピン番号 (-1 = disable)

         // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

         cfg.memory_width = 320;   // ドライバICがサポートしている最大の幅
         cfg.memory_height = 480;  // ドライバICがサポートしている最大の高さ
         cfg.panel_width = 320;    // 実際に表示可能な幅
         cfg.panel_height = 480;   // 実際に表示可能な高さ
         cfg.offset_x = 0;         // パネルのX方向オフセット量
         cfg.offset_y = 0;         // パネルのY方向オフセット量
         cfg.offset_rotation = 0;  //值在旋转方向的偏移0~7（4~7是倒置的）
         cfg.dummy_read_pixel = 8; // 在读取像素之前读取的虚拟位数
         cfg.dummy_read_bits = 1;  // 读取像素以外的数据之前的虚拟读取位数
         cfg.readable = false;      // 如果可以读取数据，则设置为 true
         cfg.invert = true;       // 如果面板的明暗反转，则设置为 true
         cfg.rgb_order = false;    // 如果面板的红色和蓝色被交换，则设置为 true
         cfg.dlen_16bit = true;   // 对于以 16 位单位发送数据长度的面板，设置为 true
         cfg.bus_shared = false;    // 如果总线与 SD 卡共享，则设置为 true（使用 drawJpgFile 等执行总线控制）

         _panel_instance.config(cfg);
      }

    { // バックライト制御の設定を行います。（必要なければ削除）
      auto cfg = _light_instance.config();    // バックライト設定用の構造体を取得します。

      cfg.pin_bl = 5;              // バックライトが接続されているピン番号
      cfg.invert = false;           // バックライトの輝度を反転させる場合 true
      cfg.freq   = 44100;           // バックライトのPWM周波数
      cfg.pwm_channel = 1;          // 使用するPWMのチャンネル番号

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);  // バックライトをパネルにセットします。
    }
    
    { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 319;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 479;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = -1;   // INTが接続されているピン番号
      cfg.bus_shared = false; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// I2C接続の場合
      cfg.i2c_port = 0;      // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
      cfg.pin_sda  = 40;     // SDAが接続されているピン番号
      cfg.pin_scl  = 41;     // SCLが接続されているピン番号
      cfg.freq = 400000;     // I2Cクロックを設定

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }

      setPanel(&_panel_instance); // 使用するパネルをセットします。
   }
};

// 準備したクラスのインスタンスを作成します。
LGFX display;

void setup(void)
{
  // SPIバスとパネルの初期化を実行すると使用可能になります。
  display.init();

  display.setTextSize((std::max(display.width(), display.height()) + 255) >> 8);

  // タッチが使用可能な場合のキャリブレーションを行います。（省略可）
  if (display.touch())
  {
    if (display.width() < display.height()) display.setRotation(display.getRotation() ^ 1);

    // 画面に案内文章を描画します。
    display.setTextDatum(textdatum_t::middle_center);
    display.drawString("touch the arrow marker.", display.width()>>1, display.height() >> 1);
    display.setTextDatum(textdatum_t::top_left);

    // タッチを使用する場合、キャリブレーションを行います。画面の四隅に表示される矢印の先端を順にタッチしてください。
    std::uint16_t fg = TFT_WHITE;
    std::uint16_t bg = TFT_BLACK;
    if (display.isEPD()) std::swap(fg, bg);
    display.calibrateTouch(nullptr, fg, bg, std::max(display.width(), display.height()) >> 3);
  }

  display.fillScreen(TFT_BLACK);
}

uint32_t count = ~0;
void loop(void)
{
  display.startWrite();
  display.setRotation(++count & 7);
  display.setColorDepth((count & 8) ? 16 : 24);

  display.setTextColor(TFT_WHITE);
  display.drawNumber(display.getRotation(), 16, 0);

  display.setTextColor(0xFF0000U);
  display.drawString("R", 30, 16);
  display.setTextColor(0x00FF00U);
  display.drawString("G", 40, 16);
  display.setTextColor(0x0000FFU);
  display.drawString("B", 50, 16);

  display.drawRect(30,30,display.width()-60,display.height()-60,count*7);
  display.drawFastHLine(0, 0, 10);

  display.endWrite();

  int32_t x, y;
  if (display.getTouch(&x, &y)) {
    display.fillRect(x-2, y-2, 5, 5, count*7);
  }
}
```

modi12jin, 2022-05-08:
@lovyan03 excuse me! I wonder if LovyanGFX supports hardware swapping of color bytes?

https://github.com/espressif/esp-idf/issues/8847#issuecomment-1111678275

[image]

m-menon, 2022-05-22:
> @lovyan03 R61529 and ILI9481 are compatible and have been tested.
> 
>  6460_1647171512.mp4

hi, can you please share how you wired the ESP32 to R61529? I have tried a few different ways but something just seems to break and the characters are not printed clearly or the whole screen is blue. I might be missing something simple here. So any wiring/ connection info? and the TFT initialization sample code? Thanks.