lovyan03/LovyanGFX #142: compile warnings when using GCC 8.4 with C++17 enabled

estshorter, 2021-08-23:
Following compilation warnings occur when using GCC8.4 with C++17 enabled.
I'm not sure this library supports C++17, but I reported it for your reference.

Note: This will not occur if the C++17 flag is not specified.
```
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp: In member function 'virtual void lgfx::v1::Bus_I2C::release()':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:46:37: warning: ignoring returned value of type 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>', declared with attribute nodiscard [-Wunused-result]
     lgfx::i2c::release(_cfg.i2c_port);
                                     ^
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:157:32: note: in call to 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::release(int)', declared here
     cpp::result<void, error_t> release(int i2c_port);
                                ^~~~~~~
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:22,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:29,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
c:\users\estshorter\src\esp\.pio\libdeps\m5paper\lovyangfx\src\lgfx\utility\result.hpp:2659:26: note: 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>' declared here
   class RESULT_NODISCARD result<void,E>
                          ^~~~~~~~~~~~~~
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp: In member function 'virtual void lgfx::v1::Bus_I2C::beginTransaction()':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:60:46: warning: ignoring returned value of type 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>', declared with attribute nodiscard [-Wunused-result]
       lgfx::i2c::endTransaction(_cfg.i2c_port);
                                              ^
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:160:32: note: in call to 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::endTransaction(int)', declared here
     cpp::result<void, error_t> endTransaction(int i2c_port);
                                ^~~~~~~~~~~~~~
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:22,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:29,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
c:\users\estshorter\src\esp\.pio\libdeps\m5paper\lovyangfx\src\lgfx\utility\result.hpp:2659:26: note: 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>' declared here
   class RESULT_NODISCARD result<void,E>
                          ^~~~~~~~~~~~~~
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:62:85: warning: ignoring returned value of type 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>', declared with attribute nodiscard [-Wunused-result]
     lgfx::i2c::beginTransaction(_cfg.i2c_port, _cfg.i2c_addr, _cfg.freq_write, false);
                                                                                     ^
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:159:32: note: in call to 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::beginTransaction(int, int, uint32_t, bool)', declared here
     cpp::result<void, error_t> beginTransaction(int i2c_port, int i2c_addr, std::uint32_t freq, bool read = false);
                                ^~~~~~~~~~~~~~~~
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:22,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:29,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
c:\users\estshorter\src\esp\.pio\libdeps\m5paper\lovyangfx\src\lgfx\utility\result.hpp:2659:26: note: 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>' declared here
   class RESULT_NODISCARD result<void,E>
                          ^~~~~~~~~~~~~~
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp: In member function 'virtual void lgfx::v1::Bus_I2C::beginRead()':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:75:76: warning: ignoring returned value of type 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>', declared with attribute nodiscard [-Wunused-result]
       lgfx::i2c::restart(_cfg.i2c_port, _cfg.i2c_addr, _cfg.freq_read, true);
                                                                            ^
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:158:32: note: in call to 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::restart(int, int, uint32_t, bool)', declared here
     cpp::result<void, error_t> restart(int i2c_port, int i2c_addr, std::uint32_t freq, bool read = false);
                                ^~~~~~~
In file included from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../esp32/common.hpp:22,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/../common.hpp:29,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp:21,
                 from .pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:22:
c:\users\estshorter\src\esp\.pio\libdeps\m5paper\lovyangfx\src\lgfx\utility\result.hpp:2659:26: note: 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>' declared here
   class RESULT_NODISCARD result<void,E>
                          ^~~~~~~~~~~~~~
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp:79:85: warning: ignoring returned value of type 'cpp::bitwizeshift::result<void, lgfx::v1::error_t>', declared with attribute nodiscard [-Wunused-result]
       lgfx::i2c::beginTransaction(_cfg.i2c_port, _cfg.i2c_addr, _cfg.freq_read, true);
[... 244 lines cut ...]
lib_deps =
  https://github.com/lovyan03/LovyanGFX
```

src/main.cpp
```cpp
#define LGFX_AUTODETECT
#include <LovyanGFX.hpp>

void setup(void)
{
}

void loop(void)
{
}
```
## Environment
- Win10 Pro 21H1 19043.1165
- VSCode 1.59.1
- PlatformIO IDE v2.3.3
- LovyanGFX: 404e56190b950cf28917e4261fa64bb9683fadeb

## Appendix
[buildlog.txt](https://github.com/lovyan03/LovyanGFX/files/7031143/buildloig.txt)

tobozo, 2021-08-23:
hey @estshorter,

Thanks for your feedback. 

Quick workaround: unless you prepend all offending function calls with `(void)!` you can suppress those warnings by adding an extra `-Wno-unused-result` flag:

```ini
build_flags = 
  -std=gnu++17
  -DBOARD_HAS_PSRAM
  -mfix-esp32-psram-cache-issue
  -Wno-unused-result
```

lovyan03, 2021-08-23:
@estshorter 
Thanks for the report !
I've adjusted the code in the develop branch to suppress the warning.

estshorter, 2021-08-23:
Thanks for the quick fix!

But,  the same warning still occurs at the following line.
- https://github.com/lovyan03/LovyanGFX/blob/883af26d698ec511aad1426de5a65e2b60f71e08/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp#L196
- https://github.com/lovyan03/LovyanGFX/blob/883af26d698ec511aad1426de5a65e2b60f71e08/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp#L211
- https://github.com/lovyan03/LovyanGFX/blob/883af26d698ec511aad1426de5a65e2b60f71e08/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp#L218
- https://github.com/lovyan03/LovyanGFX/blob/883af26d698ec511aad1426de5a65e2b60f71e08/src/lgfx/v1/platforms/esp32/Bus_I2C.cpp#L225

estshorter, 2021-08-23:
Note:  `#define LGFX_USE_V1` produces tons of unused-result warnings. 
Adding ` -Wno-unused-result`  may be a good idea as as @tobozo  said.

lovyan03, 2021-08-23:
@estshorter 
Sorry. Readjusted.
I've tweaked the result library and fundamentally killed warnings.

estshorter, 2021-08-23:
Thanks!
I've confirmed that LovyanGFX can be compiled without warnings (even when #define LGFX_USE_V1)!

closed 2021-08-23 by estshorter