lovyan03/LovyanGFX #146: M5 official library's "Ghost Macro" kills std::min etc.

9PEO0xNd, 2021-09-16:
Dear, lovyan03-san,
I'm M5Gray and M5Core2 user.
Since I have updated lovyanGFX ver0.4.2 in my system, your sample code generates error in case M5 official library is included.

I know you and mentors saying twitter comments below:
"SPRESENSEのArduinoの実装、Arduino.hにいきなり鬼畜マクロ定義されてて std::minやらstd::maxが即死なんじゃが…"
"そもそもArduinoの標準としてminとかmaxとか独自実装な件。。。"
Thus, most of users know work around and no one reports issues.

currently I'm doing work around like this:
insert "#undef min" etc. right place in my code .
Or
using brackets (std::min) in your sample code
  zoom = (float)((std::min)(lcd.width(), lcd.height())) / width;
Or
include ESP32-Chimera-Core.h instead.
これら直されますか？Twitter嫌いなのでこちらに投函したのをお許しください。

Best regards,
9PEO0xNd

lovyan03, 2021-09-16:
@9PEO0xNd 
Thanks for reporting this problem !
I've reflected the fix in the develop branch.
If everything goes well, it will be included in the next release.

9PEO0xNd, 2021-09-16:
Dear Lovyan03-san,
Thanks anyways !

closed 2021-09-16 by 9PEO0xNd