lovyan03/LovyanGFX #147: Grayscales in epaper with IT8951 Model: ED078kc2 1872*1404

martinberlin, 2021-09-21:
First I wanted to say that I really how this component works and it was a pleasure to install, just a git submodule add Repo components/LovyanGFX
And after checking the custom settings it was already working.

What I could not understand yet is how the colors are represented for this 16 grayscale epapers. I use another library for parallel that is called EPDiy and there 0 is black and 255 is white.
When I try to do 200x200 squares with incrementing colors from 0 to 70 this is what I get:
[image]

This is using epd_fast mode (in epd_quality works bad for me)
It looks like after color:30 it starts to get darker again. Can someone please explain or show an example on how to draw the 16 grayscales in  one of this epaper?

On another topic: Can be that the VCOM setting, or the Waveform for this model, is different than the one of M5 epaper?

Further info this is the jpeg-render example that I'm trying with my own component:
https://github.com/martinberlin/cale-idf/tree/feature/45-it8951/main/www-jpg-render
[image]

martinberlin, 2021-09-21:
Fantastic. I can't wait to arrive home and try it.
Will make also another photo to show how it works. Thanks a lot!
In the JPG render example link I'm experimenting to download and render a JPEG using differing techniques (Requires an ESP32 with PSRAM since needs a decompress buffer)

martinberlin, 2021-09-21:
Thanks it looks awesome now!
[image]
Your component is awesome!

martinberlin, 2021-09-21:
Reopening since sending this large buffer per SPI I'm getting an wdt alert:

``` 
0x400d76ac: task_wdt_isr at /home/martin/esp/esp-idf/components/esp_common/src/task_wdt.c:187

0x40084d89: _xt_lowint1 at /home/martin/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400dc83d: lgfx::v1::Bus_SPI::writeData(unsigned int, unsigned int) at /home/martin/esp/projects/cale-idf/components/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:196

0x400db87a: lgfx::v1::Panel_IT8951::_write_args(unsigned short, unsigned short*, int) at /home/martin/esp/projects/cale-idf/components/LovyanGFX/src/lgfx/v1/panel/Panel_IT8951.cpp:257
``` 

Will try to solve it adding a small VTaskDelay and feeding the watchdog.

lovyan03, 2021-09-21:
@martinberlin 
I looked at your program. I see that drawing with drawPixel must be slow.
This should be all you need to do in the function tjd_output.
```
  uint32_t w = rect->right - rect->left + 1;
  uint32_t h = rect->bottom - rect->top + 1;
  display.pushImage(rect->left, rect->top, w, h, (lgfx::bgr888_t*)bitmap);
```
Since the pushImage function correctly handles the case where the image extends beyond the screen, there is no need for clipping in advance.

If you need to do the gamma correction in advance, you can prepare a uint8_t array and cast it to the lgfx::grayscale_t* type for drawing.

```
  uint32_t w = rect->right - rect->left + 1;
  uint32_t h = rect->bottom - rect->top + 1;
  uint8_t buf[w*h];
  for (uint32_t i = 0; i < w * h; ++i)
  {
    uint8_t r = *(bitmap_ptr++);
    uint8_t g = *(bitmap_ptr++);
    uint8_t b = *(bitmap_ptr++);
    buf[i] = (r*38 + g*75 + b*15)>>7;
  }
  display.pushImage(rect->left, rect->top, w, h, (lgfx::grayscale_t*)buf);
```

martinberlin, 2021-09-22:
Awesome will try it today. It’s possible also to add dithering so the grayscale transitions are not that visible?
Anyways looks awesome and works great. So glad to find your component!
[image]
[image]

lovyan03, 2021-09-22:
@martinberlin 
A big thank you for your sponsorship !!

Come to think of it, you first said "This is using epd_fast mode (in epd_quality works bad for me)"
The epd_fast and epd_fastest modes allow the EPD to operate in monochrome binary.
and, LovyanGFX uses a 4×4 16pixel tile pattern to display a pseudo 17level grayscale.

If the epd_quality mode works, it will make the EPD work with 16 levels of grayscale.
and, LovyanGFX can use a 4×4 16pixel tile pattern to display a pseudo 255level grayscale.

IT8951 should be able to use epd_quality mode, but what problems are you experiencing?

martinberlin, 2021-09-22:
No problem, a pleasure to support you. Wow! Now I see so that pool picture is rendered using only monochrome tile pattern. 
Only zooming in I can see it, I would have bet it was using the 16 grayscales :)
So I will try today again. At the beginning it was off for me, like if the Waveshare of this big EPD would be different from the M5 epaper. But I had also the wrong grayscale calculation. 
I will find sometime today to try this again and report back to you. Also with some statistics of using drawPixel vs pushImage.

martinberlin, 2021-09-22:
Thanks for all your support. Confirmed that your componentworks without problems in bigger epapers driven by IT8951
Quality mode
[image]

And also fast mode works great.
[image]

martinberlin, 2022-05-19:
My bad I was using an image that had the wrong bit-depth :)
I will create a new Issue for my tests with ESP32S3 and new Goodisplay adapter

martinberlin, 2022-05-19:
After fighting with this quite long, I realize that is not Lovyan update that is making the picture to render strange, but rather that I'm testing now with a more modern IDF:
ESP-IDF v4.4

All the images look exactly like shown [here](https://github.com/lovyan03/LovyanGFX/issues/147#issuecomment-1131731863).
Sorry to bother you @lovyan03 but can you give me some advice?

I'm also designing a new PCB for Goodisplay ITE 9.7 (1800*825) [IT8951 controller](https://www.good-display.com/product/425.html) and I would really like to use your component to build my Firmware examples.
This is based on ESP32S3 and I'm still testing it since it seems a bit different than the Waveshare model.

lovyan03, 2022-05-20:
@martinberlin 
Sorry, the IT8951 I have is only M5Paper made by M5Stack, so I may not be able to reproduce your situation there.

I see you are using tjpgd provided by ESP-IDF instead of drawJpg from LovyanGFX.
Maybe something has changed with the upgrade of ESP-IDF.
For comparison, try using LovyanGFX's drawJpg and see if the image is displayed correctly.

martinberlin, 2022-05-20:
No problem, will try. Closing this since it works just like before, the only problem seems to be the IDF version >4.4

closed 2022-05-20 by martinberlin