lovyan03/LovyanGFX #150: Stange Print issue on M5Paper

chrwh, 2021-10-02:
Hi,

I get a sporadic print issue on M5Paper (see pic) when I debug.print, the value is always correct, any suggestions?

Here is the error
[image]

Here ot's OK
[image]

lovyan03, 2021-10-02:
@chrwh 
This will happen if you try to print characters that are not included in the font data.

lovyan03, 2021-10-02:
@chrwh 
I want you to check the contents of the city.

ESP_LOGD("DEBUG","%s", city);
gfx.print(city);
ESP_LOGD("DEBUG","%s", city);

chrwh, 2021-10-02:
Hi,
OK, I can understand that, but I simply don't understand where the coruption is coming from

Here is a debug print

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@hx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@dx@0␆
[
[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regstrup

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop():

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@�+@0␋␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop():

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@Yx@0␆
[
[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@K+@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@Yx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@Yx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): 

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@rx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@:E@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop():

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop():

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@mx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop():

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@5E@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regstrup

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@rx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@5E@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@Yx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@dx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@rx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@5E@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@dx@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regstrup

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): 

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regstrup

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): 

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@5E@0␆

[D][main.cpp:686] loop(): Regstrup
[D][main.cpp:688] loop(): Regst@zx@0␆

lovyan03, 2021-10-02:
@chrwh 
It seems that memory corruption is occurring for some reason.
I want to see the entire source code.

chrwh, 2021-10-02:
OK, thanks it's an VSCode Platform IO project, here is the source code files, let me know if you need anything else
[src.zip](https://github.com/lovyan03/LovyanGFX/files/7272309/src.zip)

lovyan03, 2021-10-02:
@chrwh 
I expect that the city's data is being placed in places that do not have proper memory allocation.

 line 262 of main.cpp
```
  StaticJsonDocument<1024>jsonBuffer;
```

Is the amount of this buffer sufficient?
Wouldn't changing this to 1280, for example, solve the problem?

lovyan03, 2021-10-02:
@chrwh 
I don't have an API key, so I haven't seen any data to pass to ArduinoJSON.
Can You put the string Your received from the web server here?

chrwh, 2021-10-02:
Don't think that is the issue, the returned JSON is approx. 470b give and take. If I debug print the JSON, it's always ok, and not corrupted. I do ow ever belive, that it's some sort of memory issue.

{"coord":{"lon":11.6185,"lat":55.6736},"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"base":"stations","main":{"temp":286.52,"feels_like":286.34,"temp_min":285.83,"temp_max":288.06,"pressure":1005,"humidity":93,"sea_level":1005,"grnd_level":1003},"visibility":10000,"wind":{"speed":7.04,"deg":139,"gust":14.78},"clouds":{"all":100},"dt":1633213693,"sys":{"type":2,"id":2009792,"country":"DK","sunrise":1633238410,"sunset":1633279482},"timezone":7200,"id":2620141,"name":"Regstrup","cod":200}

lovyan03, 2021-10-02:
@chrwh 
I found the cause.

The jsonBuffer variable is only valid in MakehttpRequest.
Since the data retrieved from Json is a pointer and does not have any substance, it can be destroyed outside of MakehttpRequest.
The easiest solution is to put the jsonBuffer outside the function MakehttpRequest.

Lastly, I am not your debug guy.

chrwh, 2021-10-02:
@lovyan03 Thank you so much for taking your time, and for helping me. I'm well aware, that your not my debug guy, and I want to thank you for this great lib. and all the other work you have done. I did however believe, that it was some sort of bug, but I happy, that it was me and not your great lib, that was the issue.
Thanks and best regards,
Christian

closed 2021-10-02 by chrwh