lovyan03/LovyanGFX #152: MakerFabs ESP32 Touch Camera with touch-screen FT6236 & LovyanGFX

bmedici, 2021-10-03:
Hi,

I'm currently using this lib and trying to init the touch device (based on FT6236) with LGFX as it seems to be supported in v1.
The problem is my touch device is using i2c, and neither with v0 nor v1, I'm able to initialize i2c parameters (SDA, SCL).

```
#include <LovyanGFX.hpp>

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Touch_FT5x06 touch;
```

or

```
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

static LGFX lcd;
static lgfx::Touch_FT5x06 touch;
```
can't be configured by :
```
touch.i2c_port = 1;      // 使用するI2Cを選択 (0 or 1)
touch.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
touch.pin_sda  = 23;     // SDAが接続されているピン番号
touch.pin_scl  = 32;     // SCLが接続されているピン番号
```
both will lead to 
```
lgfx::v1::Touch_FT5x06' has no member named 'i2c_port'
```

What is the correct way to use ESP32_TSC_9488_LCD_SPI_HOST and FT6236 touch through this lib, in order to have the synchronised rotate value and translations ?

Thanks for any help on this problem

Bruno

lovyan03, 2021-10-03:
@bmedici 
hello.
It is not a member of the touch structure.
Please use the config function of the touch structure to get the config value.

```
auto cfg = touch.config();
cfg.i2c_port = 1;
cfg.i2c_addr = 0x38;
cfg.pin_sda  = 23;
cfg.pin_scl  = 32;  
touch.config(cfg);
```

bmedici, 2021-10-05:
Hi,

Thank you for that so quick response! Though this projects looks really promising, I find it lacks some clear documentation, such as activating v0 or v1, the differences, usage (doc instead of juste examples).

I've managed to enable touch functions with the display, but found that it stops working when I activate basic ESP NOW functionnality, based on Wifi STA or AP_STA modes.

You'll find below a working example that easily show the problem, just by uncommenting ````#define ENABLE_ESPNOW````at top.

````
// External libs
#include <Arduino.h>

#include <WiFi.h>
#include <esp_now.h>

#define LGFX_USE_V1
#define LGFX_MAKERFABS_TOUCHCAMERA
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>
#include "makerfabs_pin.h"

// #define ENABLE_ESPNOW

// LGFX stuff
static LGFX lcd;
static LGFX_Sprite sprite(&lcd);
static lgfx::Panel_ILI948x panel;
static lgfx::Touch_FT5x06 touch;

void setup() {
  Serial.begin(115200);

  // Wifi setup
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  // ESP NOW init  
  #ifdef ENABLE_ESPNOW
    if (esp_now_init() == ESP_OK) {
      Serial.println("ESP NOW: OK");
    }
    else {
      Serial.println("ESP NOW: FAILED");
      ESP.restart();
    }
  #endif

  // Display init
  auto cfg = touch.config();
  // cfg.pin_sda = MP_ESP32_I2C_SDA;
  // cfg.pin_scl = MP_ESP32_I2C_SCL;
  // cfg.i2c_addr = MP_ESP32_SSD1306_I2C_ADDR;
  touch.config(cfg);
  panel.touch(&touch);
  lcd.setPanel(&panel);

  // Sample drawing
  lcd.init();
  lcd.setRotation(3);
  lcd.setBrightness(128);
  lcd.setColorDepth(24);

  lcd.drawFastVLine(2, 0, 100, lcd.color888(255,   0,   0));
  lcd.drawFastVLine(4, 0, 100, lcd.color565(  0, 255,   0));
  lcd.drawFastVLine(6, 0, 100, lcd.color332(  0,   0, 255));

  lcd.startWrite();
}

void loop() {
  int32_t x, y;    
  if (lcd.getTouch(&x, &y)) {
    Serial.printf("x,y = %d, %d\n", x,y);
    lcd.fillRect(x-2, y-2, 5, 5, random(65536));
  }
}

````

Any idea why I could not have touch functions while using Wifi on an ESP32 ?
Thanks agin for any suggestion !

lovyan03, 2021-10-05:
@bmedici 
LGFX_AUTODETECT is not supported by your board.
Also,  "lgfx::Panel_ILI948x" should not be used.
Use one of the following instead.
lgfx::Panel_ILI9481
lgfx::Panel_ILI9486
lgfx::Panel_ILI9488

Please refer to the following for setting instructions.
 src/lgfx_user/LGFX_ESP32_sample.hpp
 examples/HowToUse/2_user_setting/2_user_setting.ino

If you cannot read Japanese, please refer to this video.
https://www.youtube.com/watch?v=IPCvQ4o_WP8

lovyan03, 2021-10-05:
One potentially related bug has been fixed.

closed 2022-06-24 by lovyan03

SoftechEngineering, 2024-11-01:
@lovyan03 i have a question. i am using WT32 SC01 (MCU 8080 / 8 Bit Parallel lines for data, I2C for touch). 
WT32 SC01 SPECS: 320x 480
Display IC: ST7796U
Touch IC:   FT6336U

So whenever i uploaded the code with correct LVGX class, my display work properly, but touch doesn't wok. anyone can help me out with this?? Secondly, i am designing the UI from Square line studio. so help me out with this