lovyan03/LovyanGFX #153: build failure against the latest arduino-esp32

estshorter, 2021-10-09:
From commit https://github.com/espressif/arduino-esp32/commit/f87107dedb6aff148622672376ae6a4aefdfd80c in arduino-esp32, `TwoWire::readTransmission` and `TwoWire::writeTransmission` are removed and this causes build failure in LovyanGFX.

The following is a build log.

```
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp: In function 'bool lgfx::v0::i2c::writeBytes(int, uint16_t, const uint8_t*, uint8_t)':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:379:27: error: 'class TwoWire' has no member named 'writeTransmission'; did you mean 'beginTransmission'?
       return 0 == twowire.writeTransmission(addr, const_cast<uint8_t*>(data), len);
                           ^~~~~~~~~~~~~~~~~
                           beginTransmission
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp: In function 'bool lgfx::v0::i2c::writeReadBytes(int, uint16_t, const uint8_t*, uint8_t, uint8_t*, uint8_t)':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:399:24: error: 'class TwoWire' has no member named 'writeTransmission'; did you mean 'beginTransmission'?
       if (0 != twowire.writeTransmission(addr, const_cast<uint8_t*>(writedata), writelen)) return false;
                        ^~~~~~~~~~~~~~~~~
                        beginTransmission
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:400:28: error: 'class TwoWire' has no member named 'readTransmission'; did you mean 'endTransmission'?
       return (0 == twowire.readTransmission(addr, readdata, readlen));
                            ^~~~~~~~~~~~~~~~
                            endTransmission
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp: In function 'bool lgfx::v0::i2c::readRegister(int, uint16_t, uint8_t, uint8_t*, uint8_t)':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:423:24: error: 'class TwoWire' has no member named 'writeTransmission'; did you mean 'beginTransmission'?
       if (0 != twowire.writeTransmission(addr, tmp, 1)) return false;
                        ^~~~~~~~~~~~~~~~~
                        beginTransmission
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:424:28: error: 'class TwoWire' has no member named 'readTransmission'; did you mean 'endTransmission'?
       return (0 == twowire.readTransmission(addr, data, len));
                            ^~~~~~~~~~~~~~~~
                            endTransmission
*** [.pio\build\m5paper\libc1d\LovyanGFX\lgfx\v0\platforms\esp32_common.cpp.o] Error 1
```

lovyan03, 2021-10-09:
@estshorter 
Thank you for reporting this problem !
I fixed it in the develop branch. Please check it out.

estshorter, 2021-10-09:
It seems that `I2C_ERROR_OK` is not declared.  Could you look into this again?

``` 
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp: In function 'bool lgfx::v0::i2c::writeBytes(int, uint16_t, const uint8_t*, uint8_t)':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:382:14: error: 'I2C_ERROR_OK' was not declared in this scope
       return I2C_ERROR_OK == twowire.endTransmission();
              ^~~~~~~~~~~~
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp: In function 'bool lgfx::v0::i2c::writeReadBytes(int, uint16_t, const uint8_t*, uint8_t, uint8_t*, uint8_t)':
.pio/libdeps/m5paper/LovyanGFX/src/lgfx/v0/platforms/esp32_common.cpp:404:11: error: 'I2C_ERROR_OK' was not declared in this scope
       if (I2C_ERROR_OK != twowire.endTransmission(false))
           ^~~~~~~~~~~~
Compiling .pio\build\m5paper\libc1d\LovyanGFX\lgfx\v1\panel\Panel_Device.cpp.o
*** [.pio\build\m5paper\libc1d\LovyanGFX\lgfx\v0\platforms\esp32_common.cpp.o] Error 1
```

lovyan03, 2021-10-09:
@estshorter 
I'm sorry.
I was missing the include. I was able to fix it, so please try again.

lovyan03, 2021-10-09:
@estshorter 
sorry again...
I gave up on using the defined value and replaced it with 0.

estshorter, 2021-10-09:
Thank you for the fix!
Now I can build flawlessly!

closed 2021-10-09 by estshorter