lovyan03/LovyanGFX #159: How to config for TTGO T-Display ST7789

patfelst, 2021-11-04:
Hi many thanks for this great library. I have a [TTGO T_Display](http://www.lilygo.cn/prod_view.aspx?TypeId=50044&Id=1126&FId=t3:50044:3) and am unsure how to configure it. Is there some trick with these #defines?

```C++
#define LGFX_TTGO_TS // TTGO TS
#define LGFX_AUTODETECT
```

or should I be using the method shown in [this issue](https://github.com/lovyan03/LovyanGFX/issues/99)? When I try this code, all I get is random dots like static on an untuned TV screen. I've tried changing the SPI_mode betwen 0-3 and also SPI_mode_read 0-3 with no luck. I have tried the commented out section starting at `panel.freq_write` too with no luck.

I'm wondering if I have a TTGO with a different display type, for example the ST7735 has RED_TAB and GREEN_TAB types. I note the TFT_eSPI library has something called `#define CGRAM_OFFSET      // Library will add offsets required` but I don't know what the equivalent is in LovyanGFX.

Here is my code:

```C++
#include <Arduino.h>
#include "Wire.h"
#include "Adafruit_NeoPixel.h"
#include <LovyanGFX.hpp>

// #define LGFX_TTGO_TS // TTGO TS
// #define LGFX_AUTODETECT
// #define LGFX_USE_V1

#define LED_COUNT 16
#define LED_PIN 27

Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRBW + NEO_KHZ800);

struct LGFX_Config {
  static constexpr spi_host_device_t spi_host = VSPI_HOST;
  static constexpr int dma_channel = 1;
  static constexpr int spi_miso = -1;
  static constexpr int spi_sclk = 18;
  static constexpr int spi_mosi = 19;
  static constexpr int spi_dlen = 8;
};

static lgfx::LGFX_SPI<LGFX_Config> lcd;
static lgfx::Panel_ST7789 panel;

void setup() {
  strip.begin(); // Start the RGBW LED ring

  // panel.freq_write = 20000000;
  // panel.freq_fill = 40000000; //27000000;
  // panel.freq_read = 6000000;  //16000000;
  // panel.spi_mode = 0;
  // panel.spi_mode_read = 0;
  // panel.len_dummy_read_pixel = 8;
  panel.spi_read = true;
  panel.spi_3wire = false;

  panel.spi_cs = 5;
  panel.spi_dc = 16;
  panel.gpio_rst = 23;

  panel.gpio_bl = 4;

  panel.pwm_ch_bl = 7;
  panel.backlight_level = true;
  panel.reverse_invert = false;
  panel.rgb_order = false;

  panel.memory_width = 240;
  panel.memory_height = 135;
  panel.panel_width = 240;
  panel.panel_height = 135;

  panel.offset_x = 0;
  panel.offset_y = 0;

  panel.rotation = 0;
  panel.offset_rotation = 0;

  lcd.setPanel(&panel);
  lcd.init();
  // lcd.setBrightness(128);

  lcd.setFont(&fonts::FreeSerif9pt7b);
  lcd.setTextDatum(top_left);
  lcd.setTextColor(TFT_GREEN);
  lcd.drawString("CO2 monitor", 10, 10);
}
```

Thanks for any suggestions I can try.

tobozo, 2021-11-04:
Hi, 

> CGRAM_OFFSET

this is probably what you're looking for:

```C
  panel.offset_x = 0;
  panel.offset_y = 0;
```

lovyan03, 2021-11-04:
@patfelst 
Thank you for your interest.
I have prepared an example configuration, which you can use.

```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

// LGFX for TTGO T-Display
class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ST7789  _panel_instance;
  lgfx::Bus_SPI       _bus_instance;
  lgfx::Light_PWM     _light_instance;

public:

  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();

      cfg.spi_host = VSPI_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 40000000;
      cfg.freq_read  = 14000000;
      cfg.spi_3wire  = true;
      cfg.use_lock   = true;
      cfg.dma_channel = 1;
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 19;
      cfg.pin_miso = -1;
      cfg.pin_dc   = 16;

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _light_instance.config();

      cfg.pin_bl = 4;
      cfg.invert = false;
      cfg.freq   = 44100;
      cfg.pwm_channel = 7;

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);
    }

    {
      auto cfg = _panel_instance.config();

      cfg.pin_cs           =     5;
      cfg.pin_rst          =    23;
      cfg.pin_busy         =    -1;

      cfg.panel_width      =   135;
      cfg.panel_height     =   240;
      cfg.offset_x         =    52;
      cfg.offset_y         =    40;
      cfg.offset_rotation  =     0;
      cfg.dummy_read_pixel =    16;
      cfg.dummy_read_bits  =     1;
      cfg.readable         =  true;
      cfg.invert           =  true;
      cfg.rgb_order        = false;
      cfg.dlen_16bit       = false;
      cfg.bus_shared       =  true;

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance);
  }
};

LGFX lcd;

void setup(void)
{
  lcd.init();
}

void loop(void)
{
  lcd.print("hello T-Display !");
}
```

patfelst, 2021-11-04:
@lovyan03 - success, that works, thanks for your help!

While this works, I must admit OOP with new classes and inheritence are not my strong point. Is there a simpler method to use, or can you briefly describe what your initialisation code does?

lovyan03, 2021-11-04:
@patfelst 

You can also create an instance of LGFX_Device and configure it without using inheritance.
```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

lgfx::LGFX_Device   lcd;
lgfx::Bus_SPI       bus_instance;
lgfx::Light_PWM     light_instance;
lgfx::Panel_ST7789  panel_instance;

void setup(void)
{
  {
    auto cfg = bus_instance.config();

    cfg.spi_host = VSPI_HOST;
    cfg.spi_mode = 0;
    cfg.freq_write = 40000000;
    cfg.freq_read  = 14000000;
    cfg.spi_3wire  = true;
    cfg.use_lock   = true;
    cfg.dma_channel = 1;
    cfg.pin_sclk = 18;
    cfg.pin_mosi = 19;
    cfg.pin_miso = -1;
    cfg.pin_dc   = 16;

    bus_instance.config(cfg);
  }

  {
    auto cfg = light_instance.config();

    cfg.pin_bl = 4;
    cfg.invert = false;
    cfg.freq   = 44100;
    cfg.pwm_channel = 7;

    light_instance.config(cfg);
  }

  {
    auto cfg = panel_instance.config();

    cfg.pin_cs           =     5;
    cfg.pin_rst          =    23;
    cfg.pin_busy         =    -1;

    cfg.panel_width      =   135;
    cfg.panel_height     =   240;
    cfg.offset_x         =    52;
    cfg.offset_y         =    40;
    cfg.offset_rotation  =     0;
    cfg.dummy_read_pixel =    16;
    cfg.dummy_read_bits  =     1;
    cfg.readable         =  true;
    cfg.invert           =  true;
    cfg.rgb_order        = false;
    cfg.dlen_16bit       = false;
    cfg.bus_shared       =  true;

    panel_instance.config(cfg);
  }

  panel_instance.setBus(&bus_instance);
  panel_instance.setLight(&light_instance);
  lcd.setPanel(&panel_instance);

  lcd.init();
}

void loop(void)
{
  lcd.print("hello T-Display !");
}
```

The reason I recommend the example of using inheritance is that it allows you to make the configuration part a separate header file.
It's a matter of preference, so choose whatever method you feel comfortable doing.

lovyan03, 2021-11-04:
@patfelst 
The advantage of making the configuration part independent as a header file of the LGFX class is the following two points.

- The description of the configuration part can be shared among multiple projects.
- It is easy to replace the settings of multiple boards.

The user code to use it is simple and easy to change as follows
```
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

   #include "LGFX_TTGO_TDisplay.hpp"
// #include "LGFX_TTGO_TWatch.hpp"
// #include "LGFX_MY_DEVICE.hpp"

LGFX lcd;

void setup(void)
{
  lcd.init();

  ...
```

patfelst, 2021-11-05:
Thanks very much for the explanations, that's great. I like the idea of the new class in a separate header file. I will implement that I think.

closed 2021-11-05 by patfelst