lovyan03/LovyanGFX #166: Add support for ESP8266 PROGMEM support

fsender, 2021-11-16:
When I using unicode fonts for ESP8266,
the font data cost me RAM instead of flash.
```
Linking .pio\build\nodemcuv2\firmware.elf
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\firmware.elf section `.rodata' will not fit in region `dram0_0_seg'
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40015cd0 of .pio\build\nodemcuv2\firmware.elf section `.bss' is not within region `dram0_0_seg'
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40015cd0 of .pio\build\nodemcuv2\firmware.elf section `.bss' is not within region `dram0_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1
```
I hope my ESP8266 can store font data in PROGMEM.

lovyan03, 2021-11-16:
@fsender 
What is "Unicode Fonts" ?
Please provide the code that reproduces the problem.

fsender, 2021-11-16:
Oh, I meant a large font that cannot be stored in the ESP8266 RAM.
Platform: PlatformIO
Sketch: examples/Standard/LongTextScroll/LongTextScroll.ino
```
> Executing task: C:\Users\1\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LovyanGFX> 0.4.7
|   |-- <Time> 1.6
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Linking .pio\build\nodemcuv2\firmware.elf
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\firmware.elf section `.rodata' will not fit in region `dram0_0_seg'
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40050130 of .pio\build\nodemcuv2\firmware.elf section `.bss' is not within region `dram0_0_seg'
c:/users/1/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40050130 of .pio\build\nodemcuv2\firmware.elf section `.bss' is not within region `dram0_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1
================================================================= [FAILED] Took 7.11 seconds =================================================================
终端进程“C:\Users\1\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'”已终止，退出代码: 1。
```

lovyan03, 2021-11-17:
@fsender 
I have made a fix in the develop branch, so please try it out.

fsender, 2021-11-20:
Thanks for your bugfix!
New fonts work well.
In fact, small-capacity fonts should be placed in PROGMEM instead of RAM to save the RAM usage.
Large constant data such as bitmaps data should be put into PROGMEM and be able to read by the module, too.

lovyan03, 2021-11-21:
@fsender 
To be honest, I never intended to support the ESP8266.
Initially, I only supported ESP32, and I eradicated PROGMEM from the source code in the beginning.
So the current source code was forced to support the ESP8266 later on.
I don't usually use the ESP8266, so the support may be incomplete.
If you find any problems, please let me know.

fsender, 2021-11-21:
But on the ESP8266 platform, your library is still the best library.
Whether it is ESP8266 or ESP32 series, it is better, faster, and has more functions than TFT_eSPI .
Nowadays, most of my projects have completely abandoned ESPI and all use your library.
Hope you can always make it better and implement more functions on ESP8266.
I will also use your library to test the extreme performance of ESP8266. 
I ran a PDQ test with LovyanGFX and TFT_eSPI on my ESP8266 and ST7789 320x240 IPS display. Most benchmarks were faster than TFT_eSPI library except "Pixels".

fsender, 2022-03-03:
Now I'm trying to push a bitmap from my ESP8266 chip.
I found this line below:
```
          if (!(i & 7)) byte = bitmap[i >> 3];
```
In my opinion, this line need fixing to below:
```
          if (!(i & 7)) 
#ifdef ESP8266
          byte = pgm_read_byte(bitmap+(i >> 3));
#else
          byte = *(bitmap+(i >> 3));
#endif
```

Then I can still use my ESP8266 PROGMEM to save my bitmap data.
Could you find more arrays to use my ESP8266 PROGMEM?

lovyan03, 2022-03-03:
Thank you for your report !

fsender, 2022-03-07:
I think you could use some macros like '#ifdef ESP8266...#else...#endif'.
For ESP32 and other chips, I can use the old methods.

lovyan03, 2022-03-09:
https://github.com/lovyan03/LovyanGFX/blob/0.4.14/src/lgfx/utility/pgmspace.h

Since LovyanGFX defines pgmspace in this header file.
Perhaps I don't need to write processor-specific ifdef branches in each source code.

fsender, 2022-03-10:
You can view line 1019 and line 1049 in file 'lgfx/v1/LGFXBase.cpp'
```
if (!(i & 7)) byte = bitmap[i >> 3];
```
These code needed to be replaced to 'pgm_read_xxx'.
Don't forget to add __header file__ in file 'LGFXBase.cpp'.
```
#include "../utility/pgmspace.h"
...
if (!(i & 7)) byte = pgm_read_byte(bitmap+(i >> 3));
```

lovyan03, 2022-03-10:
…？
I have recently modified the develop branch after your suggestion.

https://github.com/lovyan03/LovyanGFX/commit/64041b6d473abaca8a262165ab133639735c949b

It will be a while before it is reflected in the master branch.

closed 2022-06-17 by fsender