lovyan03/LovyanGFX #173: LGFX + ESP-IDF + VSCode

Yelgurk, 2021-12-02:
Thanks for the great library for working with graphics. Already had time to work in ArduinoIDE environment, but when switching to VSCode + ESP-IDF bundle, there is a problem to connect the library for further development. Maybe you have some manual?

lovyan03, 2021-12-03:
@Yelgurk

I am able to use LovyanGFX with VSCode+ESP-IDF without any problems.

you need `REQUIRES LovyanGFX` in CMakeLists.txt

CMakeLists.txt
```
idf_component_register(SRCS "main.cpp"
                    INCLUDE_DIRS "."
                    REQUIRES LovyanGFX)
```

main.cpp
```
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "sdkconfig.h"

#define LGFX_USE_V1
#define LGFX_AUTODETECT
#include <LovyanGFX.h>
#include <LGFX_AUTODETECT.hpp>

LGFX lcd;

extern "C"
{
  void app_main(void)
  {
    lcd.init();
    while(1) {
      vTaskDelay(1);
      lcd.print("TEST");
    }
  }
}
```

Yelgurk, 2021-12-03:
Thanks for the setup code! I guess I'm one step closer to solving the problem of starting development. However, the following error occurred during the build.

Terminal:
```
 ...\esp-idf\components\lovyangfx\src\lgfx\internal\memory.h:13:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.
[1110/1117] Building C object esp-idf/u8g2/CMakeFiles/__idf_u8g2.dir/csrc/u8g2_d_setup.c.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\Windows\System32\cmd.exe /d /c cmake --build ." terminated with exit code: 1.
```

Can you tell me what the problem may be and how it can be solved?

lovyan03, 2021-12-03:
@Yelgurk 
LovyanGFX is a C++ library. (NOT c library)
In other words, it cannot be used from a *.c file.
Please use it from .cpp files as shown in the example.

Yelgurk, 2021-12-03:
Yes, I was just making sure of that while studying the library. I'm going to try it now. Thanks for the tips!

Yelgurk, 2021-12-03:
@lovyan03 
Got it! Please tell me, to select my display and set the connection pins, I need to manually change the lgfx_user file, right?
(I use ili9488)

lovyan03, 2021-12-03:
@Yelgurk 
In the lgfx_user folder is a sample configuration.
You can copy the file and give it any name you like.
You can put the file in the lgfx_user folder, but it may disappear when the library is updated.
To be safe, you may copy it to your project folder for use.

Yelgurk, 2021-12-03:
@lovyan03 
Did I understand correctly that to connect the display I need to do the following procedure:

1. Copy "LGFX_ESP32_sample.hpp" into the project folder;
2. Modify the copied settings file;
3. In `main.cpp` `#include <.../LGFX_ESP32_sample.hpp>`.

Right?

lovyan03, 2021-12-03:
@Yelgurk 
Yes, I think that's fine.
There are no strict rules, so treat it as you see fit.

Yelgurk, 2021-12-03:
@lovyan03 
With "LGFX_AUTODETECT.hpp" there is no problem to build. When using any of the examples like `#include <lgfx_user/LGFX_ESP32_sample.hpp>` the following error occurs:

```
.../esp-idf/components/LovyanGFX/src/lgfx_user/LGFX_ESP32_sample.hpp:21:7: error: using typedef-name 'using LGFX = class lgfx::v1::LGFX' after 'class'
 class LGFX : public lgfx::LGFX_Device
```

```
.../esp-idf/components/LovyanGFX/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32.hpp:1636:24: note: 'using LGFX = class lgfx::v1::LGFX' has a previous declaration here
 using LGFX = lgfx::LGFX;
```

To test the build, I used your code above, replacing only `#include <LGFX_AUTODETECT.hpp>`, with #include `<lgfx_user/LGFX_ESP32_sample.hpp>`

lovyan03, 2021-12-03:
@Yelgurk 
AUTODETECT is for auto-detection of major boards such as M5Stack and TTGO T-Watch.
If you don't want to use AUTODETECT, please remove "# define LGFX_AUTODETECT" this description.
Users should not use any functions related to AUTODETECT if they are doing their own configuration.

Yelgurk, 2021-12-03:
It's funny, as soon as I figure out the error myself and want to write about what I've got, you manage to give me some tips. That's funny! Thank you for your time and guidance!

lovyan03, 2021-12-03:
@Yelgurk 
oh…
I'm not a free support center…
so please read the errors yourself a bit…

Yelgurk, 2021-12-03:
I understand. I apologize. I'll be sure to sponsor you with my next paycheck! Thank you for your time and help.

closed 2021-12-03 by Yelgurk

Eduardo-Brandt, 2023-11-29:
> @Yelgurk
>
> I am able to use LovyanGFX with VSCode+ESP-IDF without any problems.
>
> you need `REQUIRES LovyanGFX` in CMakeLists.txt
>
> CMakeLists.txt
>
> ```
> idf_component_register(SRCS [...]

Hi [lovyan03](https://github.com/lovyan03) 

Thanks for this great Lib. I´ve used it in Arduino with ILI9486 paralell. 
But in ESP-IDF int the VSCode PlatformIO it not compiles. It shows-up some errors. What I did:

**I put a copy of the lib in project_name\components
**src\CMakeLists.txt
```
idf_component_register(SRCS "teste.cpp"
                    INCLUDE_DIRS "."
                    REQUIRES LovyanGFX)
```

**teste.cpp
```
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "sdkconfig.h"

#define LGFX_USE_V1
#define LGFX_AUTODETECT
#include <LovyanGFX.h>
#include <LGFX_AUTODETECT.hpp>

LGFX lcd;

extern "C"
{
  void app_main(void)
  {
    lcd.init();
    while(1) {
      vTaskDelay(1);
      lcd.print("TEST");
    }
  }
}
```

but it raises some errors as shown below:
```
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\platforms\esp32\common.o
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\touch\Touch_GT911.o
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\touch\Touch_NS2009.o
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\touch\Touch_STMPE610.o
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\touch\Touch_TT21xxx.o
Compiling .pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\touch\Touch_XPT2046.o
Compiling .pio\build\esp32-s3-devkitc-1\app_trace\app_trace.o
Compiling .pio\build\esp32-s3-devkitc-1\app_trace\app_trace_util.o
Compiling .pio\build\esp32-s3-devkitc-1\app_trace\host_file_io.o
Compiling .pio\build\esp32-s3-devkitc-1\app_trace\port\port_uart.o
Compiling .pio\build\esp32-s3-devkitc-1\app_update\esp_ota_ops.o
Archiving .pio\build\esp32-s3-devkitc-1\esp-idf\app_trace\libapp_trace.a
Compiling .pio\build\esp32-s3-devkitc-1\app_update\esp_ota_app_desc.o
Compiling .pio\build\esp32-s3-devkitc-1\bootloader_support\src\bootloader_common.o
Archiving .pio\build\esp32-s3-devkitc-1\esp-idf\app_update\libapp_update.a
components\LovyanGFX\src\lgfx\v1\platforms\esp32\common.cpp: In function 'void lgfx::v1::i2c::i2c_set_cmd(i2c_dev_t*, uint8_t, uint8_t, uint8_t, bool)':
components\LovyanGFX\src\lgfx\v1\platforms\esp32\common.cpp:812:14: error: 'struct i2c_dev_t' has no member named 'comd0'; did you mean 'comd'?
  812 |       (&dev->comd0)[index].val = cmd_val;
      |              ^~~~~
      |              comd
Compiling .pio\build\esp32-s3-devkitc-1\bootloader_support\src\bootloader_common_loader.o
Compiling .pio\build\esp32-s3-devkitc-1\bootloader_support\src\bootloader_clock_init.o
Compiling .pio\build\esp32-s3-devkitc-1\bootloader_support\src\bootloader_mem.o
Compiling .pio\build\esp32-s3-devkitc-1\bootloader_support\src\bootloader_random.o
*** [.pio\build\esp32-s3-devkitc-1\components\LovyanGFX\src\lgfx\v1\platforms\esp32\common.o] Error 1
====================================================== [FAILED] Took 9.53 seconds ========
```

I´m stucked on that. Is there some incompatibility of the lib? Or what can I´m doing wrong?

Thank you a lot!

mhaberler, 2023-11-29:
@Eduardo-Brandt upgrade ESP-IDF to 5.1.2 and it works.