lovyan03/LovyanGFX #177: Displaying on SH1106 shifted

johalaoui, 2021-12-23:
Hello @lovyan03 ,

I have encountered this problem:
[image]
After adding `cfg.prefix_len = 3;` it looked like the problem was fixed, but this broke other things instead:
[image]

`cfg.offset_x = 2;`:
[image]
`cfg.offset_x = -2;`:
[image]
`cfg.offset_y = 2;`:
[image]
`cfg.offset_y = -2;`:
[image]
Datasheet: https://www.berrybase.de/en/Pixelpdfdata/Articlepdf/id/8082/onumber/OLED-12864-B
The current configuration is posted below and I would appreciate any help.

Thanks in advance.

```/*
 * LGFX_user_SH1106.hpp
 *  similar to LovyanGFX/src/lgfx_user/LGFX_ESP32_sample.hpp
 *  Created on: Nov 4, 2021
 */

#pragma once

#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include "pinMapEsp32.h"

class LGFX : public lgfx::LGFX_Device {
  lgfx::Panel_SH110x _panel_instance;
  lgfx::Bus_I2C _bus_instance;

 public:

  LGFX(void) {
    {
      auto cfg = _bus_instance.config();

      cfg.i2c_port = 0;
      cfg.freq_write = 400000;
      cfg.freq_read = 400000;
      cfg.pin_sda = i2c_sda;
      cfg.pin_scl = i2c_scl;
      cfg.i2c_addr = 0x3C;

      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);

    {
      auto cfg = _panel_instance.config();

      cfg.memory_width = 128;
      cfg.memory_height = 64;
      cfg.panel_width = 128;
      cfg.panel_height = 64;
      cfg.offset_x = 0;
      cfg.offset_y = 0;

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance);
  }
};
```

tobozo, 2021-12-23:
hey :-)

This display is 132 px, not 128.

Another possibility is to use the SSD1306 driver if you don't mind the missing 4x64 block.

johalaoui, 2021-12-23:
Hello @tobozo,
I have tried using the SSD1306, but it doesn't show any image at all.
The display output looks like it is shifted by 2 pixels rather than 4.
The U8g2 library does work, but my current work is not based on it and LovyanGFX is faster.
[image]

```
#include <U8g2lib.h>
#include <pinMapEsp32.h>
U8G2_SH1106_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/U8X8_PIN_NONE, i2c_scl, i2c_sda);
void setup(void) {
  display.begin();
  display.setFontPosTop();
  display.setFont(u8g2_font_ncenB14_tr);
}

void loop() {
  display.drawStr(0, 0, "Hi");
  display.sendBuffer();
  delay(1000);
}
```

tobozo, 2021-12-23:
what about this?

```
      cfg.memory_width = 132;
      cfg.memory_height = 64;
      cfg.panel_width = 132;
      cfg.panel_height = 64;
      cfg.offset_x = 2;
      cfg.offset_y = 0;
```

johalaoui, 2021-12-23:
> what about this?
> 
> ```
>       cfg.memory_width = 132;
>       cfg.memory_height = 64;
>       cfg.panel_width = 132;
>       cfg.panel_height = 64;
>       cfg.offset_x = 2;
>       cfg.offset_y = 0;
> ```

I have tested this with a picture in the first comment.

johalaoui, 2021-12-23:
> what about this?
> 
> ```
>       cfg.memory_width = 132;
>       cfg.memory_height = 64;
>       cfg.panel_width = 132;
>       cfg.panel_height = 64;
>       cfg.offset_x = 2;
>       cfg.offset_y = 0;
> ```
There is no image.

lovyan03, 2021-12-23:
@johalaoui @tobozo 
Sorry for the late reply.

memory_width and memory_height : set the sizes supported by the driver IC.
panel_width and panel_height : set the size that can be displayed.

```
cfg.memory_width = 132;
cfg.memory_height = 64;
cfg.panel_width = 128;
cfg.panel_height = 64;
cfg.offset_x = 2;
cfg.offset_y = 0;
```

johalaoui, 2021-12-23:
> @johalaoui @tobozo Sorry for the late reply.
>
> memory_width and memory_height : set the sizes supported by the driver IC. panel_width and panel_height : set the size that can be displayed.
>
> ``` [...]
After this change the display shows this:
[image]

lovyan03, 2021-12-23:
@johalaoui 
umm... 
Then maybe offset_x = 0; is fine...?

lovyan03, 2021-12-24:
@johalaoui 
Fixed a bug in the develop branch. Please try using this.
I think this is the configuration you can use.

```
cfg.memory_width = 132;
cfg.memory_height = 64;
cfg.panel_width = 128;
cfg.panel_height = 64;
cfg.offset_x = 2;
cfg.offset_y = 0;
```

johalaoui, 2021-12-24:
Excellent work👍️
It works now as expected. In my case it also works when I only set `cfg.offset_x = 2;`. But I will use your config out of compatibility reasons.

Thank You @lovyan03

[image]

closed 2021-12-24 by johalaoui