lovyan03/LovyanGFX #179: Problem XPT2046 and Deep Sleep.

boromyr, 2022-01-03:
Hi, I am writing to ask you for help with a problem I am meeting, that is with the Deep Sleep of the ESP32.
- MCU:ESP32-S1
- Framework: Arduino
- Driver TFT: ILI9341
- Driver Touch: XPT2046
- cfg.pin_int  = PENIRQ;       
- cfg.bus_shared  = true; 

I want to implement a simple awakening to the touch in my project via the PenirQ pin of the XPT2046. I have configured everything correctly in the code, and the Sleep and Wakeup code works correctly when I use ESP_LIGHT_SLEEP_START(), but not with ESP_DEEP_SLEEP_START().

`void wakeup() {
  RESET_CD;
  // tft.wakeup();
  esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_EXT0);
  ESP.restart();
}
void sleep() {
  mess("sleeping...");
  tft.sleep();
  adc.powerDown();
  esp_sleep_enable_ext0_wakeup(PENIRQ, 0);
  esp_light_sleep_start();
}`

I checked with the multimeter that PenirQ pin always remains HIGH when touching the screen in Deep Sleep mode. Since PenirQ does not use the SPI connection, I can't understand where the problem can reside.
If I have correctly interpreted TOUCH_XPT2046.CPP, PD0 and PD1 should be {0.0}, which should be corrected for my goal.

lovyan03, 2022-01-05:
hello @boromyr 
Is there a problem with the operation of LovyanGFX?
If the problem is outside of LovyanGFX, there is nothing to answer here.
To me it looks like a purely ESP32 and XPT2046 control issue.

If you want to give me some idea about this issue, please provide the smallest code that can reproduce the problem.
The information you presented is an abstraction of the pin number and does not provide any information.

boromyr, 2022-01-16:
The problem resolved itself after updating the library, in fact I hadn't done that in a while.

I want to take this opportunity to ask you for some specific information.
Does the sleep() function just turn off the backlight, or does it also put the driver in shoot-down(for ILI9341 and ST7789)?
If I wanted to edit PD0 and PD1 and disable the internal VREF of the XPT2046, what lines should I edit?
Thanks.

closed 2022-12-25 by github-actions[bot]