lovyan03/LovyanGFX #190: Support for RM68120

Sasa1982, 2022-01-20:
Hello,

is it possible or maybe almost planed to add support for a RM68120 controller.
So it could be used with the ESP32-S2-HMI-DevKit-1

https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s2/esp32-s2-hmi-devkit-1/user_guide.html

That would be great !!

best regards

lovyan03, 2022-03-14:
@Sasa1982 
I have arranged for RM68120.
I will implement it eventually when I get time to work on it.

lovyan03, 2022-06-06:
I have now added Panel_RM68120 to the experimental branch.
However, I have not confirmed that it works.
Please give it a try if you are so inclined.
If you find something that needs to be adjusted, please try modifying it based on this.

closed 2022-06-24 by lovyan03

phil31, 2022-10-23:
Hello,

i can confirm that the RM68120 works !
i'm suing this board : HMI DevKit ESP32 LVGL ESP32-S3-CODEC LCD parallèle 8080, MPU-6050

[](https://fr.aliexpress.com/item/1005003814428825.html?spm=a2g0o.order_list.0.0.d0435e5bmnRG4C&gatewayAdapt=glo2fra)